bblanchon/ArduinoJson #1419: String class error with serializeJson on ESP8266 and PlatformIO

fabioinfo44, 2020-10-26:
```
#define ARDUINOJSON_ENABLE_STD_STREAM
#include <ArduinoJson.h>
StaticJsonDocument<1024> json;
```
then
```
File file = SPIFFS.open("config.json", "w");
serializeJson(json, file);
file.close();
```
and the output is:
```
lib\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp:22:29: error: 'const class String' has no member named 'write'
     return _dest->write(s, n);
                                         ^
```
and
```
lib\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp:18:26: error: 'const class String' has no member named 'write'
     return _dest->write(c);
                                       ^
```

bblanchon, 2020-10-27:
Hi @fabioinfo44,

Please try with [ARDUINOJSON_ENABLE_ARDUINO_STRING](https://arduinojson.org/v6/api/config/enable_arduino_string/).

Best regards,
Benoit

fabioinfo44, 2020-10-27:
I will try ASAP and let you know.
Thank you in advance

closed 2020-11-03 by bblanchon