bblanchon/ArduinoJson #1422: Copy JsonArray to 2D intArray

Fredlesgars, 2020-10-28:
Hi,

I have a problem using .copyTo --> jsonArray.copyTo(intArray) #254 and #251
I used the parsing program from de ArduinoJson assistant and I am trying to copy the JsonArray to an intarray[7][4].
According to the #254 .copyTo should be a member of the class.

Here is my code (sorry for the format, it's my first time opening an issue and I can't seam to make it work properly) :

`const size_t capacity = 29 * JSON_ARRAY_SIZE(4) + 4 * JSON_ARRAY_SIZE(7) + 4 * JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(3) + 70;
    DynamicJsonDocument doc(capacity);

    const char *json = "{\"type\":\"Order\",\"display\":\"Hermenegilde Dubois\",\"meds\":[{\"bucketID\":1,\"matrix\":[[1,0,0,1], 
    [1,0,0,1],[1,0,0,1],[1,0,0,1],[1,0,0,1],[1,0,0,1],[1,0,0,1]]},{\"bucketID\":3,\"matrix\":[[1,0,0,0],[1,0,0,0],[1,0,0,0],[1,0,0,0],[1,0,0,0], 
    [1,0,0,0],[1,0,0,0]]},{\"bucketID\":5,\"matrix\":[[0,0,0,1],[0,0,0,1],[0,0,0,1],[0,0,0,1],[0,0,0,1],[0,0,0,1],[0,0,0,1]]}, 
    {\"bucketID\":7,\"matrix\":[[0,0,0,1],[0,0,0,1],[0,0,0,1],[0,0,0,1],[0,0,0,1],[0,0,0,1],[0,0,0,1]]}]}";

    deserializeJson(doc, json);

    const char *type = doc["type"];       // "Order"
    const char *display = doc["display"]; // "Hermenegilde Dubois"

    JsonArray meds = doc["meds"];

    int meds_0_bucketID = meds[0]["bucketID"]; // 1

    JsonArray meds_0_matrix = meds[0]["matrix"];

    JsonArray meds_0_matrix_0 = meds_0_matrix[0];

    int med[7][4];
    meds_0_matrix.copyTo(med);`

I get this error : 
[image]

I'm using PlatformIO 5.0.2 in VScode with a Teensy 3.5

Compiler : g++.exe (x86_64-posix-seh-rev0, Built by MinGW-W64 project) 8.1.0

Let me know what I am doing wrong!

Thanks in advance.

bblanchon, 2020-10-29:
Hi @Fredlesgars,

Please see [copyArray()](https://arduinojson.org/v6/api/misc/copyarray/).

Best regards,
Benoit

Fredlesgars, 2020-10-29:
Salut Benoit,

Merci beaucoup pour la réponse rapide, ça fonctionne très bien!

Bonne journée,

Fred

closed 2020-10-29 by Fredlesgars

bblanchon, 2020-10-30:
De rien, Fred. Merci d'utiliser ArduinoJson.
Bonne chance pour ton projet!