bblanchon/ArduinoJson #1424: [Bug:]  DeserializationError error -> TooDeep (v6.17)

hasenradball, 2020-10-29:
Hi Benoit,

with the v 6.17 I get the following error:
```
Loading configuration...
 >>> ERROR: deserializeJson() failed:  TooDeep
```
Using the following Code:
```
bool cConfigIO::readConfig(void) {
  DBG__PRINT(F("\n Loading configuration..."));
  // Open file for reading
  File file = LittleFS.open(filename, "r");
  if (!file) {
    DBG__PRINT(F(">>> ERROR: Failed to open \"config.txt\" file!"));
    return false;
  }
  StaticJsonDocument<512> doc;

  // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, file);
  if (error) {
    DBG__PRINT(F(">>> ERROR: deserializeJson() failed:"), " ");
    DBG__PRINT(error.f_str());
    return false;
  }
  else
    //serializeJsonPretty(doc, Serial);
    serializeJson(doc, Serial);
    DBG__PRINT("\n");
```
The code runs with v6.16
Is there an issue with the v6.17?

The serial output looks like:
```
 >>> Starte Filesystem...
        filesystem mounted successfully!

        FILE: config.txt - 0.33 kB
        FILE: index.html - 6.01 kB
        FILE: script.js - 10.89 kB
        FILE: wordclock.css - 4.59 kB
 Total size: 21.83 kB
 
 Loading configuration...
 >>> ERROR: deserializeJson() of >config.txt< failed: TooDeep!

 ERROR: read of "config.txt" not sucessful!
 
 Booting...
 ERROR: keine Konfiguration!

### --- Starte Access Point --- ###
 INFO: Set Access Point Config: true
 INFO: Access Point gestarted
 SSID: Word-Clock IP Adresse: 192.168.55.1
```

bblanchon, 2020-10-30:
Hi @hasenradball,

Thank you for reporting this problem.
No, this is not a known bug.

Please provide a [reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) that works on 6.16 and fails on 6.17.
Feel free to use wandbox to share the reprex: [6.16.1](https://wandbox.org/permlink/jjU20aywLlxbnNWw), [6.17.0](https://wandbox.org/permlink/qfo3YnnE3bXY0rSH).

Best regards,
Benoit

hasenradball, 2020-10-30:
As discussed this Issue was introduced by me ...
I forgot to save a default Configuration... so the reading of it can not work.

closed 2020-10-30 by hasenradball