bblanchon/ArduinoJson #1425: serializeJSON asking for initialization

aaronfield33, 2020-10-29:
Hey,

I am very unfamiliar with ArduinoJSON as well as programming. I am learning a lot as I go but have been running into some issues. Basically, I am trying to pull two temperature and humidity readings from DHT22's (on an ESP-12E) into Losant for monitoring. I have been trying this for some time and getting almost nowhere. I tried following a Losant tutorial on how to do it for a single sensor and converting it to double sensors. The problem with the tutorial seems to be that it uses ArduinoJSON 5 and 6. I tried installing 5 just to see if it would work and I got more errors. I tried converting some of the serialization (starting on line 75) over to ArduinoJSON 6 and cannot figure out what I am doing wrong. The error codes I am getting are attached in a TXT. 

The tutorial is here: https://www.losant.com/blog/getting-started-with-the-esp8266-and-dht22-sensor

My code for running the dual sensors just over serial is here: https://github.com/aaronfield33/Gecko-Sensing

My code for the Losant dual sensors is also on my GitHub. I have not looked further into the next serialization portion of the code (of which I am getting errors) but I figured once I have the first section figured out, the second section should be much easier to understand. 

[Losant_Dual_Sensor_Errors.txt](https://github.com/bblanchon/ArduinoJson/files/5461477/Losant_Dual_Sensor_Errors.txt)

closed 2020-10-30 by aaronfield33

bblanchon, 2020-10-30:
Hi @aaronfield33,

It's hard to tell what's wrong without seeing the code.
Did you fix your problem?

Best regards,
Benoit

aaronfield33, 2020-10-30:
Hey @bblanchon,

Yes, I figured it out. All I missed was a semicolon and that caused me all sorts of pain. Now what I am stuck on is some of the Losant code for my device.

Thanks,
Aaron Field