bblanchon/ArduinoJson #1428: Copy of std::string not being created when atrributing to JsonObject

Majubs, 2020-11-04:
Hello,

I'm having some difficulty understanding why ArduinoJson is not creating a copy of a `std:string`. Here is the situation, I have an `std::unordered_map<std::string, std::string>` creatred in a function, meaning it'll be destructed when function terminates. In this function I call another function that takes a pointer to the unordered_map as parameter:

    void function1()
    {
        std::unordered_map<std::string, std::string> temp_umap = {/*fill umap*/};
        function2(&temp_umap);
    }

Inside `function2()` I fill a JsonObject from the unordered_map like this:

```
void function2(std::unordered_map<std::string, std::string> * umap)
{
    /* jsonRoot is a  StaticJsonDocument */
    JsonObject jsonObj = this->jsonRoot.createNestedObject(String(key));

    for(auto & it = umap->begin(); it != umap->end(); ++it)
    {
        jsonObj[String(it->first.c_str())] = it->second;
    }
}
```

From my understanding, `it->second` should return a `std::string`, which according to the [documentation](https://arduinojson.org/v6/api/jsondocument/subscript/) should cause a copy of that string to be created, instead of being referenced. I want a copy since I know umap will cease to exist.

I tried to look at C++ docs to undestand how `it->second` works, but I'm a little confused since I have a pointer to the unordered_map.

I could create a temp `char[]` and copy `it->second` to it (using `sprintf()` for example), or do the same as the key with `String(it->second.c_str())` and see if it works, but I'd like to properly understand what is happening with all the pointers, and also know what would be the best way to do this, be it one of the things I suggested, or something else.

I know this more of wuestion that a issue, but I don't know a better forum to ask this. 

Thanks in advance.

bblanchon, 2020-11-05:
Hi @Majubs,

What makes you think that the string is not copied?

Best regards,
Benoit

Majubs, 2020-11-05:
The output when I print the json contents. The kays are correct, but the values are not. 

The first time I call `function2()` it's correct, but on subsequent calls previous values get overwritten with the new values, or with gibberish. For example, at first I have:

```
{
    "1": {
        "key1": 10,
        "key2": 100,
        "key3": 1000,
    }
}
```

In a second call to `function2()` I get:

```
{
    "1": {
        "key1": 10,
        "key2": 200,
        "key3": 2000,
    }
    "2": {
        "key1": 20,
        "key2": 200,
        "key3": 2000,
    }
}
```

And if call `function2()` one more time:

```
{
    "1": {
        "key1": 10,
        "key2": 200,
        "key3": 2000,
    }
    "2": {
        "key1": 20,
        "key2": 200,
        "key3": 2000,
    }
    "3": {
        "key1": randomstuff,
        "key2": 300,
        "key3": morerandomstuff,
    }
}
```

Sometimes the "randomstuff" are the contents of other variables, which seems like behaviour when pointers get lost. 

My guess is that it's some interaction with `it->second` that is not bahaving as expected, but I don't know C++ or ArduinoJson enough to know what it is.

bblanchon, 2020-11-06:
The code you posted inserts strings in the nested object; how can you have integers in the output?

Majubs, 2020-11-11:
I just forgot to put quotes in my examples, but it's all strings. I used numbers to make it easier to see the differences.

bblanchon, 2020-11-12:
There is nothing special with `pair::second`; see this demo:
https://wandbox.org/permlink/7ep1fYgl7S5XXCdI

The only explanation I see is that the [StaticJsonDocument](https://arduinojson.org/v6/api/staticjsondocument/) is too large and causes a stack overflow.

1. What is the capacity of this document?
2. On what platform is this code running?
3. Did you with a [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/)?

closed 2020-11-19 by bblanchon