bblanchon/ArduinoJson #1430: IncompleteInput

Findus76, 2020-11-07:
Hi

Now I have been looking at the code for hours without realizing what is wrong.
Version, 6.17.1
Get error: "IncompleteInput"
I have tried to increase 500 to 1500 with the same result.

Brg

Code
bool CheckJSONfunctionString(char* functionList, uint16_t Listleght)
{
	bool status = false;
	Serial.print("Checking function string\r\n");
	Serial.print("input string: ");
	Serial.println(functionList);
	DynamicJsonDocument FunclistDoc(Listleght + 500);
	DeserializationError err = deserializeJson(FunclistDoc, (const char*)functionList, Listleght);
	if (!err) 
	{

Console output
Checking function string
input string: {"Type":"SystemFunction","Function list":{"Relay ch1":{"Settings":{},"Info":{"Position":"Garage","Description":"Solidstate max 2A"},"Relay":{"Head Type":"Relay","Sub Type":"Main Voltage","R/W":"W","State":"ON/OFF"}},"Relay ch2":{"Settings":{},"Info":{"Position":"Garage","Description":"Solidstate max 2A"},"Relay":{"Head Type":"Relay","Sub Type":"Main Voltage","R/W":"W","State":"ON/OFF"}},"Relay ch3":{"Settings":{},"Info":{"Position":"Garage","Description":"Solidstate max 2A"},"Relay":{"Head Type":"Relay","Sub Type":"Main Voltage","R/W":"W","State":"ON/OFF"}},"Relay ch4":{"Settings":{},"Info":{"Position":"Garage","Description":"Solidstate max 2A"},"Relay":{"Head Type":"Relay","Sub Type":"Main Voltage","R/W":"W","State":"ON/OFF"}},"Temperature #1":{"Settings":{"Sensor":{"Calibration(°C)":0,"DeltaChangeUpdate(°C)":1}},"Info":{"Position":"Garage","Description":"Range -55°C to +125°C, ±0.5°C Accuracy"},"Sensor":{"Head Type":"Sensor","Sub Type":"Temperature","R/W":"R","Unit":"°C"}},"Temperature #2":{"Settings":{"Sensor":{"Calibration(°C)":0,"DeltaChangeUpdate(°C)":2}},"Info":{"Position":"Outside Front","Description":"Range -55°C to +125°C, ±0.5°C Accuracy"},"Sensor":{"Head Type":"Sensor","Sub Type":"Temperature","R/W":"R","Unit":"°C"}}},"System":{"Settings":{},"Info":{"Type":"DIN module v1","Description":"4ch Relay, RS485, OneWire, 1ch analog input"}}
deserializeJson() failed: IncompleteInput

Findus76, 2020-11-08:
Update
Downgrade to 6.15.0
Get error
Leght: 1380
deserializeJson() failed: NoMemory
Strange. This routine is run before WiFi, MQTT, etc. have been set up. There should be plenty of memory

bblanchon, 2020-11-08:
Hi @Findus76,

You called the overload of [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) that takes the length of the input as the third parameter.
Please verify that you pass the right length or remove this third argument.

ArduinoJson 6.15 requires more memory because it doesn't deduplicate strings; this feature was added in [version 6.16](https://arduinojson.org/news/2020/08/01/version-6-16-0/).

Best regards,
Benoit

Findus76, 2020-11-08:
Thanks.

Has upgrade back to 6.17.1
 I have tried to remove the third argument, get the same error message.
If I change `"DynamicJsonDocument FunclistDoc (Listleght + 500);"` to `"DynamicJsonDocument FunclistDoc (Listleght + 200);"`
I get error messages: NoMemory

I have validated that the string length is 1380bytes
Tried changing `"DeserializationError err = deserializeJson (FunclistDoc, (const char *) functionList, Listleght);"` to `"DeserializationError err = deserializeJson (FunclistDoc, (const char *) functionList, Listleght-1);"` Makes no difference, error message: IncompleteInput

Brg

bblanchon, 2020-11-09:
Hi @Findus76,

Regarding the [`NoMemory`](https://arduinojson.org/v6/api/misc/deserializationerror/#nomemory), please read [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)

Regarding the [`IncompleteInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#incompleteinput), the document you provided in your first message is, indeed, incomplete: it lacks the last closing brace. The input should end with three braces, as you can see below:

```json
{
    "Type": "SystemFunction",
    "Function list": {
        "Relay ch1": {
            "Settings": {},
            "Info": {
                "Position": "Garage",
                "Description": "Solidstate max 2A"
            },
            "Relay": {
                "Head Type": "Relay",
                "Sub Type": "Main Voltage",
                "R/W": "W",
                "State": "ON/OFF"
            }
        },
        "Relay ch2": {
            "Settings": {},
            "Info": {
                "Position": "Garage",
                "Description": "Solidstate max 2A"
            },
            "Relay": {
                "Head Type": "Relay",
                "Sub Type": "Main Voltage",
                "R/W": "W",
                "State": "ON/OFF"
            }
        },
        "Relay ch3": {
            "Settings": {},
            "Info": {
                "Position": "Garage",
                "Description": "Solidstate max 2A"
            },
            "Relay": {
                "Head Type": "Relay",
                "Sub Type": "Main Voltage",
                "R/W": "W",
                "State": "ON/OFF"
            }
        },
        "Relay ch4": {
            "Settings": {},
            "Info": {
                "Position": "Garage",
                "Description": "Solidstate max 2A"
            },
            "Relay": {
                "Head Type": "Relay",
                "Sub Type": "Main Voltage",
                "R/W": "W",
                "State": "ON/OFF"
            }
        },
        "Temperature #1": {
            "Settings": {
                "Sensor": {
                    "Calibration(°C)": 0,
                    "DeltaChangeUpdate(°C)": 1
                }
            },
            "Info": {
                "Position": "Garage",
                "Description": "Range -55°C to +125°C, ±0.5°C Accuracy"
            },
            "Sensor": {
                "Head Type": "Sensor",
                "Sub Type": "Temperature",
                "R/W": "R",
                "Unit": "°C."
            }
        },
        "Temperature #2": {
            "Settings": {
                "Sensor": {
                    "Calibration(°C)": 0,
                    "DeltaChangeUpdate(°C)": 2
                }
            },
            "Info": {
                "Position": "Outside Front",
                "Description": "Range -55°C to +125°C, ±0.5°C Accuracy"
            },
            "Sensor": {
                "Head Type": "Sensor",
                "Sub Type": "Temperature",
                "R/W": "R",
                "Unit": "°C"
            }
        }
    },
    "System": {
        "Settings": {},
        "Info": {
            "Type": "DIN module v1",
            "Description": "4ch Relay, RS485, OneWire, 1ch analog input"
        }
    }
}
```

After adding the missing brace, [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) returns `Ok`, as you can see in this online demo: https://wandbox.org/permlink/8h7vXUJ3SOKfdN4o

Best regards,
Benoit

Findus76, 2020-11-09:
Embarrassing! 
I use the plugin for notepad ++ to do syntax check. But this does not plug in get any fault about. 
Do you have a program you recommend for syntax checking by JSON?

Brg

bblanchon, 2020-11-10:
Use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/): it says "Unexpected end of JSON input" in this case.

closed 2020-11-10 by bblanchon