bblanchon/ArduinoJson #1431: ESP8266:12e => corrupted output from jsonDocument in case of usage inside class constructor

Abhimanyu6676, 2020-11-08:
**BOARD AND FRAME**
  * platform = espressif8266
  * board = nodemcuv2
  * framework = arduino
  * arduinoJson = 6.17.1

#To reproduce

## main.cpp

```
#include "main.hpp"
//#include "ws.hpp"
#inclide <ArduinoJson.h>

void setup(){
Serial.begin(115200);
delay(1000);
Serial.println("<<<<<<<<<<<<<<>>>>>>>>>>>>>>");
printJsonTest();
}

void loop(){

}

void printJsonTest()
{
	DynamicJsonDocument doc(500);
	JsonObject obj = doc.to<JsonObject>();
	doc["a"] = 34;
	doc["d"] = 194;
	doc["temp1"] = String("gcouglb.j");
	doc["temp2"] = String("gcouglb.j");
	doc["temp3"] = String("gco uglb.j");
	doc["temp4"] = String("gco uglb.j");
	doc["temp5"] = String("gco uglb.j");
	doc["temp6"] = String("gco uglb.j");
	doc["temp7"] = String("gco uglb.j");
	serializeJson(doc, Serial);
        String temp;
        serializeJson(doc, temp);
        //if(wsClientConnected())
             //wsSendAll(temp);
	doc.clear();
}
```

## main.hpp
```

void printJsonTest();
```

## customClass.hpp

```
#pragma once
#include "main.hpp"

class customClass 
{
private:

public:
    customClass();
    ~customClass();

    void init();

};
```

## customClass.cpp

```
#include <Arduino.h>

customClass::customClass()
{
     printJsonTest();
}

customClass::~customClass()
{
}

void customClass::init()
{
}

```

# The output from the code is 

**{"a":34,"d":194,"temp1":"\N\ \ \N","temp2":"\N\ \ \N","temp3":"gco\Nuglb.j","temp4":"gco\Nuglb.j","temp5":"gco\Nuglb.j","temp6":"gco\Nuglb.j","temp7":"gco\Nuglb.j"}**

#### this not something specific to serial cause I am getting the same string over websocket and in http test

#### I have also tested with ArduinoJson version 5.13.1 and 5.13.4. there are some changes to output pattern but it comes corrupted somehow

## however if the **printJsonTest()** code is changed as such

{
	StaticJsonDocument<500> doc;
	//JsonObject obj = doc.to<JsonObject>();
	doc["a"] = 34;
	doc["d"] = 194;
	doc["temp1"] = String("gco uglb.j");
	doc["temp2"] = String("gco uglb.j");
	doc["temp3"] = String("gco uglb.j");
	doc["temp4"] = String("gco uglb.j");
	doc["temp5"] = String("gco uglb.j");
	doc["temp6"] = String("gco uglb.j");
	doc["temp7"] = String("gco uglb.j");
	serializeJson(doc, Serial);
	//doc.clear();
}

**output is =>**
**{"a":34,"d":194,"temp1":"\N\ \ \N","temp2":"\N\ \ \N","temp3":"\N\ \ \N","temp4":"\N\ \ \N","temp5":"\N\ \ \N","temp6":"\N\ \ \N","temp7":"\N\ \ \N"}**

bblanchon, 2020-11-08:
Hi @Abhimanyu6676,

I'm pretty sure the problem comes from the fact that you call this function *before* the `setup()` function.
I don't know the details but I don't think ArduinoJson is to blame here.

Best regards,
Benoit

Abhimanyu6676, 2020-11-10:
Have tried after setup also. over http server now that is inside main loop but results are same

bblanchon, 2020-11-11:
Please provide an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) to I can reproduce the problem here.

Abhimanyu6676, 2020-11-18:
[u can find it here](https://github.com/bblanchon/ArduinoJson/issues/1431)

bblanchon, 2020-11-19:
In a previous message, you said you could reproduce the issue when `serializeJson()` is called after `setup()`.
Would you mind sharing the code?

closed 2020-12-18 by bblanchon