bblanchon/ArduinoJson #1432: BUG ON UPDATE

laercionit, 2020-11-09:
I am not able to identify the error, but it is having a problem in the compilation after the update. Can someone guide me please?

bool SensorLoadSD(const char* Filename, const byte Total = SensorTotal) {
  if (FilesSDOpen(Filename)) {
    MyFileSD.find("[");
    byte ObjectID = 0;
    do {
      DynamicJsonDocument JSON(JsonSettingsBuffer);
      DeserializationError err = deserializeJson(JSON, MyFileSD);
      if (!err) {
        JsonObject Object = JSON.as<JsonObject>();
        Sensor[ObjectID].ID                   = Object[F("ID")]   | Sensor[ObjectID].ID;
        Sensor[ObjectID].Pin                  = Object[F("PIN")]  | Sensor[ObjectID].Pin;
        Sensor[ObjectID].Type                 = Object[F("TP")]   | Sensor[ObjectID].Type;
        Sensor[ObjectID].Model                = Object[F("MDL")]  | Sensor[ObjectID].Model;
        strlcpy(Sensor[ObjectID].Name         , Object[F("NM")]   | Sensor[ObjectID].Name,          sizeof(Sensor[ObjectID].Name));
        Sensor[ObjectID].Rate                 = Object[F("RT")]   | Sensor[ObjectID].Rate;
        Sensor[ObjectID].LowAlert             = Object[F("LW")]   | Sensor[ObjectID].LowAlert;
        Sensor[ObjectID].HighAlert            = Object[F("HG")]   | Sensor[ObjectID].HighAlert;
        Sensor[ObjectID].Favorite             = Object[F("FAV")]  | Sensor[ObjectID].Favorite;
        ObjectID++;
      } else return 0;
    } while ((MyFileSD.findUntil(",", "]")) && (MyFileSD.available()) && (ObjectID < Total));
    String Formated = F("SensorLoadSD: %s"); snprintf_P(Buffer, sizeof(Buffer), Formated.c_str(), Filename); DebugSerial.println(Buffer);
    FilesSDClose();
    return 1;
  }
  return 0;
}

bblanchon, 2020-11-09:
Hi @laercionit,

Thank you very much for reporting this bug.
It'll be fixed very soon; until then, you can revert back to 6.17.0 or cast `Sensor[ObjectID].Name` to a `const char*`:

```c++
strlcpy(Sensor[ObjectID].Name, Object[F("NM")] | (const char*)Sensor[ObjectID].Name, sizeof(Sensor[ObjectID].Name));
```

Best regards,
Benoit

closed 2020-11-10 by bblanchon

bblanchon, 2020-11-14:
The fix was published in [version 6.17.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.2).

commit 30da9201 (2020-11-10): Fixed invalid conversion in `operator|(Variant, char*)` (fixes #1432)