bblanchon/ArduinoJson #1433: Particle compiler error in an empty project

chinesefellow, 2020-11-09:
Tried including the ArduinoJson library today. Version 6.17.1. Without writing any additional code besides having an empty setup and loop function, a compiler error is generated.

I have another project/app in the Particle environment using version 6.14.1 and that one is still working. 

Here's the raw error:
```
Processing  arduinojsonexample.ino
Checking library ArduinoJson...
Installing library ArduinoJson 6.17.1 to lib/ArduinoJson ...
Library ArduinoJson 6.17.1 installed.
make -C ../modules/argon/user-part all
make[1]: Entering directory '/firmware/modules/argon/user-part'

mkdir -p target/
make -f ../../../modules/shared/nRF52840/build_linker_script.mk PREBUILD=1
make[2]: Entering directory '/firmware/modules/argon/user-part'
Creating target/platform_user_ram.ld ...
make[2]: Leaving directory '/firmware/modules/argon/user-part'

make -C ../../../user 
make[2]: Entering directory '/firmware/user'
Building cpp file: src/arduinojsonexample.cpp
Invoking: ARM GCC CPP Compiler
mkdir -p ../build/target/user/platform-12-msrc/
arm-none-eabi-gcc -DSTM32_DEVICE -DnRF52840 -DNRF52840_XXAA -DPLATFORM_THREADING=1 -DPLATFORM_ID=12 -DPLATFORM_NAME=argon -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00C -DUSBD_PID_CDC=0xC00C -DSPARK_PLATFORM -g3 -gdwarf-2 -Os -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DINCLUDE_PLATFORM=1 -DSOFTDEVICE_PRESENT=1 -DS140 -DPRODUCT_ID=12 -DPRODUCT_FIRMWARE_VERSION=65535 -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-config-project.h\" -DENABLE_FEM=1 -DNRF_802154_PROJECT_CONFIG=\"openthread-platform-config.h\" -DRAAL_SOFTDEVICE=1 -D_WIZCHIP_=W5500 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DLFS_CONFIG=lfs_config.h -DSYSTEM_VERSION_STRING=1.5.2 -DRELEASE_BUILD -I./inc -I../wiring/inc -I../system/inc -I../third_party/miniz/miniz -I../services/inc -I../third_party/nanopb/nanopb -I../communication/inc -I../hal/inc -I../hal/shared -I../hal/src/argon -I../hal/network/lwip/esp32 -I../hal/src/nRF52840 -I../hal/src/armv7 -I../hal/src/nRF52840 -I../hal/src/nRF52840/lwip -I../hal/src/nRF52840/freertos -I../hal/src/nRF52840/openthread -I../hal/src/nRF52840/mbedtls -I../hal/src/nRF52840/littlefs -I../hal -I../hal/network/api -I../hal/network/lwip -I../hal/network/lwip/posix -I../hal/network/openthread -I../hal/network/lwip/wiznet -I../hal/network/ncp -I../hal/network/ncp/at_parser -I../third_party/lwip/lwip/src/include -I../third_party/freertos/freertos/FreeRTOS/Source/include -I../third_party/nrf5_sdk/nrf5_sdk/external/freertos/portable/GCC/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/external/freertos/portable/CMSIS/nrf52 -I../third_party/openthread/openthread/include -I../third_party/openthread/openthread/src/core -I../third_party/openthread/openthread/examples/platforms -I../third_party/openthread/openthread/radio -I../third_party/openthread/openthread/radio/hal -I../third_party/openthread/openthread/radio/rsch -I../third_party/openthread/openthread/radio/rsch/raal -I../third_party/openthread/openthread/radio/rsch/raal/softdevice -I../third_party/wiznet_driver/wiznet_driver/Ethernet -I../gsm0710muxer/gsm0710muxer/include -I../platform/shared/inc -I../third_party/nrf5_sdk -I../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/include -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/mdk -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/src/prs -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/util -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/common -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fstorage -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_section_vars -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/crc32 -I../third_party/nrf5_sdk/nrf5_sdk/integration/nrfx/legacy -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/delay -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/log -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/log -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/log/src -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_section_vars -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/delay -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/queue -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/common -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/mbr/nrf52840/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140 -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fifo -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic_fifo -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/strerror -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fstorage -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic_flags -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd/class/cdc -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/usbd/class/cdc/acm -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/common -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/nrf_ble_gatt -I../third_party/nrf5_sdk/nrf5_sdk/components/ble/ble_advertising -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/t2t_lib -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/platform -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/generic/message -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/generic/record -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/lauchapp -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/text -I../third_party/nrf5_sdk/nrf5_sdk/components/nfc/ndef/uri -I../third_party/nrf5_sdk/nrf5_sdk/external/nrf_cc310/include -I../third_party/nrf5_sdk/nrf5_sdk/external/utf_converter -I../third_party/littlefs/littlefs -I../platform/MCU/nRF52840/inc -I../dynalib/inc -Isrc -I./libraries -Ilib/ArduinoJson/src -I. -MD -MP -MF ../build/target/user/platform-12-msrc/arduinojsonexample.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DSPARK_PLATFORM_NET=ESP32 -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc  -DLOG_INCLUDE_SOURCE_INFO=1 -DPARTICLE_USER_MODULE -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0xD4000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=6 -DMODULE_FUNCTION=5 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=4,1,1512 -DMODULE_DEPENDENCY2=0,0,0 -D_GNU_SOURCE -DLOG_MODULE_CATEGORY="\"app\""  -fno-exceptions -fno-rtti -fcheck-new -std=gnu++14 -c -o ../build/target/user/platform-12-msrc/arduinojsonexample.o src/arduinojsonexample.cpp
In file included from lib/ArduinoJson/src/ArduinoJson/Polyfills/static_array.hpp:11:0,
                 from lib/ArduinoJson/src/ArduinoJson/Numbers/FloatTraits.hpp:14,
                 from lib/ArduinoJson/src/ArduinoJson/Numbers/convertNumber.hpp:18,
                 from lib/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:9,
                 from lib/ArduinoJson/src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from lib/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from lib/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from lib/ArduinoJson/src/ArduinoJson.hpp:17,
                 from lib/ArduinoJson/src/ArduinoJson.h:9,
                 from arduinojsonexample.ino:2:
lib/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In function 'typename ArduinoJson6171_F1::enable_if<ArduinoJson6171_F1::is_pointer<T>::value, T>::type ArduinoJson6171_F1::pgm_read(const void*)':
lib/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp:14:44: error: there are no arguments to 'pgm_read_ptr' that depend on a template parameter, so a declaration of 'pgm_read_ptr' must be ava [... cut ...]
../build/module.mk:274: recipe for target '../build/target/user/platform-12-msrc/arduinojsonexample.o' failed
make[2]: *** [../build/target/user/platform-12-msrc/arduinojsonexample.o] Error 1
make[2]: Leaving directory '/firmware/user'
../../../build/recurse.mk:11: recipe for target 'user' failed
make[1]: *** [user] Error 2
make[1]: Leaving directory '/firmware/modules/argon/user-part'
../build/recurse.mk:11: recipe for target 'modules/argon/user-part' failed
make: *** [modules/argon/user-part] Error 2
```

closed 2020-11-11 by bblanchon

bblanchon, 2020-11-11:
Hi @chinesefellow,

Thank you very much for reporting this bug.

The problem came from the fact that Particle defines some of the `pgm_read_XXX` macros in [spark_wiring_arduino.h](https://github.com/particle-iot/device-os/blob/7b7d28bf76db44111271abac23d247f09ebb8540/wiring/inc/spark_wiring_arduino.h), like `pgm_read_byte` and `pgm_read_word`, but lacks `pgm_read_ptr`, `pgm_read_float`, and `pgm_read_dword`.
Particle does define these macros in [avr/pgm_space.h](https://github.com/particle-iot/device-os/blob/1b1805fb4afcb93b0e92fbecfd28c082d6310285/wiring/inc/avr/pgmspace.h), but I cannot include this header from ArduinoJson because [the Arduino Core API deprecated it](https://github.com/arduino/ArduinoCore-API/blob/cd4ba4ca9ebc3421c65d3b19abe60820b8c4cf25/api/deprecated-avr-comp/avr/pgmspace.h) so it may not be available on every platform.

To fix this issue, I added a check before enabling support for `PROGMEM` in ArduinoJson: it now verifies that all macros are defined.

Moreover, [I added Particle Argon to the Continuous Integration](https://github.com/bblanchon/ArduinoJson/commit/aa7cc5351cd9ea04a1b5dfedf3fc19c5c9a920ed) so this kind of problem never happens again.

Until the fix is published, you can simply define [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to `0`.

Best regards,
Benoit

chinesefellow, 2020-11-11:
Thank you for such a great library! My students and I use it every year.

Daniel Law
he / him / his
Teacher of Awesome Things
FIRST Robotics Competition Coach
Computer Science Department Chair
Lane Tech High School
drlaw@cps.edu

*One who makes no mistakes makes nothing. - **Giacomo Casanova*

bblanchon, 2020-11-14:
The fix was published in [version 6.17.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.2).

commit 16e51b83 (2020-11-11): Changed the default value of ARDUINOJSON_ENABLE_PROGMEM (fixes #1433)

It now checks that the pgm_read_XXX macros are defined before enabling PROGMEM.