bblanchon/ArduinoJson #1434: ESP32 crashes when sending data to Queue

rikki78, 2020-11-10:
Hi,
first of all thanks for the great library!

I have an ESP32 (DOIT devkit) and I would like to send data from one task to another. 
When I fill the struct with the hardcoded values, it works. Whenever I fill the struct with the library (the commented part) the ESP32 crashes. See the exception below. 
I'm not sure if this is caused by the library, but since it doesn't happen when hardcoding the values, maybe I'm forgetting something?

thanks for your help in advance!

```

boolean jsonToRf2(JsonObject& RF2data)
{
    rfData * rfToSend;
    Serial.printf("Adding data to rf2 queue\n");
    /* rfToSend->code = RF2data["address"];
    rfToSend->unit = RF2data["unit"] | -1;
    rfToSend->period = RF2data["period"];
    rfToSend->group  = RF2data["groupBit"];
    rfToSend->dimValue  = RF2data["dim"] | 5;
    rfToSend->repeats = RF2data["repeats"]; 
    rfToSend->isDimCommand = (RF2data["switchType"] == 2);
    rfToSend->type = 2; */

    rfToSend->code = 27;
    rfToSend->unit = 2;
    rfToSend->period = 273;
    rfToSend->group  = 0;
    rfToSend->dimValue  = 0;
    rfToSend->repeats = 3; 
    rfToSend->isDimCommand = 1;
    rfToSend->type = 2;
    if (rfQueue != NULL) 
    {
      int ret = xQueueSend( /* The handle of the queue. */
                rfQueue,
                /* The address of the xMessage variable.  sizeof( struct AMessage )
                bytes are copied from here into the queue. */
                ( void * ) &rfToSend,
                /* Block time of 0 says don't block if the queue is already full.
                Check the value returned by xQueueSend() to know if the message
                was sent to the queue successfully. */
                ( TickType_t ) 0 );

      Serial.printf("Adding data to rf2 bfr. Ret %d\n", ret);
    }
  else
    return false;
  return true;
}
``` 

```
Adding data to rf2 queue
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400dcbb1  PS      : 0x00060030  A0      : 0x800d9695  A1      : 0x3ffde100
A2      : 0x3ffde1cc  A3      : 0x00000001  A4      : 0x3ffc483c  A5      : 0x3ffde2ac  
A6      : 0x00000001  A7      : 0x40404040  A8      : 0x800dcbaf  A9      : 0x3ffde0e0  
A10     : 0x0000001b  A11     : 0x3f4111b3  A12     : 0xc8a630df  A13     : 0x3ffc4c28
A14     : 0x3ffde09c  A15     : 0x3ffde09c  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000001  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffff7

Backtrace: 0x400dcbb1:0x3ffde100 0x400d9692:0x3ffde140 0x400f6a0a:0x3ffde4f0 0x4019bbbf:0x3ffde700 0x400d71a1:0x3ffde720 0x400f7085:0x3ffde760 0x400f709f:0x3ffde780 0x400dc6a7:0x3ffde7a0 0x40088d09:0x3ffde7c0
  #0  0x400dcbb1:0x3ffde100 in jsonToRf2(ArduinoJson6171_91::ObjectRef&) at src/rfFunctions.cpp:336
  #1  0x400d9692:0x3ffde140 in callback(char*, unsigned char*, unsigned int) at lib/MySensors_ID548/core/MyMessage.cpp:242
  #2  0x400f6a0a:0x3ffde4f0 in mqttCallback(char*, unsigned char*, unsigned int) at .pio/libdeps/esp32doit-devkit-v1/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:115
  #3  0x4019bbbf:0x3ffde700 in std::_Function_handler<void (char*, unsigned char*, unsigned int), void (*)(char*, unsigned char*, unsigned int)>::_M_invoke(std::_Any_data const&, char*&&, unsigned char*&&, unsigned int&&) at lib/MySensors_ID548/core/MyMessage.cpp:242
  #4  0x400d71a1:0x3ffde720 in std::function<void (char*, unsigned char*, unsigned int)>::operator()(char*, unsigned char*, unsigned int) const at lib/MySensors_ID548/core/MyMessage.cpp:242
      (inlined by) PubSubClient::loop() at lib/MySensors_ID548/drivers/PubSubClient/PubSubClient.cpp:421
  #5  0x400f7085:0x3ffde760 in mqttHandle() at .pio/libdeps/esp32doit-devkit-v1/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:115
  #6  0x400f709f:0x3ffde780 in iotLoop() at .pio/libdeps/esp32doit-devkit-v1/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:115
  #7  0x400dc6a7:0x3ffde7a0 in CommunicationTaskCode(void*) at src/main.cpp:472
  #8  0x40088d09:0x3ffde7c0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

```

bblanchon, 2020-11-11:
Hi @rikki78,

It it possible that the [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) refers to a destructed [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/)?

Also, I'm surprised that you pass the [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) by reference because [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) is already a "smart" reference. I recommend that you remove the `&` in the function declaration.
If you really want to squeeze every byte you can, use a [`JsonObjectConst`](https://arduinojson.org/v6/api/jsonobjectconst/) instead.

Best regards,
Benoit

rikki78, 2020-11-11:
Hi,
Yes that could be right. I will test this and let you know. 
The odd thing is, when not passing the structure to a queue, but handling it in a function, no crash occurs.
Anyway I will test it. Thanks.

bblanchon, 2020-11-12:
> The odd thing is, when not passing the structure to a queue, but handling it in a function, no crash occurs.

It sounds like the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) is scoped in the function, and therefore, is destructed as soon as the function returns.
Instead of pushing the [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) to the queue, you should push the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/).

rikki78, 2020-11-14:
Hi, 
I modified like you suggested and I hope I understood it right.
I'll add some more details here. I've stripped the program to show only the relevant lines.

```

void callback(char *topic, byte *payload, unsigned int length)
{
...
  DynamicJsonDocument doc(MQTT_MAX_PACKET_SIZE);
  DeserializationError error = deserializeJson(doc, payload);
  jsonToRf2(doc);
}

boolean jsonToRf2(const JsonDocument&  RF2data)
{
  rfData * rfToSend;
  Serial.printf("Adding data to rf2 queue\n");
  rfToSend->code = RF2data["address"];
  rfToSend->unit = RF2data["unit"] | -1;
  rfToSend->period = RF2data["period"];
  rfToSend->group  = RF2data["groupBit"];
  rfToSend->dimValue  = RF2data["dim"] | 5;
  rfToSend->repeats = RF2data["repeats"]; 
  rfToSend->isDimCommand = (RF2data["switchType"] == 2);
  rfToSend->type = 2;
  if (rfQueue != NULL) 
  {
    int ret = xQueueSend( /* The handle of the queue. */
              rfQueue,
              /* The address of the xMessage variable.  sizeof( struct AMessage )
              bytes are copied from here into the queue. */
              ( void * ) &rfToSend,
              /* Block time of 0 says don't block if the queue is already full.
              Check the value returned by xQueueSend() to know if the message
              was sent to the queue successfully. */
              ( TickType_t ) 0 );

    Serial.printf("Adding data to rf2 bfr. Ret %d\n", ret);
  }
}

```
This crashes the ESP32.

If I instead copy the contents to another struct, no crash occurs.
```

void callback(char *topic, byte *payload, unsigned int length)
{
...
  DynamicJsonDocument doc(MQTT_MAX_PACKET_SIZE);
  DeserializationError error = deserializeJson(doc, payload);
  jsonToRf2(doc);
}

rfData  rfDataBfr[RF_MAX_BFR];
boolean jsonToRf2(const JsonDocument&  RF2data)
{
  uint8_t i = rfiWr + 1;
  if (i >= RF_MAX_BFR)
    i = 0;
    
  Serial.printf("Adding data to rf2 queue\n");
  rfDataBfr[i].code = RF2data["address"];
  rfDataBfr[i].unit = RF2data["unit"] | -1;
  rfDataBfr[i].period = RF2data["period"];
  rfDataBfr[i].group  = RF2data["groupBit"];
  rfDataBfr[i].dimValue  = RF2data["dim"] | 5;
  rfDataBfr[i].repeats = RF2data["repeats"]; 
  rfDataBfr[i].isDimCommand = (RF2data["switchType"] == 2);
  rfDataBfr[i].type = 2;
  Serial.printf("Adding data to rf2 bfr\n");
}
```

bblanchon, 2020-11-14:
Hi @rikki78,

Look a the following lines:

```c++
rfData * rfToSend;  // <- uninitialized pointer
rfToSend->code = RF2data["address"];  //<- write to a random address
```

You need to initalize the pointer first; for instance, you could allocate the structure from the heap:

```c++
rfData * rfToSend = new rfData;
rfToSend->code = RF2data["address"];  //OK
```

If you do that, don't forget to `delete` the pointer after treating the message.

By the way, if `callback` is called from an interrupt handler, you should call `xQueueSendFromISR` instead.

If you need a refresher on pointers, stack, heap, new/delete, please check out "The Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/).

Regards,
Benoit

rikki78, 2020-11-14:
Thanks, that did it!
I think you're right, some refresher would be useful 😄

closed 2020-11-14 by rikki78