bblanchon/ArduinoJson #1435: passing doc from function to function to deserialize String or file

rinty35, 2020-11-12:
Hello.

Like issue #989 i want to pass on a function a SaticJsonDocument to realize a deserialize opération

> void mafonction(String fichier, const JsonDocument& obj ){
>   File file = SPIFFS.open(fichier, "r");
>   DeserializationError error = deserializeJson(obj, file);
> }

When i compile the code. i've this error 

no matching function for call to 'deserializeJson(const ArduinoJson6171_91::JsonDocument&, fs::File&)'

Do you have an idea ?

bblanchon, 2020-11-13:
Hi @rinty35,

You need to remove the `const` qualifier because it makes the `JsonDocument` read-only.

Best regards,
Benoit

rinty35, 2020-11-13:
I try this solution before posting issue.

In this case arduino compilator say : 
_In file included from ...\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:10:0,

                 from ...\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:35,

                 from ..\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from ...\Documents\Arduino\ESP8266_LED_DS18B20\ESP8266_LED_DS18B20.ino:11:

...\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp: In instantiation of 'size_t ArduinoJson6171_91::doSerialize(const TSource&, TWriter) [with TSerializer = ArduinoJson6171_91::JsonSerializer; TSource = char [10000]; TWriter = ArduinoJson6171_91::Writer<HardwareSerial, void>; size_t = unsigned int]':

...\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:22:49:   required from 'size_t ArduinoJson6171_91::serialize(const TSource&, TDestination&) [with TSerializer = ArduinoJson6171_91::JsonSerializer; TSource = char [10000]; TDestination = HardwareSerial; size_t = unsigned int]'

...\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:113:55:   required from 'size_t ArduinoJson6171_91::serializeJson(const TSource&, TDestination&) [with TSource = char [10000]; TDestination = HardwareSerial; size_t = unsigned int]'

...\Documents\Arduino\ESP8266_LED_DS18B20\ESP8266_LED_DS18B20.ino:237:35:   required from here

...\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:15:34: error: request for member 'accept' in 'source', which is of non-class type 'const char [10000]'

   return source.accept(serializer);

                                  ^

exit status 1

Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini_

bblanchon, 2020-11-14:
Hi @rinty35,

Since you didn't say what's on line 237 of ESP8266_LED_DS18B20.ino, I can only guess.
It seems that you called [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/) with a first argument of type `char [10000]` instead of [JsonDocument](https://arduinojson.org/v6/api/jsondocument/).

As a reminder, here is the right syntax:

```c++
char result[10000]:
serializeJson(doc, result);
```

Please check out the documentation on arduinojson.org, and the [Serialization Tutorial](https://arduinojson.org/v6/doc/serialization/) in particular.

Cordialement,
Benoit

rinty35, 2020-11-14:
Sorry i find. I have a var with two declaration one global on string and the second local  Char[10000]

It's ok now.

closed 2020-11-14 by rinty35