bblanchon/ArduinoJson #1436: Does it work with Arducam?  I get errors.

billbee, 2020-11-13:
I was trying to use this with a product I got from Amazon called arducam.  I could not get the cam to work so wanted to try it with the weather thing where your software worked perfectly the first time!!

If there is an easy fix I would be eternally grateful thank you.

In file included from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/static_array.hpp:11:0,
                 from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Numbers/FloatTraits.hpp:14,
                 from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Numbers/convertNumber.hpp:18,
                 from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:9,
                 from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/billb/BackupArduino/FloridaScrollerMAX7219_ArduCamNov2020JSON/FloridaScrollerMAX7219_ArduCamNov2020JSON.ino:55:
/Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6170_91::enable_if<ArduinoJson6170_91::is_pointer<T>::value, T>::type ArduinoJson6170_91::pgm_read(const void*) [with T = const __FlashStringHelper*; typename ArduinoJson6170_91::enable_if<ArduinoJson6170_91::is_pointer<T>::value, T>::type = const __FlashStringHelper*]':
/Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/DeserializationError.hpp:103:12:   required from here
/Users/billb/BackupArduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp:14:44: error: 'pgm_read_ptr' was not declared in this scope
   return reinterpret_cast<T>(pgm_read_ptr(p));

bblanchon, 2020-11-14:
Hi @billbee,

This looks very similar to #1433.
Can you check?

Best regards,
Benoit

billbee, 2020-11-14:
Merci beaucoup!Merci beaucoup!Merci beaucoup!Merci beaucoup!Merci beaucoup!Merci beaucoup!

Exactly right!  Great software!!!!

closed 2020-11-14 by billbee