bblanchon/ArduinoJson #1437: cannot enable STD_STRING

fabioinfo44, 2020-11-23:
```
lib\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp:18:26: error: 'const class String' has no member named 'write'
     return _dest->write(c);
                          ^

```Searching here for this issue I came across on PlatformIO, I found that I should do the following defines, but I tried both together and once at a time and the error persists.

```
#define ARDUINOJSON_ENABLE_STD_STREAM 1
#define ARDUINOJSON_ENABLE_STD_STRING 1
#include <ArduinoJson.h>
```

bblanchon, 2020-11-23:
Hi @fabioinfo44,

Please try [ARDUINOJSON_ENABLE_ARDUINO_STRING](https://arduinojson.org/v6/api/config/enable_arduino_string/).

Best regards,
Benoit

fabioinfo44, 2020-11-23:
@bblanchon
Thanks for your reply!
Done that but still not working...

bblanchon, 2020-11-24:
Hi @fabioinfo44,

`const class String`: the string is read-only, that's why you can't write to it.
Please try to remove the `const` keyword.

Best regards,

fabioinfo44, 2020-11-24:
@bblanchon 

Sorry, where should I find that declaration?

bblanchon, 2020-11-25:
I get no errors when I compile the code above.
My setup:

* PlatformIO 5.0.3
* ESP8266 2.6.2
* ArduinoJson 6.17.2

fabioinfo44, 2020-11-25:
@bblanchon 
Sorry, I read a wrong line!

`serializeJson(json, request->getParam("body", true)->value());`

Thank to your amazing and punctual support I found that it was necessary to cast the second parameter with (String &)
We finally did it, so I'll be able to go on with my project. Thank you again!

bblanchon, 2020-11-26:
Hi @bblanchon,

I'm glad you found this issue.
However, removing the constness of a returned value like that is a very bad practice.
Why don't you create a temporary `String` and then call `request->send()`?

Best regards,
Benoit

closed 2020-12-18 by bblanchon