bblanchon/ArduinoJson #1438: ESP8266, PubSubClient - Speed parity between publish methods.

porkyneal, 2020-11-23:
Before I begin, this is a fantastic library and the eBook has been a huge game changer for me with respect to optimizing the use of this library in my projects, so thanks 👍.

This issue, is not exactly a bug but more an observation which I would appreciate some help in understanding / resolving.

Essentially, if I use the method of sending a payload using the PubSubClient as described [here](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/) everything works as expected and the data finishes publishing rapidly.

However, the PubSubClient has a write method --> [DOCS](https://pubsubclient.knolleary.net/api#beginPublish), that means we don't have to create a char buffer before sending, helping preserve some precious memory. Below is my usage

```
   mqttClient.beginPublish(topic, measureJson(payload), false);
   serializeJson(payload, mqttClient);
   if (mqttClient.endPublish()) {
           DEBUG_APP("MQTT Send Success to %s\n", topic);
           return true;
    } else {
           DEBUG_APP("MQTT Send Failed: Len --> %d Topic:%s\n",measureJson(payload), topic);
           return false;
    }    
```

This works, the data gets sent as expected, however there is a huge time penalty, in the order of seconds for the same payload.
(I haven't had a chance to get the actual time penalty.)

Would you have any pointers as to what I can look into to use this method and gain the speed back, not creating temp buffers to me seems like a win, but not at the speed penalty.

Many Thanks,

bblanchon, 2020-11-24:
Hi,

Thanks for the kind words :-)

Please read [How to improve (de)serialization speed?](https://arduinojson.org/v6/how-to/improve-speed/)
(spoiler: you'll still need a buffer)

Best regards,
Benoit

porkyneal, 2020-11-24:
Brilliant, I have actually used your streamUtils to save a JSON file to EEPROM on the STM32, so will give it a whirl on this.

porkyneal, 2020-11-24:
That works great,  although the PubSubClient cannot be used as a Stream object, so I used the underlying WiFiClient instead.

Here is my example if others come along with the same observation.

```
        mqttClient.beginPublish(topic, measureJson(payload), false);
        WriteBufferingStream mqttWriteBuffer(wifiClient, 64);
        serializeJson(payload, mqttWriteBuffer);
        mqttWriteBuffer.flush();

        if (mqttClient.endPublish()) {
            DEBUG_APP("MQTT Send Success to %s\n", topic);
            return true;
        } else {
            DEBUG_APP("MQTT Send Failed: Len --> %d Topic:%s\n",measureJson(payload), topic);
            return false;
        }    
```

closed 2020-11-24 by porkyneal

bblanchon, 2020-11-24:
[PubSubClient inherits from `Print`](https://github.com/knolleary/pubsubclient/blob/2d228f2f862a95846c65a8518c79f48dfc8f188c/src/PubSubClient.h#L88), not `Stream`, so you must use a `BufferingPrint` instead of a `WriteBufferingStream`.

```c++
mqttClient.beginPublish(topic, measureJson(payload), false);
BufferingPrint bufferedMqttClient(mqttClient, 64);
serializeJson(payload, bufferedMqttClient);
bufferedMqttClient.flush();
mqttClient.endPublish();
```

Alternatively, you can pass the `WriteBufferingStream` to the constructor of `PubSubClient`.
Unfortunately, you'll still have to call `flush()` explicitly because `PubSubClient` doesn't call it in `endPublish()`; maybe one of us should open a PR...