bblanchon/ArduinoJson #1439: Json writing directly to a stream

rvdbreemen, 2020-11-23:
Is it possible to write directly to a stream with AduinoJson, like a JsonWriter class? Is it at all possible to directly write to a stream using your library and do serialization on the fly? 

Much like the JsonWriter classes out there:  e.g. https://github.com/timnew/ArduinoJsonWriter

I have seen many "direct writing" classes out there, but to me it is really not clear if your library always writes to memory first, and then serializes into a (buffered) stream. 

How does that work with your library, maybe I am just missing something. 

Thanks for explaining.

bblanchon, 2020-11-24:
Hi @rvdbreemen,

Yes. You can pass a `Stream` instance to [`serializeJson()`](https://arduinojson.org/v6/api/json/serializejson/) and [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/).
However, unlike `JsonWriter`, you must construct a [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) in memory.

Best regards,
Benoit

rvdbreemen, 2020-11-24:
@bblanchon thanks for that. That clears up my understanding of your library. Any change your library ever would support a truly streaming implementation of the serialization or is it just a totally different strategy?

bblanchon, 2020-11-25:
ArduinoJson will never provide an API similar to `JsonWriter`; as you say, this is a totally different strategy.
However, many users have successfully implemented [JSON streaming](https://en.wikipedia.org/wiki/JSON_streaming) with ArduinoJson: you simply need to call [`serializeJson()`](https://arduinojson.org/v6/api/json/serializejson/) or [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) repeatedly. It's not the same thing, but it's pretty close.

closed 2020-12-18 by bblanchon