bblanchon/ArduinoJson #1440: ESP32 wifi connection won't work with credentials taken from JSON SPIFFS

AdamMiltonBarker, 2020-11-25:
Hi guys, have an annoying issue, if I hardcode network and broker credentials I get an immediate connection, if I use SPIFFS and  JSON config file, the variables print fine and are correct but both connections fail with no error. Any suggestion, TIA,

```
#include <ArduinoJson.h>
#include <WiFi.h>
#include <PubSubClient.h>
#include <WiFiClientSecure.h> 
#include <SPIFFS.h> 

struct Config {
  const char* ssid;
  const char* password;
  const char* mqtt_server;
  int mqtt_port;
  String locationID;
  String zoneID;
  String deviceID;
  char deviceName[];
  char mqttUsername[];
  char mqttPassword[];
};

struct Config config; 
char willTopic[150]; 
char charBuf[150];  

WiFiClientSecure espClient;
PubSubClient client(espClient);

void loadConfiguration(Config &config)
{
  if (!SPIFFS.begin())
  {
    Serial.println("Failed to mount SPIFFS");
  }
  else
  {
    File file = SPIFFS.open("/config.json", FILE_READ);
    if (!file)
    {
      Serial.println("There was an error opening the file");
      return;
    }
    else
    {
      Serial.println("Loading configuration!");

      StaticJsonDocument<1048> doc;

      DeserializationError error = deserializeJson(doc, file);
      if (error)
      {
        Serial.println("error...");
        Serial.println(error.c_str());
      }
      else
      {
        config.ssid = doc["wifi"]["ssid"].as<const char*>();
        config.password = doc["wifi"]["key"].as<const char*>();
        config.mqtt_server = doc["iotJumpWay"]["host"].as<const char*>();
        config.mqtt_port = doc["iotJumpWay"]["port"];
        config.locationID = doc["iotJumpWay"]["lid"].as<String>();
        config.zoneID = doc["iotJumpWay"]["zid"].as<String>();
        config.deviceID = doc["iotJumpWay"]["did"].as<String>();
        strcpy(config.deviceName, doc["iotJumpWay"]["dn"]);
        strcpy(config.mqttUsername, doc["iotJumpWay"]["un"]);
        strcpy(config.mqttPassword, doc["iotJumpWay"]["pw"]);
        Serial.println("Configuration loaded!");
      }
      Serial.println("");
    }
    file.close();
  }
}

void setupWiFi() {
  
  delay(10);
  
  Serial.println();
  Serial.print("Connecting to WiFi network: ");
  Serial.println(config.ssid);
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(config.ssid, (const char*)config.password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  
  Serial.println("");
  Serial.println("WiFi connected succesfully!");
  Serial.print("Device IP address: ");
  Serial.println(WiFi.localIP());
  
}

void callback(char* topic, byte* payload, unsigned int length) {
  
  Serial.println("Unused");
  
}

void publishToDeviceStatus(const char* data){
  String statusTopic = config.locationID+"/Devices/"+config.zoneID+"/"+config.deviceID+"/Status";
  statusTopic.toCharArray(charBuf, 150);
  client.publish(charBuf, data);
}

void publishToDeviceZone(const char* data){
  String commandTopic = config.locationID+"/Devices/"+config.zoneID+"/"+config.deviceID+"/Zone";
  commandTopic.toCharArray(charBuf, 150);
  client.publish(charBuf, data);
}

void reconnect() {
  while (!client.connected()) {
    
    Serial.println("Attempting connection to HIAS iotJumpWay Broker...");
    
    String willTopicString = config.locationID+"/Devices/"+config.zoneID+"/"+config.deviceID+"/Status";
    willTopicString.toCharArray(willTopic, 150);
    
    if (client.connect(config.deviceName, config.mqttUsername, config.mqttPassword, willTopic, 0, 0, "OFFLINE")) {
      Serial.println("Connected to HIAS iotJumpWay Broker!");
      publishToDeviceStatus("ONLINE");
    } else {
      Serial.print("Failed to connect to HIAS iotJumpWay Broker, rc=");
      Serial.println(client.state());
      Serial.println("... trying again in 5 seconds");
      delay(5000);
    }
    
  }
}

void setup() {
  
  Serial.begin(9600);

  if(!SPIFFS.begin(true)){
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }
  loadConfiguration(config);
  setupWiFi();
  client.setServer(config.mqtt_server, config.mqtt_port);
  client.setCallback(callback);
  
}

void loop() {
  
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
  
  delay(1000);
}
```

AdamMiltonBarker, 2020-11-26:
Just to update I replaced the struct with individual variables and still the same, when printing them out right before connection they are fine, but unless I hardcode them wifi connection fails still. If I hardcode the wifi creds it connects instantly, then MQTT connection fails, if I hardcode the MQTT to creds it connects......

AdamMiltonBarker, 2020-11-26:
Solved been a while since I have worked with Ardunio :D For anyone else in the future, I changed the const char*s to String then used .c_str() and initialized the chars with a size.

closed 2020-11-26 by AdamMiltonBarker

bblanchon, 2020-11-26:
Hi @AdamMiltonBarker,

Congratulations on fixing this!
I'm sorry I didn't answer soon enough.

Best regards,
Benoit