bblanchon/ArduinoJson #1441: Not compiling: VariantAs.hpp:94:27

hergiz, 2020-11-25:
Extending example of **ArduinoJson/examples/JsonConfigFile/** and am constantly getting an error while trying to compile:

> C:\...\.platformio\lib\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp:94:27: error: invalid conversion from ArduinoJson6172_91::enable_if<true, const char*>::type {aka const char*}' to 'char*' [-fpermissive]

Code of main.cpp:

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ArduinoJson.h>

#include <Ticker.h>

#define DEBUGLEVEL 3
#include <DebugUtils.h>

/////////////////////
// Sve vezano za WiFi
/////////////////////
#include <LittleFS.h>

struct Config {
  char ssid[35];
  char pass[35];
  
  int ip1;
  int ip2;
  int ip3;
  int ip4;
  
  int sub1;
  int sub2;
  int sub3;
  int sub4;
  
  int gat1;
  int gat2;
  int gat3;
  int gat4;

  int dns1;
  int dns2;
  int dns3;
  int dns4;
};
Config config;
const char *filename = "/config.txt";

bool configLoaded = false;

int wifiReqConnect = 0 ; 
int wifiIsConnected = 0 ;
const long wifiReconnectDelay = 500;          
const long wifiReqConnectNum = 15; // number of intervals to wait for connection
unsigned long wifiPreviousMillis = 0;
WiFiEventHandler mConnectHandler;
WiFiEventHandler mDisConnectHandler;
WiFiEventHandler mGotIpHandler;
AsyncWebServer webServer (80); //port na kojem se sluša

void WiFionConnected(const WiFiEventStationModeConnected& event){
	DEBUGPRINTLN1 ( "Connected to AP." );
	wifiIsConnected = 1;    
}
 
void WiFionDisconnect(const WiFiEventStationModeDisconnected& event){
	DEBUGPRINTLN1 ( "WiFi On Disconnect." );
	wifiIsConnected = 0;
[... 296 lines cut ...]
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
Compiling .pio\build\esp12e\lib4eb\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pio\build\esp12e\lib4eb\ESP8266WiFi\ESP8266WiFiSTA-WPS.cpp.o
Compiling .pio\build\esp12e\lib4eb\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pio\build\esp12e\lib4eb\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pio\build\esp12e\lib4eb\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp12e\lib4eb\ESP8266WiFi\WiFiClientSecureAxTLS.cpp.o
In file included from C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:14:0,
                 from C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson.hpp:17,
                 from C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:5:
C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp: In instantiation of 'T ArduinoJson6172_91::variantAs(ArduinoJson6172_91::VariantData*, ArduinoJson6172_91::MemoryPool*) [with T = char*]':
C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:259:37:   required from 'ArduinoJson6172_91::VariantRef::operator T() const [with T = char*]'
C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:77:30:   required from 'ArduinoJson6172_91::MemberProxy<TParent, TStringRef>::operator T() const [with T = char*; TObject = ArduinoJson6172_91::JsonDocument&; TStringRef = const char*]'
Compiling .pio\build\esp12e\lib4eb\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
src\main.cpp:223:56:   required from here
C:\Users\Kles\.platformio\lib\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp:94:27: error: invalid conversion from 'ArduinoJson6172_91::enable_if<true, 
const char*>::type {aka const char*}' to 'char*' [-fpermissive]
   return variantAs<T>(data);
                           ^
*** [.pio\build\esp12e\src\main.cpp.o] Error 1
```

bblanchon, 2020-11-26:
Hi @hergiz,

Please see:

* [invalid conversion from 'const char*' to 'char*' [-fpermissive]](https://arduinojson.org/v6/error/invalid-conversion-from-const-char-to-char/)

The error is located at `src\main.cpp:223:56`.

Best regards,
Benoit

hergiz, 2020-11-26:
I have seen that post, but did not help me.

Finally I found solution:
This is not working
`String val = request->getParam("ssid", true)->value();`
`strlcpy(doc["ssid"], val.c_str(), sizeof(config.ssid));`

This is working
`doc["ssid"] = val.c_str();`

I am sorry for bothering you all.

closed 2020-11-26 by hergiz