bblanchon/ArduinoJson #1442: problem to compile samples for ESP8266 target !

phil31, 2020-11-27:
Hello all,

new with Arduino design, i'm trying to do some stufs with this librarie !
i try to compile the samples for Mega2560 or ESP32, that works (at least, compilation is right !)
but as soon as i change the target to ESP8266 generic (or ESP8285 generic which is my final target), with the same sample, i'v got some compilation errors that i can't solve by myself (newbie ...)

here it it the logs :

----------------------------------------------------------------------------------------------------------------------------------
In file included from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/static_array.hpp:11:0,
from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Numbers/FloatTraits.hpp:14,
from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Numbers/convertNumber.hpp:18,
from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantData.hpp:9,
from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/SlotFunctions.hpp:8,
from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:7,
from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:8,
from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:17,
from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
from D:\Arduino\WP-FP\WP-FP.ino:48:
D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6172_91::enable_if<ArduinoJson6172_91::is_pointer::value, T>::type ArduinoJson6172_91::pgm_read(const void*) [with T = const __FlashStringHelper*; typename ArduinoJson6172_91::enable_if<ArduinoJson6172_91::is_pointer::value, T>::type = const __FlashStringHelper*]':
D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/DeserializationError.hpp:103:12: required from here
D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/pgmspace_generic.hpp:14:45: error: 'const void*' is not a pointer-to-object type
return reinterpret_cast(pgm_read_ptr(p));
^
exit status 1
Error compiling for board Generic ESP8285 Module.
----------------------------------------------------------------------------------------------------------------------------------

may i get some tips or info to solve this please ?
Arduino 1.8.13 with ArduinoJson 6.17.2

thanks, regards

bblanchon, 2020-11-28:
Hi @phil31,

Thank you for reporting this problem.

> error: 'const void*' is not a pointer-to-object type

This is the same error we encountered in #1381.
It was caused by a bug in Arduino Core API and was fixed by arduino/ArduinoCore-API#118 in October.

What surprises me most is that you shouldn't be affected by this bug because the [ESP8266 core for Arduino](https://github.com/esp8266/Arduino) is not based on Arduino Core API. And also, we can see that [their definition is correct](https://github.com/esp8266/Arduino/blob/2843a5ace10eb696f19c9d754be7bbf6331046b5/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h#L110).

Are you using an alternative core for ESP8266?
If so, please open an issue and tell them about arduino/ArduinoCore-API#118.

Best regards,
Benoit

phil31, 2020-11-28:
Hi Benoit

thanks for your fast reply.
what do you mean by "alternative core for ESP8266" ? .. i dont know, i just install Arduino IDE  1.8.13 and all required from the IDE.
i can perfectly compile my program when i comment the "include <ArduinoJson.h> ".

i can perfectly compile the examples when the target are others than ESP8266 or ESO8285 (wich is my final target).
can you compile the samples for Generic ESP8266 ?

regards
Phil

bblanchon, 2020-11-28:
Hi Phil,

As far as I know, the Arduino IDE 1.8.13 doesn't support ESP8266 natively.
To program an ESP8266 with the Arduino IDE, you must install an additional toolkit.
As an example, see [the installation instructions for the ESP8266 core for Arduino](https://github.com/esp8266/Arduino#installing-with-boards-manager).

Of course, the examples compile with ESP8266; they are even part of the Continuous Integration.
Also, if they didn't, I would receive a massive amount of complaints. 😄

I suggest you upgrade or reinstall the [ESP8266 core for Arduino](https://github.com/esp8266/Arduino#installing-with-boards-manager).
If that's still not working, you can disable support for PROGMEM until we figure what's wrong (see [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/)).

Best regards,
Benoit

phil31, 2020-11-28:
no upgrade available, wii lty uninstall/reinstall IDE and core

regards

bblanchon, 2020-11-30:
Please let me know if it fixed the problem.

phil31, 2020-11-30:
well i just try to compile the samples for ESP targets, on another computer with a fresh install.
the compilation is working as expected !

i dont know how to remove properly the ESP core from arduino to try a new install ?

regards

bblanchon, 2020-12-01:
To do a fresh install, you may need to delete the folder `%LOCALAPPDATA%\Arduino15`, that's where the cores are installed.

closed 2020-12-01 by bblanchon