bblanchon/ArduinoJson #1444: Feature Request:  Serialize IPAddress

EmperorArthur, 2020-12-03:
Currently compilation fails if ArdunioJson attempts to serialize an IPAddress object.

On the surface, this makes sense because it is not something which can be directly converted to JSON.  However, Serial.print(..) explicitly works, so it catches users off gaurd.

Code to convert IPAddress to a string, which can be serialized.
```
out = String(address[0]) + "." + String(address[1]) + "." + String(address[2]) + "." + String(address[3])
```

bblanchon, 2020-12-04:
We can make this feature more general by including any `Printable` type.

EmperorArthur, 2020-12-04:
That would be great!

Hmm, Printable types implement `printTo(Print& p) const`.  The good news, is as your "PrintWriter" shows, the only things that have to be implemented in a Print subclass are `size_t write(uint8_t c)` and `size_t write(const uint8_t* s, size_t n)`.  The base class handles everything else.

So, it looks pretty easy to do as well.

closed 2021-03-27 by bblanchon

bblanchon, 2021-05-05:
Feature published in [ArduinoJson 6.18.0](https://arduinojson.org/news/2021/05/04/version-6-18-0/)

commit 347ac422 (2021-03-27): Added support for `Printable` (closes #1444)