bblanchon/ArduinoJson #1447: Problems with the missing copy Constructor since 6.15.0

Apfelwurm, 2020-12-08:
Hi @all,
i have problems when updating to ArduinoJson 6.15.0 or above in all my projects.

The error is:
`.pio/libdeps/d1_mini/ArduinoJson@6.15.0/src/ArduinoJson/Document/JsonDocument.hpp:324:3: error: 'ArduinoJson6150_0000010::JsonDocument::JsonDocument(const ArduinoJson6150_0000010::JsonDocument&)' is private
   JsonDocument(const JsonDocument&);`

Since i'm not that good with c++ i'm currently not able to fix it by myself.
While taking a look on the changes which were made between 6.14.1 and 6.15.0 i've noticed this commit and the linked issue: https://github.com/bblanchon/ArduinoJson/commit/9cb0ddb5e75de729a965476afbcb1d05c4ab9df8 

The affected part of my code is this function which merges two JsonDocuments:

`void WebserverHandler::mergesjson(JsonDocument dest, const JsonDocument src)
{
   for (auto kvp : src.as<JsonObject>()) {
     dest[kvp.key()] = kvp.value();
   }
}`

it is called by two objects which are initialized as DynamicJsonDocument doc(1024)

according to https://github.com/bblanchon/ArduinoJson/issues#issuecomment-594464272 i tried to add the & behind the JsonDocument type but that does not work.

Can someone give me advice or even better an explanation what i did wrong here and how to do it the right way?

Thanks!

bblanchon, 2020-12-09:
Hi @Apfelwurm,

You must change the signature, like so:

```c++
void WebserverHandler::mergesjson(JsonDocument& dest, const JsonDocument& src)
```

See also:

* [ArduinoJson 6.15: Copy constructor of JsonDocument](https://arduinojson.org/news/2020/03/22/version-6-15-0/#copy-constructor-of-jsondocument)
* [How to merge JSON objects?](https://arduinojson.org/v6/how-to/merge-json-objects/)

Need a refresher on C++ references? Check out "The Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/).

Best regards,
Benoit

Apfelwurm, 2020-12-09:
Hi @bblanchon, thank you for the fast reply :)

I had only changed the Datatypes in the cpp file, not in the hpp file. Since i changed it in both everything works again. My fail :D

Your "The Missing C++ Course" seems like something i should check out in the near future!

Alex

closed 2020-12-09 by Apfelwurm