bblanchon/ArduinoJson #1449: SerializeJSON from PSRAM | SPIRAM | External RAM to MQTT

GagahGK, 2020-12-14:
Hello I'm trying to utilize PSRAM that was included in my ESP32CAM and found the documentation about it using the `SpiRamAllocator`.
i also found the documentation to send a MQTT Message by storing the data in the buffer.

after reading both documentation i tried to create `jsonDocument `using the `SpiRamAllocator` and created temporary buffer for the mqtt
```
  //Create Json Document
  SpiRamJsonDocument MQTTPayload(20000); // 20 KB here
  
  //Put data here
  MQTTPayload["HardwareID"] = HardwareID;
  //Serialize it
  char * buffer = (char *) ps_malloc (20000 * sizeof (char));
  size_t n = serializeJson(MQTTPayload,buffer);
  client.publish("TubesIoT/Device/" + HardwareID, buffer, n);
```

and it gives me error like this

```
In file included from C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:7:0,
                 from C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:10,
                 from C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:35,
                 from C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\Gagah\Documents\Arduino\sketch_dec09a\sketch_dec09a.ino:8:
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp: In instantiation of 'size_t ArduinoJson6172_91::Writer<TDestination, Enable>::write(const uint8_t*, size_t) [with TDestination = char*; Enable = void; size_t = unsigned int; uint8_t = unsigned char]':
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/CountingDecorator.hpp:21:12:   required from 'void ArduinoJson6172_91::CountingDecorator<TWriter>::write(const uint8_t*, size_t) [with TWriter = ArduinoJson6172_91::Writer<char*, void>; uint8_t = unsigned char; size_t = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/TextFormatter.hpp:132:5:   required from 'void ArduinoJson6172_91::TextFormatter<TWriter>::writeRaw(const char*) [with TWriter = ArduinoJson6172_91::Writer<char*, void>]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:90:5:   required from 'size_t ArduinoJson6172_91::JsonSerializer<TWriter>::visitNull() [with TWriter = ArduinoJson6172_91::Writer<char*, void>; size_t = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantFunctions.hpp:18:30:   required from 'typename TVisitor::result_type ArduinoJson6172_91::variantAccept(const ArduinoJson6172_91::VariantData*, TVisitor&) [with TVisitor = ArduinoJson6172_91::JsonSerializer<ArduinoJson6172_91::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:350:25:   required from 'typename TVisitor::result_type ArduinoJson6172_91::VariantConstRef::accept(TVisitor&) const [with TVisitor = ArduinoJson6172_91::JsonSerializer<ArduinoJson6172_91::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:20:39:   required from 'typename TVisitor::result_type ArduinoJson6172_91::JsonDocument::accept(TVisitor&) const [with TVisitor = ArduinoJson6172_91::JsonSerializer<ArduinoJson6172_91::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:15:34:   required from 'size_t ArduinoJson6172_91::doSerialize(const TSource&, TWriter) [with TSerializer = ArduinoJson6172_91::JsonSerializer; TSource = ArduinoJson6172_91::BasicJsonDocument<SpiRamAllocator>; TWriter = ArduinoJson6172_91::Writer<char*, void>; size_t = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:22:34:   required from 'size_t ArduinoJson6172_91::serialize(const TSource&, TDestination&) [with TSerializer = ArduinoJson6172_91::JsonSerializer; TSource = ArduinoJson6172_91::BasicJsonDocument<SpiRamAllocator>; TDestination = char*; size_t = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:113:35:   required from 'size_t ArduinoJson6172_91::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson6172_91::BasicJsonDocument<SpiRamAllocator>; TDestination = char*; size_t = unsigned int]'
C:\Users\Gagah\Documents\Arduino\sketch_dec09a\sketch_dec09a.ino:94:46:   required from here
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp:22:29: error: request for member 'write' in '*((ArduinoJson6172_91::Writer<char*, void>*)this)->ArduinoJson6172_91::Writer<char*, void>::_dest', which is of non-class type 'char*'
     return _dest->write(s, n);
                             ^
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp: In instantiation of 'size_t ArduinoJson6172_91::Writer<TDestination, Enable>::write(uint8_t) [with TDestination = char*; Enable = void; size_t = unsigned int; uint8_t = unsigned char]':
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/CountingDecorator.hpp:17:12:   required from 'void ArduinoJson6172_91::CountingDecorator<TWriter>::write(uint8_t) [with TWriter = ArduinoJson6172_91::Writer<char*, void>; uint8_t = unsigned char]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/TextFormatter.hpp:149:5:   required from 'void ArduinoJson6172_91::TextFormatter<TWriter>::writeRaw(char) [with TWriter = ArduinoJson6172_91::Writer<char*, void>]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/TextFormatter.hpp:71:15:   required from 'void ArduinoJson6172_91::TextFormatter<TWriter>::writeFloat(T) [with T = float; TWriter = ArduinoJson6172_91::Writer<char*, void>]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:60:5:   required from 'size_t ArduinoJson6172_91::JsonSerializer<TWriter>::visitFloat(ArduinoJson6172_91::Float) [with TWriter = ArduinoJson6172_91::Writer<char*, void>; size_t = unsigned int; ArduinoJson6172_91::Float = float]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantData.hpp:43:51:   [ skipping 2 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:350:25:   required from 'typename TVisitor::result_type ArduinoJson6172_91::VariantConstRef::accept(TVisitor&) const [with TVisitor = ArduinoJson6172_91::JsonSerializer<ArduinoJson6172_91::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:20:39:   required from 'typename TVisitor::result_type ArduinoJson6172_91::JsonDocument::accept(TVisitor&) const [with TVisitor = ArduinoJson6172_91::JsonSerializer<ArduinoJson6172_91::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:15:34:   required from 'size_t ArduinoJson6172_91::doSerialize(const TSource&, TWriter) [with TSerializer = ArduinoJson6172_91::JsonSerializer; TSource = ArduinoJson6172_91::BasicJsonDocument<SpiRamAllocator>; TWriter = ArduinoJson6172_91::Writer<char*, void>; size_t = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:22:34:   required from 'size_t ArduinoJson6172_91::serialize(const TSource&, TDestination&) [with TSerializer = ArduinoJson6172_91::JsonSerializer; TSource = ArduinoJson6172_91::BasicJsonDocument<SpiRamAllocator>; TDestination = char*; size_t = unsigned int]'
C:\Users\Gagah\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:113:35:   required from 'size_t ArduinoJson6172_91::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson6172_91::BasicJsonDocument<SpiRamAllocator>; TDestination = char*; size_t = unsigned int]'
C:\Users\Gagah\Documents\Arduino\sketch_dec09a\sketch_dec09a.ino:94:46:   required from here
[... cut ...] _91::Writer<char*, void>*)this)->ArduinoJson6172_91::Writer<char*, void>::_dest', which is of non-class type 'char*'
     return _dest->write(c);
                          ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Gagah\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board ESP32 Wrover Module.

```

am i doing something wrong here ? how do i fix this ?
thankyou

bblanchon, 2020-12-14:
Hi @GagahGK,

When you pass a pointer to [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/), you must also pass the size as the third argument:

```c++
size_t n = serializeJson(MQTTPayload, buffer, 20000);
```

This is a security feature to prevent buffer overrun.

Best regards,
Benoit

GagahGK, 2020-12-15:
i see i didn't read the documentation for the  [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/) properly, sorry 
i works flawlessly now thankyou 😀.

I'm going to leave my code here in case someone also having the same problem
```
  //Create Json Document
  SpiRamJsonDocument MQTTPayload (20000); // 20 KB here
  //Put data here
  MQTTPayload["HardwareID"] = HardwareID.c_str();
  MQTTPayload["image"] = imageBuffer.c_str();
  //Serialize it
  serializeJson(MQTTPayload, Serial);
  char * buffer = (char *) ps_malloc (20000 * sizeof (char));
  size_t n = serializeJson(MQTTPayload,buffer,20000);
  client.publish("TubesIoT/Device/" + HardwareID, buffer, n);
```

closed 2020-12-15 by GagahGK

bblanchon, 2021-02-03:
See also [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/) (#1491)