bblanchon/ArduinoJson #1450: InvalidInput

Talkwondo, 2020-12-17:
Hi,
I am sending a request to a mock server and get the following json by **TEXT**:
{
  "userId": 1,
  "id": 1,
  "title": "delectus aut autem",
  "completed": false
}

When I try to parse the json I get an error of InvalidInput.

I am using version 6. I see in the documents to use http1.0 but I can not config it because I use sim module 800L with AT commands.

Please if someone has an idea to solve the problem.

Thanks,

bblanchon, 2020-12-18:
Hi @Talkwondo,

[InvalidInput](https://wandbox.org/permlink/n9TSMbnSMcvsB6Nc) means that the input is invalid, it's up to you to figure out what is wrong with this input.
In case you think the ArduinoJson parser is at fault, check out [this online demo](https://wandbox.org/permlink/n9TSMbnSMcvsB6Nc) and you'll see that everything works as expected.

If I were you, I'd start by printing the content of the input; if it comes from a stream, [you can use ReadLoggingStream](https://arduinojson.org/v6/how-to/debug-data-coming-from-a-stream/).

Best regards,
Benoit

Talkwondo, 2020-12-18:
Hello,
Thanks for you advice.
When I try to log the json with your recommendation i get this error:
no matching function for call to 'StreamUtils::ReadLoggingStream::ReadLoggingStream(String&, HardwareSerial&)'
     ReadLoggingStream loggingStream(data, Serial);

Thanks

bblanchon, 2020-12-19:
Please apologize if my message was unclear, `ReadLoggingStream` is only for `Stream`.
If, instead you use a `String`, you must call `Serial.print(data)`.

Talkwondo, 2020-12-19:
Thanks, so I have already done that before you told me and this is the log:
{
  "userId": 1,
  "id": 1,
  "title": "delectus aut autem",
  "completed": false
}

I notice that if you encapsulate this "JSON" string with R() so it works. so maybe the solution is to modify the string?

bblanchon, 2020-12-20:
`R()` is not mandatory, see here:
https://wandbox.org/permlink/kO0Kte0st4kcwb2v

Talkwondo, 2020-12-20:
Maybe the problem is the blank spaces?

bblanchon, 2020-12-21:
The problem is in the input.
I know you showed the JSON document twice, but I assure you that what you pass to `deserializeJson()` is different.
Why don't you share a __small__ piece of code?

Talkwondo, 2020-12-21:
This is th code:
`#include <SoftwareSerial.h>
#include <ArduinoJson.h>
#include <StreamUtils.h>
#include <Arduino_JSON.h>

SoftwareSerial sim(13, 12); 

void setup()
{
  Serial.begin(9600);

  sim.begin(9600);
  

  Serial.println("Begin..."); 
  delay(10000);

  sim.println("AT");
  updateSerial();
  sim.println("AT+CSQ");
  updateSerial();
  sim.println("AT+CCID");
  updateSerial();
  sim.println("AT+CREG?");
  updateSerial();
  sim.println("AT+CGREG?");
  updateSerial();
  
  //SEND SMS
//    sim.println("AT+CMGF=1");
//    updateSerial();
//    sim.println("AT+CMGS=\"+972524420370\"");
//    updateSerial();
//    sim.print("SHEEN | MA KORE?");
//    updateSerial();
//    sim.write(26);
  delay(500);
  sim.println("AT+CGATT=1");
  updateSerial(); 
  sim.println("AT+SAPBR=3,1,CONTYPE,GPRS");
  updateSerial();
  sim.println("AT+SAPBR=3,1,\"APN\",\"" + String("we") + "\"");
  updateSerial();
  
  sim.println("AT+SAPBR=1,1");
  updateSerial();
  
  sim.println("AT+SAPBR=2,1");
  updateSerial();
  
  sim.println("AT+HTTPINIT");
  updateSerial();
  sim.println("AT+HTTPPARA=\"CID\",1");
  updateSerial();

  sim.println("AT+HTTPPARA=URL,http://jsonplaceholder.typicode.com/todos/1");
  updateSerial();
  
  sim.println("AT+HTTPPARA=CONTENT,application/json");
  updateSerial();

  sim.println(F("AT+IPR=9600"));
  updateSerial();

  sim.println("AT+HTTPACTION=0");
  updateSerial();
  delay(5000);
  
  sim.println("AT+HTTPREAD");
  readSerial();

  sim.println("AT+HTTPTERM");
  updateSerial();
}

void readSerial() {
  while(sim.available()) 
  {
    String data = sim.readString();    
    Serial.print(data);
    DynamicJsonDocument doc(256);
    DeserializationError error = deserializeJson(doc, data);

    if (error) {
      Serial.print("ERROR PARSING! ");
      Serial.println(error.c_str());
    }
    const char* userID = doc["title"].as<char*>();
    int id = doc["userId"].as<int>();
    Serial.println(userID);
    Serial.println(id);
    delay(1000);
  }
  
}

void loop()
{
}

void updateSerial()
{
  delay(500);
  while (Serial.available()) 
  {
    sim.write(Serial.read());
  }
  while(sim.available()) 
  {
    Serial.write(sim.read());
  }
}`

Talkwondo, 2020-12-22:
SOLVED!
The issue it the response from the server was big string and the json was just a little bit of it. 
Slice the string to the JSON!

Thanks @bblanchon

closed 2020-12-22 by Talkwondo