bblanchon/ArduinoJson #1451: Design pattern for JSON command and response on limited memory devices

geekfactory, 2020-12-21:
I have a small device based on ATMEGA328 which I´m working on. I want to implement bidirectional JSON communication. For example, to set the date on the device RTC I´d send the following JSON to the serial port:

`{
  "command":"settimedate",
  "args":{
    "date":"19/12/20",
    "time":"16:06:00",
    "dst":true
  }
}`

Then, after the device sets the time on the RTC, I´d like to get a response like this:

`{
    "res":"ok",
    "date":"19/12/20",
    "time":"16:06:00",
    "dst":true
  }`

I want a "dispatcher" function to deserialize the outer object and then a specific function to process the args object. The design pattern would be like this (please excuse me if you find errors), this is pseudo-code grade:

`
/**
   Decode logic for command strings
*/
int jsoncmd_decode(const char * cmd);

/**
   Function implementing the desired functionality for each command
*/
void command_setdatetime(StaticJsonDocument& doc)
{
  // process data in json
  // parse strings and save the required data
  uint32_t currenttime = parseTime(doc["args"]["time"]);
  uint32_t parsedate = parseDate(doc["args"]["date"]);
  hardware_rtc_set(currenttime, parsedate);

  // now I consumed input data, the buffer can be used to format response
  // how can I do this? is the following method correct?

  doc.clear();

  // send command confirmation or errors if any
  doc["res"] = "ok";
  doc["date"] = hardware_rtc_date();
  doc["time"] = hardware_rtc_time();
  doc["dst"] = hardware_rtc_dst();

  // return after response is written to serial port
  serializeJson(doc, Serial);
}

/**
   The input of this function is the JSON data from the serial port
   which is buffered until it is complete and then passed to this function
*/
void jsoncmd_dispatcher(char * input)
{
  StaticJsonDocument<128> doc;
  deserializeJson(doc, input);

  switch (jsoncmd_decode(doc["cmd"]))
  {
    case E_CMD_SETDATETIME:
      // perform the requested operation to set date and time
      command_setdatetime(doc);
      break;

    case E_CMD_SETIO:
      // perform the requested operation on IO pines
      command_setio(doc);
      break;

    case E_CMD_DEVID:
      // send device identification back to PC
      command_devid(doc)
      break
  }

  // prepare the response and send it here using a new JsonDocument?

}
// prepare the response and send it here using a new JsonDocument?
`

The code assumes the following:

- Data will be buffered from serial until a complete JSON object is received, then the string will be passed to deserializeJSON().
- deserializeJSON() would operate on a writable buffer, so no need to copy strings

I have the following questions and thoughts:

- What would be the best pattern to use the least possible amount of memory?
- I understand that object reuse is not recommended, but given the small memory on the target device, what would be the recommended approach to avoid declaring two JsonDocument objects?
- In case any copying is done when serializing, Is it possible to use the "input" buffer (it is writable) to do the serialization process?
- I know I could manually format the JSON output on the input buffer and send it back to the serial, but this is not pretty.

Any comments on this use case would be greatly appreciated.

Thank you!

bblanchon, 2020-12-21:
Hi @geekfactory,

1. I think this pattern is pretty good: reusing the same `JsonDocument` for the response avoids having two in memory simultaneously.
2. Object reuse is not recommended when there are better alternatives, but I think it's good usage in your case.
3. No. ArduinoJson cannot use the input buffer during the serialization process.
   If I were you, I would not use a buffer to store the input; instead, I would increase the capacity of the `JsonDocument` and let `deserializeJson()` do the copy. It would use less memory overall and solve your serialization issue.

My advice about Atmega328:

* never use the heap (in particular, avoid `String` and `DynamicJsonDocument`)
* the RAM is indeed tiny, but in my experience, the Flash size is the real living factor, so keep your code simple
* move **long** strings to Flash (but remember that `F()` doesn't deduplicate)

I hope this helps.

Best regards,
Benoit

geekfactory, 2020-12-21:
Thank you for a fast response.

I´ll implement this and let you know how it goes. I understand the benefits of letting the object handle the strings (using a read-only object as input, Stream in this case), however, one major issue when letting it manage the strings is that it will block program execution at some point while waiting for the JSON to arrive. This is not intended, and as far as I understand, the library cannot return until it receives all the JSON.

The main program should be able to continue execution on the main loop, as there are other tasks being performed aside from parsing the JSON input from the serial port.

Please let me know if I´m missing something.

bblanchon, 2020-12-22:
You're right `deserializeJson()` reads the `Stream` until it gets the complete document or encounters an error.
The library doesn't currently offer an alternative solution.

closed 2021-01-08 by bblanchon