bblanchon/ArduinoJson #1452: need help with serialising uint16_t array

caster1, 2020-12-26:
hi im using esp8266 ir library to retreieve raw data into an array but im not sure how to convert that into a json array to send that to my server, ive googled two days continuously but no much results so confused with json5 ad json6 codes
`StaticJsonDocument<1024> doc;
  doc["result"] = resultToHumanReadableBasic(&results);
  String description = IRAcUtils::resultAcToString(&results);
  doc["Ac results"] = description;
  uint16_t *raw_array = resultToRawArray(&results);
  JsonArray array = doc.to<JsonArray>();
  copyArray(raw_array, array);

  yield();
 char buffer[1024];
size_t n = serializeJson(doc, buffer);
client.publish("/test/light1", buffer,n);`

the error that im getting is
`no matching function for call to 'copyArray(uint16_t*&, ArduinoJson6141_0000010::VariantTo<ArduinoJson6141_0000010::ArrayRef>::type)'
`

Yash1903, 2020-12-26:
Hello, caster
you share only String value or you want send whole file on Server ??

caster1, 2020-12-27:
only string values of that array so i can deserialise the same on another module .

bblanchon, 2020-12-27:
Hi @caster1,

> no matching function for call to 'copyArray(uint16_t*&, ArduinoJson6141_0000010::VariantTo<ArduinoJson6141_0000010::ArrayRef>::type)'

The compiler tells you that there is no function `copyArray(uint16_t*, JsonArray)`.
Indeed, a quick look at [the documentation for `copyArray()`](https://arduinojson.org/v6/api/misc/copyarray/) reveals that this function expects three parameters when called with a pointer:

```c++
bool copyArray(uint16_t* src, size_t len, JsonArray dst);
```

As you can see, you must specify the length of the array.

Best regards,
Benoit

(@Yash1903, I had to delete an unrelated message you wrote in another thread. Please [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new/choose) if you need assistance, but don't hijack existing conversations).

caster1, 2020-12-27:
thank you my problem solved

Yash1903, 2020-12-27:
Hi @bblanchon,
i wrote code for Temperature and humidity detection but result or output of this code presser and humidity value not show like this....
and i'm using library ArduinoJson 6.17.2
****OUTPUT
`{"Node0":{"Time":13518241,"Temperature":[48.75608,2.30203],"Humidity":[]}}
 {"Node2":{"Time":13518241,"Temperature":[25.4,25.4],"Temperaturea":[]}}
`

`#include "DHT.h"
#include <ArduinoJson.h>
#define DHTPIN 2     // what pin we're connected to
#define DHTTYPE DHT22   // DHT 22  (AM2302)

int maxHum = 60;
int maxTemp = 40;

DHT dht(DHTPIN, DHTTYPE);

void setup() {
  // put your setup code here, to run once:
// pinMode(fan, OUTPUT);
  Serial.begin(9600); 
  dht.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
  // put your main code here, to run repeatedly:
  float h = dht.readHumidity();
  // Read temperature as Celsius
  float t = dht.readTemperature();
////////////////////////////////////////////////////////////////////////////////
DynamicJsonDocument no2(96);

JsonObject Node2 = no2.createNestedObject("Node2");
Node2["Time"] = 13518241;

JsonArray Node2_Temperature = Node2.createNestedArray("Temperature");
Node2_Temperature.add(t);
Node2_Temperature.add(t);

JsonArray Node2_Temperaturea = Node2.createNestedArray("Temperaturea");
Node2_Temperaturea.add(h);
Node2_Temperaturea.add(h);

serializeJson(no2, Serial);
Serial.println();
delay(2000);
////////////////////////////////////////////////
DynamicJsonDocument doc(96);

JsonObject Node0 = doc.createNestedObject("Node0");
Node0["Time"] = 13518241;

JsonArray Node0_Temperature = Node0.createNestedArray("Temperature");
Node0_Temperature.add(48.75608);
Node0_Temperature.add(2.30203);

JsonArray Node0_Humidity = Node0.createNestedArray("Humidity");
Node0_Humidity.add(2243511);
Node0_Humidity.add(13165115);

JsonArray Node0_Presser = Node0.createNestedArray("Presser");
Node0_Presser.add(2122);
Node0_Presser.add(2231);

serializeJson(doc, Serial);
Serial.println();
}`

bblanchon, 2020-12-28:
Hi @caster1,

It looks like you wanted to write:

```c++
StaticJsonDocument<1024> doc;
uint16_t *raw_array = resultToRawArray(&results); // memory leak?
uint16_t length = getCorrectedRawLength(&results);
copyArray(raw_array, length, doc);
char buffer[1024];
size_t n = serializeJson(doc, buffer);
client.publish("/test/light1", buffer,n);
```

Also, remember that the maximum message size for PubSubClient is 256 by default.

Best regards,
Benoit

(@Yash1903, I deleted the message you wrote because it didn't belong here; please [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new))

caster1, 2020-12-28:
hi when i replaced my ode with yours i.e
`StaticJsonDocument<1024> doc;
uint16_t *raw_array = resultToRawArray(&results); // memory leak?
uint16_t length = getCorrectedRawLength(&results);
copyArray(raw_array, length, doc);
char buffer[1024];
size_t n = serializeJson(doc, buffer);
client.publish("/test/light1", buffer,n);`

and I'm getting this error 
no matching function for call to 'copyArray(uint16_t*&, uint16_t&, ArduinoJson6141_0000010::StaticJsonDocument<256u>&)'

bblanchon, 2020-12-29:
Please upgrade ArduinoJson to 6.15.2 or above.

caster1, 2021-01-03:
```
StaticJsonDocument<256> doc;
  deserializeJson(doc, payload, length);

  uint16_t *rarray = doc;
  uint16_t len = length;
 irsend.sendRaw(rarray, len ,38);
```
im trying to desrialise like this but its throwing me this error. 
cannot convert 'ArduinoJson6172_91::StaticJsonDocument<256u>' to 'uint16_t* {aka short unsigned int*}' in initialization

uint16_t *rarray = doc;

bblanchon, 2021-01-06:
To extract a C array from a `JsonDocument`, you must use [`copyArray()`](https://arduinojson.org/v6/api/misc/copyarray/):

```c++
uint16_t rarray[64];
uin16_t len = copyArray(doc, rarray);
irsend.sendRaw(rarray, len, 38);
```

closed 2021-01-13 by bblanchon