bblanchon/ArduinoJson #1453: Make GET and PUT Request to the same Server

v11, 2020-12-28:
Hello Benoit

You already helped me with a few questions this summer. Thank you very much for this. I made progress, but now i am stuck. I am not sure if the question is really related to ArduinoJson. But maybe you can help design my process better.

Environment:
Controller: ESP 32
Backend: Google Firebase

There are 2 steps in the function:
1. GET Data from Google via REST API
2. POST or PUT Data back to the Firebase Realtime Database
After that, the Controller is going to Deep-Sleep and wakes up after certain time.

Problem:
The GET Requests always working. The POST or PUT Request sometimes don't work.
See: http://vision11.ch/share/temp/missing-data.png

Question:
Maybe you have an idea, about how can i write the code better to make sure every request get's fired in every circumstances. I also would like to separate the code in two functions. I tried but then, even more requests were missing.

Do you have an idea on this problem?

Lukas

Code:
```
void getDisplayConfigSaveStatsToFirebase(char *host)
{

  // Wifi Client for secure connection (HTTPS)
  WiFiClientSecure client;

  // Add Certificate
  client.setCACert(cert_rest);

  // Connect to Host
  Serial.println("Connect to server via port 443");

  // HTTPS Connection: Check if Secure Connection is successfull
  if (!client.connect(host, 443))
  {
    // Connection Failed
    Serial.println("Connection failed!");

    char err_buf[100];
    if (client.lastError(err_buf, 100) < 0)
    {
      // Show the last Error-Message when something goes wrong
      Serial.println(err_buf);
      return;
    }
    else
    {
      Serial.println("Connection error");
      return;
    }
  }
  else
  {
  }

  // Get Display Configuration Data from Firebase

  // HTTPS Connection: Request URL with
  String url = "/devices/" + macAddress + "/config.json?print=pretty";
  Serial.print("requesting URL: ");
  Serial.println(host + url);

  client.print(String("GET ") + url + " HTTP/1.0\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: ESP32\r\n" +
               "Connection: keep-alive\r\n\r\n");

  Serial.println("request sent");

  // HTTPS Connection: Client is connected. Read Headers
  // HTTPS Connection: Header is the first response we get from the Server
  // HTTPS Connection: Example: HTTP/1.1 200 OK: Standard response for successful HTTP requests.
  while (client.connected() || client.available())
  {
    if (client.available())
    {
      String line = client.readStringUntil('\n');

      if (line == "\r")
      {
        Serial.println("headers received");
        break;
      }
    }
  }

  // HTTPS Connection: Once server sends all requested data it will disconnect, then once all received data are read, program will exit the while loop.

  // JSON Parse: ArduinoJson
  // JSON Parse: Allocate the JSON document
  // JSON Parse: Use https://arduinojson.org/v6/assistant/ to calculate the capacity
  DynamicJsonDocument doc(96);

  // JSON Parse: Parse JSON object and receive possible errors
  DeserializationError error = deserializeJson(doc, client);

  // JSON Parse: Print error and stop the program
  if (error)
  {
    Serial.print(("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }
  else
  {
    Serial.println("serialization successful");
  }

  // JSON Parse: Get the data
  displayType = doc["displayType"];
  deepSleepInSeconds = doc["deepSleepInSeconds"];

  // JSON Parse: Print the result
  Serial.print("Display Type received: ");
  Serial.println(displayType);
  Serial.print("Deep Sleep in Seconds received: ");
  Serial.println(deepSleepInSeconds);

  // End GET Request

  // ++++++++++++++++++++++++++++++++++++++

  // Start Post Request

  // JSON Parse: ArduinoJson
  // JSON Parse: Allocate the JSON document
  // JSON Parse: Use https://arduinojson.org/v6/assistant/ to calculate the capacity
  DynamicJsonDocument doc_put(96);

  // Prepare the data
  doc_put["bootCount"] = String(bootCount);
  doc_put["lastUpdate"] = String(bootCount) + ": " + currentTime;

  // HTTPS Connection: Request URL with
  String url_put = "/devices/" + macAddress + "/stats.json";
  Serial.print("requesting URL: ");
  Serial.println(host + url_put);

  client.println("POST " + url_put + " HTTP/1.1");
  // client.println("PUT " + url_put + " HTTP/1.1");

  // Send the HTTP headers
  client.print("Host: ");
  client.println(host);
  client.println("Connection: keep-alive");
  client.print("Content-Length: ");
  client.println(measureJson(doc_put));
  client.println("Content-Type: application/json");

  // Terminate headers with a blank line
  client.println();

  // Send JSON document in body
  serializeJson(doc_put, client);

  Serial.println("PUT request sent");

  client.stop();
}

```

v11, 2020-12-28:
I believe it is only the POST who is making trouble. That is the request i am sending:
```
{
    "bootCount" : "3",
    "lastUpdate" : "Monday, 28. December 2020 13:46:36"
}
```

v11, 2020-12-29:
OK, it is working now.

I have now to separated functions.
- GET Request
- PUT Request

I had a lot of trouble with my PUT Request but now it is working reliable.

What did I changed:
- In the HTTP header i changed "connection: keep-alive" to "connection: close"
- I am waiting for the response of the server

Can someone lighten my up? Why does it only work reliable when i wait for this response?

```
// Wait for response
  while (client.connected())
  {
    String line = client.readStringUntil('\n');
    if (line == "\r")
    {
      Serial.println("headers received");
      break;
    }
  }

  // Read response from Server
  while (client.available())
  {
    char c = client.read();
    Serial.write(c);
  }
  Serial.println("");
```

Here is the complete Code of the PUT Request:
```
void setStats(char *host)
{

  // Note: For sucessful PUT / POST Request it is necessary to listen for the response of the Server

  Serial.println("");
  Serial.println("set stats");

  // Wifi Client for secure connection (HTTPS)
  WiFiClientSecure client;

  // Add Certificate to client
  client.setCACert(cert_rest);

  // Connect to Host
  Serial.println("connect to server via port 443");

  // HTTPS Connection: Check if Secure Connection is successfull
  if (!client.connect(host, 443))
  {
    // Connection Failed
    Serial.println("connection failed!");

    char err_buf[100];
    if (client.lastError(err_buf, 100) < 0)
    {
      // Show the last Error-Message when something goes wrong
      Serial.println(err_buf);
      return;
    }
    else
    {
      Serial.println("connection error");
      return;
    }
  }

  // JSON Parse: ArduinoJson
  // JSON Parse: Allocate the JSON document
  // JSON Parse: Use https://arduinojson.org/v6/assistant/ to calculate the capacity
  DynamicJsonDocument doc(96);

  // Asign the data
  doc["bootCount"] = String(bootCount);
  doc["lastUpdate"] = currentTime;

  // HTTPS Connection: Request URL with
  String url = "/devices/" + macAddress + "/stats.json";
  Serial.print("requesting URL: ");
  Serial.println(host + url);

  // Assign URL
  client.println("PUT " + url + " HTTP/1.1");

  // Send the HTTP headers
  client.print("Host: ");
  client.println(host);
  client.println("Connection: close");
  client.print("Content-Length: ");
  client.println(measureJson(doc));
  client.println("Content-Type: application/json");
  client.println();

  // Send JSON document in body
  serializeJson(doc, client);

  // Wait for response
  while (client.connected())
  {
    String line = client.readStringUntil('\n');
    if (line == "\r")
    {
      Serial.println("headers received");
      break;
    }
  }

  // Read response from Server
  while (client.available())
  {
    char c = client.read();
    Serial.write(c);
  }
  Serial.println("");

  client.stop();
}
```

bblanchon, 2020-12-29:
Hi @v11,

It seems that the server cancels the requests if the client prematurely closes the connection.
This explains why you need to wait for the complete response before calling `client.stop()`.

Anyway, this question is not about ArduinoJson; I think you'll find more helpful answers on StackOverflow.

Best regards,
Benoit

v11, 2020-12-29:
Thank you Benoit.

closed 2020-12-29 by v11