bblanchon/ArduinoJson #1454: Searching for a example: Changing a value in a large JSON file.

BlueAndi, 2020-12-28:
Is there an example how to change a value in a large and complex JSON file, using ReadBufferingStream/WriteBufferingStream? Loading the whole file is not an option. Additional the new written file should look pretty (indention).

My current idea is to have a loop with deserializeJson and serializeJsonPretty. If the right object is found, it will be replaced. Additional the ',' must be handled, right? I guess to keep the indention right, I can not use serializeJsonPretty directly and write my own PrettyJsonSerializer.

bblanchon, 2020-12-29:
Hi,

If I understand correctly, you have a huge file JSON that cannot fit in RAM, and you want to replace a single value.

Unfortunately, this is currently not possible with ArduinoJson, and I don't think it ever will because I don't see how it could work.
Indeed, replacing even a single value often requires rewriting the whole file; for example, if you replace a string with a longer one, all following characters must be shifted as well.

As a workaround, I recommend splitting the huge JSON file into several smaller files that fit easily in RAM.

Best regards,
Benoit

BlueAndi, 2020-12-29:
Hi Benoit,
thanks for your fast answer. I feared something like that.
If the filter mechanism could be used similiar for changing values during parsing, this may be possible ... just an idea.
Ok, but thinking now about to break the file down.

Best wishes
Andi

closed 2021-01-08 by bblanchon