bblanchon/ArduinoJson #1455: when i append file in v5 its append  correctly but in v6 of json library give out put in number or bytes for appending file

Yash1903, 2020-12-29:
below my code..

`#include <ArduinoJson.h>
#include "FS.h"
#include "SPIFFS.h"
String y;
unsigned long stime=65000;
unsigned long Itime=0;
//const char* filename = "/data.json";

void setup() {
//  pinMode(fan, OUTPUT);
  Serial.begin(9600);
//////////////
if(!SPIFFS.begin()){
        Serial.println("An Error has occurred while mounting SPIFFS");
        return;
   }
  
    //--------- Write to file ---------------
    File fileToWrite = SPIFFS.open("/test.json", "w");
  
    if(!fileToWrite){
        Serial.println("There was an error opening the file for writing");
        return;
    }
  
    if(fileToWrite.println(y)){
        Serial.println("File was written");
    } else {
        Serial.println("File write failed");
    }
  
    fileToWrite.close();
}

void loop() {
   
/////////////////////////////////////////////////////////
//unsigned long rtime=millis();
//Serial.print("Ruing");
//Serial.println(rtime);

DynamicJsonDocument doc(192);

JsonObject Node0 = doc.createNestedObject("Node0");
Node0["time"] = 13518241;

JsonArray Node0_Phase1 = Node0.createNestedArray("Phase1");
Node0_Phase1.add(48.75608);
Node0_Phase1.add(2.30203);

JsonArray Node0_Phase2 = Node0.createNestedArray("Phase2");
Node0_Phase2.add(48.75608);
Node0_Phase2.add(2.30203);

JsonArray Node0_Phase3 = Node0.createNestedArray("Phase3");
Node0_Phase3.add(48.75608);
Node0_Phase3.add(2.30203);

JsonObject Strock = doc.createNestedObject("Strock");
Strock["time"] = 13518241;

JsonArray Strock_valueofsensor = Strock.createNestedArray("valueofsensor");
Strock_valueofsensor.add(48.75608);
Strock_valueofsensor.add(2.30203);

String y;
y=serializeJson(doc, Serial);
Serial.println(y);

//File fileToAppend = SPIFFS.open("/test.json", "a");  
//    if(fileToAppend.println(y))
//    fileToAppend.close();
///////////////////
delay(1000);
//const size_t capacity = 4*JSON_ARRAY_SIZE(2) + 2*JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(4);
DynamicJsonDocument doc1(192);

JsonObject Node1 = doc1.createNestedObject("Node1");
Node1["time"] = 13518241;

JsonArray Node1_axis1 = Node1.createNestedArray("axis1");
Node1_axis1.add(48.75608);
Node1_axis1.add(2.30203);

JsonArray Node1_axis2 = Node1.createNestedArray("axis2");
Node1_axis2.add(48.75608);
Node1_axis2.add(2.30203);

JsonArray Node1_axis3 = Node1.createNestedArray("axis3");
Node1_axis3.add(48.75608);
Node1_axis3.add(2.30203);

JsonObject Strock1 = doc1.createNestedObject("Strock1");
Strock["time"] = 13518241;

JsonArray Strock1_valueofsensor = Strock1.createNestedArray("valueofsensor");
Strock1_valueofsensor.add(48.75608);
Strock1_valueofsensor.add(2.30203);

String ya;
ya=serializeJson(doc1, Serial);
Serial.println(ya);

//File fileToAppend2 = SPIFFS.open("/test.json", "a");  
//    if(fileToAppend2.println(ya))
//    fileToAppend2.close();
////////////////////////////////////
//const size_t capacity = 3*JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(4);
DynamicJsonDocument doc2(192);

JsonObject Node2 = doc2.createNestedObject("Node2");
Node2["Time"] = 13518241;

JsonArray Node2_Temperature = Node2.createNestedArray("Temperature");
Node2_Temperature.add(48.75608);
Node2_Temperature.add(2.30203);

JsonArray Node2_Humidity = Node2.createNestedArray("Humidity");
Node2_Humidity.add(2243511);
Node2_Humidity.add(13165115);

JsonArray Node2_Presser = Node2.createNestedArray("Presser");
Node2_Presser.add(2122);
Node2_Presser.add(2231);

String yas;
yas=serializeJson(doc2, Serial);
Serial.println(yas);

// File fileToAppend3 = SPIFFS.open("/test.json", "a");
//    if(fileToAppend3.println(yas))
//    fileToAppend3.close();
//unsigned long etime=millis();
//Serial.print("etime");
//Serial.println(etime);
//////////////////////////////////////

 //---------- Read from file --------------
    File fileToRead = SPIFFS.open("/test.json", "r");
  
    if(!fileToRead){
        Serial.println("Failed to open file for reading");
        return;
    }
  
    Serial.println("File Content:");
  
    while(fileToRead.available()){
  
        Serial.write(fileToRead.read());
    }
  
    fileToRead.close();
}`

bblanchon, 2020-12-30:
Hi @Yash1903,

Please explain what the problem is.

Best regards,
Benoit

Yash1903, 2020-12-30:
thank you for Replay@bblanchon, i have 2 question 
1) in above Program run ArduinoJson 5.13.5 run completely but i modify to Arduinojson V6 then show me bytes instead of sensor data.
2) can i send whole data.json file to Mqtt ? if yes then can you share example of it.
or How to write,append and read .json file in Arduinojson v6

bblanchon, 2021-01-01:
Hi,

1. I'm sorry, I don't understand what you mean by "show me bytes instead of sensor data". Can you provide an example?
2. Please see [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/)
3. Please see [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/)

Best regards,
Benoit

Yash1903, 2021-01-09:
hello @bblanchon I try example of JsonConfigFile.ino..but Shaw me Error..
`Failed to initialize SD library` how i can resolve this Error ??

bblanchon, 2021-01-12:
This is not a problem related to ArduinoJson.
Please compare the working examples with the non-working one to see how they differ.

closed 2021-01-28 by bblanchon