bblanchon/ArduinoJson #1456: Stop receiving data with AltSoftSerial when deserializeJson has an error

Sindarius, 2020-12-29:
Platform : Arduino Uno
Compiler : Visual Studio Code / PlatformIO

I have a problem where sometimes I get garbage data into my buffer and when I attempt to deserialize the data in the buffer I get an InvalidInput as expected. The issue is that after I get a deserialization error I no longer receive input from AltSoftSerial. 

I commented out the deserializeJson call and even after getting bad data I continue to receive data properly. 

I need ideas to get the AltSoftSerial start reading again after getting bad data.

bblanchon, 2020-12-30:
Hi,

ArduinoJson stops reading as soon as it finds an error in the stream.
So, before calling `deserializeJson()` again, you must flush the remaining bytes.
Please see [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/) for an example.

BTW, you can try to reduce the error ratio by using Hamming codes.
StreamUtils provides a [HammingStream](https://github.com/bblanchon/ArduinoStreamUtils#how-to-use-error-correction-codes-ecc) decorator for that.

Best regards,
Benoit

Sindarius, 2020-12-30:
Hi Benoit

Thank you for the additional information. I will take a look into those.  Every now and then I get a random bad read from a 3d printer board to the Arduino (likely wire length playing a role). I am reading the byte that come in into a byte array and that is what I am sending to the deserializeJson method. I find it odd that when it fails to read that buffer it would cause problems with the serial connection as I thought I had them disconnected by using the byte array. 

It could just be how I am using it. I am fairly new to programming on the Arduino. 

https://github.com/Sindarius/Sindarius_Duet_Neopixels/blob/master/src/DuetData.cpp

closed 2021-01-05 by Sindarius