bblanchon/ArduinoJson #1457: JsonArray.add(JsonArray) doesn't make an exact clone.

timkoers, 2020-12-30:
As per documentation (https://arduinojson.org/v6/api/jsonarray/add/#arguments), when calling add on an JsonArray it is expected to clone the JsonArray argument into the JsonArray that add is being called on.

Example;

```C++
JsonArray setPoints = output.createNestedArray("set_points");
setPoints.add(root["set_points"].as<JsonArray>()); // This will copy the values into the new array
```

`root["set_points"]` looks like:

```json
"set_points" : [
		{
			"temperature": 120,
			"duration": 90
		},
		{
			"temperature": 165,
			"duration": 30
		},
		{
			"temperature": 182,
			"duration": 120
		},
		{
			"temperature": 210,
			"duration": 30
		},
		{
			"temperature": 182,
			"duration": 30
		}
	]
```

Result:

```json
	"set_points": [
		[
			{
				"temperature": 120,
				"duration": 90
			},
			{
				"temperature": 165,
				"duration": 30
			},
			{
				"temperature": 182,
				"duration": 120
			},
			{
				"temperature": 210,
				"duration": 30
			},
			{
				"temperature": 182,
				"duration": 30
			}
		]
	]

```
Please note the extra pair of array indicators above.

I am expecting this behaviour:

```json

	"set_points": [{
			"temperature": 120,
			"duration": 90
		},
		{
			"temperature": 165,
			"duration": 30
		},
		{
			"temperature": 182,
			"duration": 120
		},
		{
			"temperature": 210,
			"duration": 30
		},
		{
			"temperature": 182,
			"duration": 30
		}
	]

```

timkoers, 2020-12-30:
When using `set` instead of `add`, the library provides me with the expected result.

bblanchon, 2020-12-30:
Hi @timkoers,

That's the expected behavior:

* [JsonArray::add()](https://arduinojson.org/v6/api/jsonarray/add/) appends to the array
* [JsonArray::set()](https://arduinojson.org/v6/api/jsonarray/set/) replaces the values

BTW, you can simply your code like so:

```c++
output["set_points"] = root["set_points"];
```

Best regards,
Benoit

timkoers, 2020-12-31:
Ah ok, I think I misread the documentation. My fault.

Have a nice year's ending and all the best for 2021

closed 2020-12-31 by timkoers

bblanchon, 2021-01-01:
Thanks! Best wishes to you as well.
Good luck with your project.