bblanchon/ArduinoJson #1458: PSRAM decision at runtime?

mike-s123, 2020-12-31:
Arduino IDE w/Arduino-ESP32 core (latest). Target ESP32 with or without PSRAM.

I'd like to make the decision whether to use PSRAM at runtime, depending on whether it's available (so the same code can take advantage of available hardware). Something like:

    bool jsonLarge = false;       // this is later used to determine how much can be returned via json
    psRAMavail = psramInit();
    struct SpiRamAllocator {
      void* allocate(size_t size) {
        return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
      }
      void deallocate(void* pointer) {
        heap_caps_free(pointer);
      }
    };
    if (psRAMavail) {
      using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;
      SpiRamJsonDocument jsonOut(0x40000);
      jsonLarge = true;
     } else {
      DynamicJsonDocument jsonOut(0x10000);
    }

But, of course, that doesn't work because it declares jsonOut twice, even if one isn't going to be used at runtime. I can pick one way or the other using hardcoded #defines, but that required different compiles to support different hardware. I could use different names, but then the rest of the code would be loaded with conditionals based on which one is available.

I'm not an expert or professional coder, and I did what research I could, and am now here asking if there's a way to do this.

bblanchon, 2021-01-01:
Hi @mike-s123,

You'll can use `new`/`delete`, like so:

```c++
JsonDocument* doc;
if (psRAMavail)
  doc = new SpiRamJsonDocument(0x40000)
else
  doc = new DynamicJsonDocument(0x10000);
deserializeJson(*doc, input);
int value = (*doc)["value"];
// ...
delete doc;
```

Or, better, you can use `unique_ptr`, like so:

```c++
std::unique_ptr<JsonDocument> doc;
if (psRAMavail)
  doc.reset(new SpiRamJsonDocument(0x40000));
else
  doc.reset(new DynamicJsonDocument(0x10000));
deserializeJson(*doc, input);
int value = (*doc)["value"];
// ...
```

BTW, if the size is constant, you can save a call to `malloc()` by replacing `new DynamicJsonDocument(0x10000)` with `new StaticJsonDocument<0x10000>()`.

Best regards,
Benoit

bblanchon, 2021-01-02:
A simpler solution would be to let the allocator class decide which heap to use:

```c++
struct SpiRamAllocator {
  void* allocate(size_t size) {
   if (psRAMavail)
     return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
   else
     return malloc(size);
  }
  void deallocate(void* pointer) {
    if (psRAMavail)
      heap_caps_free(pointer);
    else
      return free(size);
  }
};
```

Another option is to ask the ESP32 to [map the external RAM into the data address space](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#integrate-ram-into-the-esp32-memory-map). This way you can use `DynamicJsonDocument` as usual.

mike-s123, 2021-01-04:
Wow, thanks for the fast response with multiple suggestions.  Changing the allocator worked well.

    struct SpiRamAllocator {
      void* allocate(size_t size) {
        if (psRAMavail) return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
        else return (malloc(size));
      }
      void deallocate(void* pointer) {
        if (psRAMavail) heap_caps_free(pointer);
        else free(pointer);
      }
    };
    using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;
    uint32_t jsonOutSize;
    (psRAMavail) ? jsonOutSize = 0x40000 : jsonOutSize = 0x10000;
    SpiRamJsonDocument jsonOut(jsonOutSize);

closed 2021-01-04 by mike-s123

bblanchon, 2021-01-04:
You're welcome.
Thank you for using ArduinoJson, @mike-s123, and good luck with your project!