bblanchon/ArduinoJson #1461: i need help in arduinojson

Mouka007, 2021-01-02:
hello
I have problem when convert DynamicJsonBuffer in my code from arduinojson 5 to arduinojson 6 , i need to resolve it without install libraries to version 5.13 , i know there are way by replace DynamicJsonDocument , i am try more than one time but the code is not work , i need help to replace DynamicJsonDocument in my code instead of DynamicJsonBuffer.

this my code i need to convert it to DynamicJsonDocument , i will show the the two part in the code which i need to solve it , if anyone need to show the full code i will upload it

>  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        std::unique_ptr<char[]> buf(new char[size]);
        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;         
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);   //print config data to serial on startup
        if (json.success()) {
          Serial.println("\nparsed json");
          
          strcpy(MyApiKey, json["MyApiKey"]);
          strcpy(deviceID_1, json["deviceID_1"]);
          strcpy(deviceID_2, json["deviceID_2"]);
          strcpy(deviceID_3, json["deviceID_3"]);

          
        } else {
          Serial.println("failed to load json config");
          initialConfig = true;
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  

and this the second part

> if (shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
   
    json["MyApiKey"] = MyApiKey
    json["deviceID_1"] = deviceID_1;
    json["deviceID_2"] = deviceID_2;
    json["deviceID_3"] = deviceID_3;

    
    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {Serial.println("failed to open config file for writing");}    
    json.prettyPrintTo(Serial);
    json.printTo(configFile);
    configFile.close();
    Serial.println("config saved");
    shouldSaveConfig = false;
    initialConfig = false; 
    ticker.detach();
    digitalWrite(status_led, HIGH);
    WiFi.mode(WIFI_STA);
    ESP.restart();
    delay(5000); 
  }

bblanchon, 2021-01-02:
Hi,

Please see the [migration guide](https://arduinojson.org/v6/doc/upgrade/).

Best regards,
Benoit

Mouka007, 2021-01-02:
thanks for your replay , 

i maked it but , i have problem in the function ,it dont gave any errro , but dont make same function , it dont save the given input .

this my code after replacement 

>  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        std::unique_ptr<char[]> buf(new char[size]);
        configFile.readBytes(buf.get(), size);
        DynamicJsonDocument doc(200);   
      // char json[] =
       //"{\"MyApiKey\":\"MyApiKey\",\"deviceID_1\":\"deviceID_1\",\"deviceID_2\":\"deviceID_2\",\"deviceID_3\":\"deviceID_3\"}";      
        DeserializationError error = deserializeJson(doc , buf.get());
        serializeJson(doc, Serial);   //print config data to serial on startup
        if (error) {
          Serial.print("deserializeJson() failed with code");
          Serial.println(error.c_str());
          strcpy(MyApiKey, doc["MyApiKey"]);//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
          strcpy(deviceID_1, doc["deviceID_1"]);
          strcpy(deviceID_2, doc["deviceID_2"]);
          strcpy(deviceID_3, doc["deviceID_3"]);

          
        } else {
          Serial.println("failed to load json config");
          initialConfig = true;
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  

and this second  part 

> if (shouldSaveConfig) {
    Serial.println("saving config");
   
    DynamicJsonDocument doc(200);
    
    doc["MyApiKey"] = MyApiKey;/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
    doc["deviceID_1"] = deviceID_1;
    doc["deviceID_2"] = deviceID_2;
    doc["deviceID_3"] = deviceID_3;

    
    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {Serial.println("failed to open config file for writing");}    
    serializeJsonPretty(doc, Serial);
    serializeJson(doc, configFile);
    configFile.close();
    Serial.println("config saved");
    shouldSaveConfig = false;
    initialConfig = false; 
    ticker.detach();
    digitalWrite(status_led, HIGH);
    WiFi.mode(WIFI_STA);
    ESP.restart();
    delay(5000); 
  }

>

bblanchon, 2021-01-13:
Hi @Mouka007,

Did you solve your issue?
If not can you summarize what works and what doesn't?

Best regards,
Benoit

closed 2021-01-28 by bblanchon