bblanchon/ArduinoJson #1462: Deserialization an array in ArduinoJson 6 with Esp32

JuzKid, 2021-01-02:
Hi, I am new in ArduinoJson so I had some of the question here. I already include the library ArduinoJson 6.17.2 and can compile the code but there was no result/ output in the Serial Monitor. I also tried the sample code (deserialization), the output is same (I get nothing on the Serial Monitor). So I was here to seek a solution. Below is my code:

`#include <ArduinoJson.h>

int token;
String ID;
int card_number; 
String full_name;

void setup() {
    Serial.begin(115200);

    char json[] = "{\"token\":123456,\"result\":[{\"ID\":1001,\"card_number\":1234567890,\"full_name\":\"Bradley\"},{\"ID\":1002,\"card_number\":789789789,\"full_name\":\"John\"}]}";
    Serial.println(json);       
          
    //Allocate getlist file
    const int capacity = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(2)+ 2*JSON_OBJECT_SIZE(3);
    StaticJsonDocument<capacity> doc;
    DeserializationError err = deserializeJson(doc, json);

    //Test if parsing succeeds
    if (err) {
       Serial.print("deserializeJson() failed with code");
       Serial.println(err.c_str());
    return;
    }
          
    int token = doc["token"];
    JsonArray arr = doc.as<JsonArray>();
    for (JsonObject repo : arr) {
    int ID        = repo["result"]["ID"];
    int card_number  = repo["result"]["card_number"];
    char full_name        = repo["result"]["full_name"];
    }
      
    Serial.println("Start Printing:");
    Serial.println(token);
    Serial.println(ID);
    Serial.println(card_number);
    Serial.println(full_name);
}

void loop() {

}`

Here is the sample code which provided but there was no result/ output in Serial Monitor:

`#include <ArduinoJson.h>

void setup() {
    Serial.begin(115200);
   
    const char* json = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.75608,2.302038]}";

    DynamicJsonDocument doc(1024);
    DeserializationError err = deserializeJson(doc, json);

    //Test if parsing succeeds
    if (err) {
        Serial.print("deserializeJson() failed with code");
        Serial.println(err.c_str());
        return;
    }

    const char* sensor = doc["sensor"]; // "gps"
    long time = doc["time"]; // 1351824120
    float data_0 = doc["data"][0]; // 48.75608
    float data_1 = doc["data"][1]; // 2.302038

    Serial.println(sensor);
    Serial.println(time);
    Serial.println(data_0);
    Serial.println(data_1);

}

void loop() {
  // put your main code here, to run repeatedly:

}`

[image]

Thank you.

bblanchon, 2021-01-04:
Hi @JuzKid,

Does a simple `Serial.println("hello world!");` works?
If not, then it's unrelated to ArduinoJson.

Can you try to restart the board by pressing the RESET button?
Sometimes, when we open the Serial Monitor a few seconds after the program started, we miss the output.

Best regards,
Benoit

JuzKid, 2021-01-05:
Hi, 

Yes, simple `Serial.println("hello, world!")` works (Every time it works). 
I already tried restart the board and open Serial Monitor before the code is uploaded but there was no result in the Serial Monitor. 

I tried uploaded about 10 times but just 1 time it worked. (the ArduinoJson sample code).  There was no error and nothing else, just it do not have output in the Serial Monitor. if it was my code, the failure rate (no result on Serial Monitor) is more than ArduinoJson sample code.

I also have some question about the array (my code).  When the 1 times works (the array part), Serial Monitor print out 0 from my code. The token had no problem is 123456, but the array part show there was zero/ null  on the Serial Monitor so I was wonder is my coding part for array had wrong ?

Thank you. Regards.

bblanchon, 2021-01-07:
Hi @JuzKid,

I cannot help you with the information I currently have.
Can you rephrase your question?
In particular, please detail what works and what doesn't.
If you have a second question, please open a new issue or wait until the first one is resolved.
Thank you for your comprehension.

Best regards,
Benoit

closed 2021-01-20 by bblanchon