bblanchon/ArduinoJson #1464: Json::Array iterator skips one entry?

ykolerov, 2021-01-02:
Hi, I have the followinf function:

```
void TMsg::pull () {

  serializeJson(*msgStorage, Serial); // msgStorage is a TMsg member with a three nodes in it

  JsonArray arr = msgStorage->as<JsonArray>();
  serializeJson(arr, Serial);

  for (JsonArray::iterator it = arr.begin(); it != arr.end(); ++it) {
    serializeJson(*it, Serial);
...
  }
}

```

I get the following output (splitted into lines manually):

// this is *msgStorage - it contains three nodes
[
 {
  "event_type":"Initialization",
  "event_data":{"status":"started"},
  "cntrl_id":"DC","ts_created":0,
  "ts_sent":0
 },
 {
  "event_type":"Initialization",
  "event_data":{"status":"success"},
   "cntrl_id":"DC",
   "ts_created":1609635230,
   "ts_sent":1609635230
 },
 {
  "event_type":"Info",
  "event_data":{"software_ver":"CT-Jan  3 2021-00:50:43"},
  "cntrl_id":"DC0645C40A24",
  "ts_created":1609635231,
  "ts_sent":1609635231
 }
]

// this is arr, it's equal to *msgStorage and also contains three nodes
[
 {
  "event_type":"Initialization",
  "event_data":{"status":"started"},
  "cntrl_id":"DC",
  "ts_created":0,
  "ts_sent":0
 },
 {
  "event_type":"Initialization",
  "event_data":{"status":"success"},
  "cntrl_id":"DC",
  "ts_created":1609635230,
  "ts_sent":1609635230},
 {
  "event_type":"Info",
  "event_data":{"software_ver":"CT-Jan  3 2021-00:50:43"},
  "cntrl_id":"DC",
  "ts_created":1609635231,
  "ts_sent":1609635231
 }
]

// and this is what I get inside the cycle (second node won't appear here)
{
 "event_type":"Initialization",
 "event_data":{"status":"started"},
  "cntrl_id":"DC",
  "ts_created":0,
  "ts_sent":0
}

{
 "event_type":"Info",
 "event_data":{"software_ver":"CT-Jan  3 2021-00:50:43"},
 "cntrl_id":"DC",
 "ts_created":1609635231,
 "ts_sent":1609635231
}

Why the second node would be skipped?

thanks,

bblanchon, 2021-01-03:
Hi @ykolerov,

I cannot reproduce this issue.
See this online demo: https://wandbox.org/permlink/LvgWt7Ll4LAzrCGF

Best regards,
Benoit

ykolerov, 2021-01-05:
Hi,
I realized that I misuse JsonDocument::remove() in my code, and that (predictably) caused this weird behaviour. Sorry for borthering.

closed 2021-01-05 by ykolerov

bblanchon, 2021-01-06:
Well, that's good news :-)
Thank you for using ArduinoJson, @ykolerov.
Good luck with your project!