bblanchon/ArduinoJson #1465: GET requests on server accepting HTTPS & "Let's Encrypt" certificate

jmlocci, 2021-01-04:
Good'day all

I want to implement a GET request on a site that only accepts HTTPS with a "Let's Encrypt" certificate. I try to mimic the REDDIT example found in the book, advising the use of WiFiClientSecure.

The site is "api.meteo-concept.com", it requires the implementation of SSL/TLS on port 443.

The piece of code:

` const char* URL = "api.meteo-concept.com";
  const int PORT = 443;

  // Connect to host
  WiFiClientSecure client;
  // BearSSL::WiFiClientSecure client;  // alternative to the above
  client.setInsecure(); // bypass the "Let's Encrypt" certificate to allow succesful connection to server
  client.setTimeout(10000);
  client.connect(URL, PORT));

  // Send HTTP request
  client.println(F("GET /api/forecast/daily?insee=65292 HTTP/1.0"));
  client.println(F("Authorization: MY_TOKEN"));                                  // Happy to provide MY_TOKEN through PM
  client.println(F("Host: api.meteo-concept.com"));
  client.println(F("Connection: close"));

  // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  Serial.print("Server response code : ");
  Serial.println(status);
`
If the "client.setInsecure()" line is inactive, the connection fails
If the "client.setInsecure()" line is active, the connection succeds but the server response is empty

I suspect the GET request is unsuccessful due to incorrect credentials, probably because the "Let's Encrypt" certification authotity is not delt with in Arduino.

Can anyone confirm or advise?

Many thanks

bblanchon, 2021-01-05:
Hi @jmlocci,

I don't know how the certificate verification works in this library, but you probably need to feed it with root certificates (take a look at `setTrustAnchors()`).
Unfortunately, that means updating them every year, and I don't know if we can securely automate the process.

Regarding the empty response, do you at least get a status code?
Empty responses are often due to a redirect, so I wouldn't be surprised to see a 302 or 301.

Best regards,
Benoit

jmlocci, 2021-01-05:
Hello Benoit.

I will look into the `setTrustAnchors()`

No, no error code, that's weird. In fact the response to the GET is just "1" (Serial.println(client)). The  word "status" is therefore not found. The "invalid response" code in the serial output below is generated by the "Skip HTTP headers" instruction.

Serial output in setDebugOutput mode:

dhcp client start ...
ip: 192.168.1.99,mask: 255.255.255.0, gw: 192.168.1.1
Connected to server : api.meteo-concept.com, Port 443
Server response code :
1
pm open,type :2 0
Invalid response

Well, never mind... Thanks!!!

martinius96, 2021-01-05:
@jmlocci 
Hello there, can you please specify also what board are you using?
Based on code I think it is ESP32? I tried it with ESP32 and edited it how I am using HTTPS connections.
I have put DST ROOT CA X3 certificate in .pem format to code, connection is working and I get response from webserver about autorization. (THAT ROOT CA IS ISSUING CERTIFICATES FOR LET'S ENCRYPT).
[image]
As you can see at screenshot below from UART monitor, return code HTTP 401 - 401 Unauthorized. So I think, if you put there your token it will work and you will be able to see what server will print you behind Authorization :).
[image]

**Program implementation on ESP32** (if not good readable, visit for code: https://pastebin.com/341FAXm7)
`/*|-------------------------------------------------|*/
/*|Project: Authorization ESP32 with DST ROOT CA X3 |*/
/*|Autor: Martin Chlebovec (martinius96 - Github)   |*/
/*|-------------------------------------------------|*/

#include <WiFi.h>
#include <WiFiClientSecure.h>
const char* ssid = "MY_WIFI_NAME";
const char* password = "MY_WIFI_PASSWORD";
const char* host = "api.meteo-concept.com";
WiFiClientSecure client;

// DST ROOT CA X3 in .pem format
const static char* test_root_ca PROGMEM = \
    "-----BEGIN CERTIFICATE-----\n" \
    "MIIDSjCCAjKgAwIBAgIQRK+wgNajJ7qJMDmGLvhAazANBgkqhkiG9w0BAQUFADA/\n" \
    "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
    "DkRTVCBSb290IENBIFgzMB4XDTAwMDkzMDIxMTIxOVoXDTIxMDkzMDE0MDExNVow\n" \
    "PzEkMCIGA1UEChMbRGlnaXRhbCBTaWduYXR1cmUgVHJ1c3QgQ28uMRcwFQYDVQQD\n" \
    "Ew5EU1QgUm9vdCBDQSBYMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB\n" \
    "AN+v6ZdQCINXtMxiZfaQguzH0yxrMMpb7NnDfcdAwRgUi+DoM3ZJKuM/IUmTrE4O\n" \
    "rz5Iy2Xu/NMhD2XSKtkyj4zl93ewEnu1lcCJo6m67XMuegwGMoOifooUMM0RoOEq\n" \
    "OLl5CjH9UL2AZd+3UWODyOKIYepLYYHsUmu5ouJLGiifSKOeDNoJjj4XLh7dIN9b\n" \
    "xiqKqy69cK3FCxolkHRyxXtqqzTWMIn/5WgTe1QLyNau7Fqckh49ZLOMxt+/yUFw\n" \
    "7BZy1SbsOFU5Q9D8/RhcQPGX69Wam40dutolucbY38EVAjqr2m7xPi71XAicPNaD\n" \
    "aeQQmxkqtilX4+U9m5/wAl0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNV\n" \
    "HQ8BAf8EBAMCAQYwHQYDVR0OBBYEFMSnsaR7LHH62+FLkHX/xBVghYkQMA0GCSqG\n" \
    "SIb3DQEBBQUAA4IBAQCjGiybFwBcqR7uKGY3Or+Dxz9LwwmglSBd49lZRNI+DT69\n" \
    "ikugdB/OEIKcdBodfpga3csTS7MgROSR6cz8faXbauX+5v3gTt23ADq1cEmv8uXr\n" \
    "AvHRAosZy5Q6XkjEGB5YGV8eAlrwDPGxrancWYaLbumR9YbK+rlmM6pZW87ipxZz\n" \
    "R8srzJmwN0jP41ZL9c8PDHIyh8bwRLtTcm1D9SZImlJnt1ir/md2cXjbDaJWFBM5\n" \
    "JDGFoqgCWjBH4d1QB7wCCZAA62RjYJsWvIjJEubSfZGL+T0yjWW06XyxV3bqxbYo\n" \
    "Ob8VZRzI9neWagqNdwvYkQsEjgfbKbYK7p2CNTUQ\n" \
    "-----END CERTIFICATE-----\n";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("Connecting to WiFi network: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  client.setCACert(test_root_ca);
  Serial.println("");
  Serial.println("Wifi connected, IP address of ESP:");
  Serial.println(WiFi.localIP());

  client.stop();
  if (client.connect(host, 443)) {
    String url = "/api/forecast/daily?insee=65292";
    Serial.println("Connection to server was successful");
    String token = "MY_TOKEN";
    client.print(String("GET ") + url + " HTTP/1.0\r\n" + "Authorization: " + token + "\r\n" + "Host: " + host + "\r\n" + "User-Agent: ESP32\r\n" + "Connection: close\r\n\r\n");
    while (client.connected()) {
      String line = client.readStringUntil('\n'); //HTTP HEADER
      Serial.println(line); //comment this line if u dont want to see server HTTP header
      if (line == "\r") {
        break;
      }
    }
    String line = client.readString(); //PAYLOAD
    Serial.println(line); //comment this line if u dont want to see server response (our payload)
  } else if (!client.connect(host, 443)) {
    Serial.println("Connection to website wasnt successful, ");
  }
}

void loop() {

}`

jmlocci, 2021-01-05:
Hello Martin

Many thanks for that. I am using an nodeMCU board 12E type based on an ESP8266, so the code will probably not compile straight away and will need some modifications, I have to use the ESP8266WiFi library. Or... I can get myself a ESP32, It was planned anyway.

martinius96, 2021-01-05:
This one is working for ESP8266, but it  is with setInsecure() option: https://pastebin.com/2nJYxL3e
I was testing it under Arduino Core 2.5.2 for ESP8266 :-) A bit older....
When I was trying with SHA1 fingerprint, I was unable to connect to host, idk why...
At that host, there will be end of DST Root CA X3 at 27th February, it will expire. 

Also server cert will expire until new CA will issue it (it will also change fingerprint). 
Fingerprint nowadays is working for year or two (if certificate is issued for 2 years). 
Year is more often...

jmlocci, 2021-01-05:
Ni again Martin.

I have used your updated script with my own API key. Good progress (check serial output below), but not quite there. The certificate must be the issue. Curiously, when I modify randomly my token the serial output remains the same (and I am sure my token is correct, it works on a browser).

I have ordered a set of ESP32 to use with the appropriate library, I will need some anyway. I will keep you posted.

Thanks again.

`.
Wifi connected, IP address of ESP:
192.168.1.99
Connection to server was successful
HTTP/1.1 401 Unauthorized
Server: nginx
Date: Tue, 05 Jan 2021 22:00:55 GMT
Content-Type: application/json
Connection: close
Cache-Control: private, must-revalidate
pragma: no-cache
expires: -1
Vary: Authorization

{"message":"Authentication Required"}
`

martinius96, 2021-01-15:
Hello there. 
Can i try it with some testing API key?
If it is possible, you can send it to me: martinius96@gmail.com
And i can try it and let u know.

martinius96, 2021-01-15:
I have tested it again and also I checked documentation...
Based on documentation, there are 2 available methods for Authorization:
- Bearer token header in request
- parameter in GET request (like: /api/forecast/daily?insee=65292&token=XYZKEY)

Last time we were using without Authorization Bearer, so it was working as Basic Auth that require username and password.
There is example for Bearer token (tested with ESP8266 - NodeMCU v3 Lolin) - insecured: https://pastebin.com/vJZgupYh
Now working :-)
[image]
ESP32 example (havent tested, but should work too): https://pastebin.com/8ibA8iGv

bblanchon, 2021-01-21:
@jmlocci, what is the status of this issue?

jmlocci, 2021-01-21:
Sorry Benoît, I thought you had closed the issue. I have received some help
that resolved my problem (from Martin, thanks to him).

Merci !!!

closed 2021-01-21 by bblanchon