bblanchon/ArduinoJson #1467: platformio native serializeMsgPack UnicodeDecodeError

jschoch, 2021-01-07:
I am getting this error using platformio native:  all versions should be the most recent.

 UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 55: character maps to <undefined>

serializeJson below works fine, but swapping msgPack fails with that error.
cpp
```c++
DynamicJsonDocument msgDoc(100);
char outBuffer[100];

void initMsgDoc(){
    msgDoc["e"] = 0;
    msgDoc["id"] = 0;
    msgDoc["mid"] = 0;
    msgDoc["s"] = 0;
    msgDoc["t"] = 0; 
}

...

void Hap::send(Evt e){
    msgDoc["e"] = 0;
    //size_t len = serializeMsgPack(msgDoc,outBuffer);
    size_t len = serializeJson(msgDoc, outBuffer);
    printf(" sending event: %d len: %d\n%s",e,len,outBuffer);
}
```

bblanchon, 2021-01-08:
Hi @jschoch,

> UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 55: character maps to

This looks like a Python error; how did you get it?

Best regards,
Benoit

jschoch, 2021-01-08:
compiling with vscode and platformio.

bblanchon, 2021-01-09:
This looks like a bug in PlatformIO's serial monitor, not in ArduinoJson.
It perfectly normal for [`serializeMsgPack()`](https://arduinojson.org/v6/api/msgpack/serializemsgpack/) to generate invalid Unicode characters.
In MessagePack `0x81` is the first byte of an object with one element ([see specs](https://msgpack.org/)).

jschoch, 2021-01-12:
Ok, thanks!

closed 2021-01-12 by jschoch

pharapeti, 2021-01-30:
Running into the same issue here:

```
void test_arduino_json_serialization(void) {
  ArduinoJson::StaticJsonDocument<200> doc;
  doc["hello"] = "world";
  doc["name"] = "peter krauss";

  char output [255];;
  int no_bytes_written = ArduinoJson::serializeMsgPack(doc, output);
  TEST_ASSERT_EQUAL(no_bytes_written, 31);

  char expected [255] = "82 A5 68 65 6C 6C 6F A5 77 6F 72 6C 64 A4 6E 61 6D 65 AC 70 65 74 65 72 20 6B 72 61 75 73 73";
  TEST_ASSERT_EQUAL_STRING(expected, output);
}
```

bblanchon, 2021-01-30:
@pharapeti, I don't think your problem is related to the current issue.
Please [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new/choose) if you need assistance.

BTW, if you want to verify the output of `serializeMsgPack()`, you must compare it with actual binary data and not with a hexadecimal representation.
For example: `char expected [255] = "\x82\xA5hello...`
See the [test suite](https://github.com/bblanchon/ArduinoJson/blob/v6.17.2/extras/tests/MsgPackSerializer/serializeObject.cpp#L41) for an example.
Also, remember that MessagePack documents can contain zeros, which is probably not supported by `TEST_ASSERT_EQUAL_STRING`.

pharapeti, 2021-01-31:
@bblanchon I ended up solving the issue after posting this comment. The solution was to create the string using the binary data (as you also suggested) 😄 

I ran into the same issue described in by the creator or this issue. I believe it occurs when attempting to print a hexadecimal character with `printf("%c", some_char_array_containing_hex_characters[x])`. Due to this, the platformio debugger was unable to represent the character in Unicode encoding resulting in a `UnicodeDecodeError`. The solution was to print to the character using the `x` or `X` specifier. For example: `printf("%X", some_hex_char)` - however still not an issue with this project.

bblanchon, 2021-01-31:
> I ended up solving the issue after posting this comment

When this happens, please say so immediately to avoid wasting the time of others.