bblanchon/ArduinoJson #1468: Max. DynamicJsonDocument doc size in esp32?

bhavikbhansali, 2021-01-11:
I am developing an IoT project where I want to parse a large JSON string. so I want to know what is the max size we can parse with this library. or what is the best way to do it. I am having trouble with the same.

Currently can mention DynamicJsonDocument doc(82000; only.

**Current Library Version: 6.15.2
(below all details taken from website)
Input length: 48070+
Data structures: 99344+
Strings: 308
Total(Strings): 99652+
Total (recommended): 131072+**

Arduino IDE 1.8.5
**Code Size:**
Sketch uses 895158 bytes (68%) of program storage space. Maximum is 1310720 bytes.
Global variables use 57288 bytes (17%) of dynamic memory, leaving 270392 bytes for local variables. Maximum is 327680 bytes.

Board: Custom Design

**ESP32 Details: WROOM32D (8MB)**

Thanks.

bblanchon, 2021-01-12:
Hi @bhavikbhansali,

The largest consecutive block of memory limits the capacity of `DynamicJsonDocument`.
On ESP32, you can get the size of this block with `ESP.getMaxAllocHeap()`.

Also, the 8MB you mentioned are in an external PSRAM chip; to use it with ArduinoJson, please see [How to use external RAM on an ESP32?](https://arduinojson.org/v6/how-to/use-external-ram-on-esp32/).

Best regards,
Benoit

bhavikbhansali, 2021-01-12:
> Benoit

Thanks for the information. Will go through it and get back here.

bhavikbhansali, 2021-01-12:
> Hi @bhavikbhansali,
>
> The largest consecutive block of memory limits the capacity of `DynamicJsonDocument`.
> On ESP32, you can get the size of this block with `ESP.getMaxAllocHeap()`.
>
> Also, the 8MB [...]

I tried to get it on a serial monitor and the results are as below.
getMaxAllocHeap:  113792
doc.capacity(): 0

If I understand it correctly. allowed size is 113792 but when I tried to define it will not able to allocate it.

Please correct me if I am wrong.?

bhavikbhansali, 2021-01-12:
What I found is that there is no PSRAM in WROOM32D. So is there any alternate way for this?

bblanchon, 2021-01-13:
Hi @bhavikbhansali,

I don't know why the allocation fails in your case, because on my ESP32:

```c++
DynamicJsonDocument doc(ESP.getMaxAllocHeap());
Serial.println(doc.capacity());
```

prints:

```
124952
```

And, of course, if I add just one byte, like so:

```c++
DynamicJsonDocument doc(ESP.getMaxAllocHeap() + 1);
Serial.println(doc.capacity());
```

then it prints:

```
0
```

If the Assistant says you need a capacity of 128 KB, then this is not possible to store the entire document on an ESP32 because, as we just saw, there is only 111 KB of consecutive RAM available.
There are, however, some workarounds; see [How to deserialize a very large document?](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/)

Best regards,
Benoit

bhavikbhansali, 2021-01-15:
Is there any way to check the Flash size of the WROOM32D I have. I have dought that the mentioned 8MB memory by the vendor is not true or not?

bblanchon, 2021-01-17:
I'm sorry, @bhavikbhansali, I don't know how to verify that your board does have 8MB of Flash as claimed by the vendor.
Do you have any questions about ArduinoJson, or can we close this issue?

bhavikbhansali, 2021-01-22:
@bblanchon Thanks for the help. I think we can close this issue now as I need more time to try few things. If I found that I need any help will get in touch with you.

Thanks for the help. :-)
Bhavik

closed 2021-01-22 by bhavikbhansali