bblanchon/ArduinoJson #1469: Compile problem with JSON Doc

wogoos, 2021-01-11:
Dear Blanchon,
I use your latest (14-11-20) JSON library for a while, and it works nice for me on flat JSON Key-Value files.  I need to generate a more complicated JSON structure with arrays of JSON object which I manually made and checked on JSON syntax errors.  
I have used this JSON as input for https://arduinojson.org/v6/assistant/  (ESP8266, Serialze, Stream) and copied the code it generates in my program.  I have compiler errors on this piece of code.
I hope you can tell me what the problem is and what I need to do to correct it. 
Thanks Oscar Goos

The error is :
lib\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp:18:26: error: 'class ArduinoJson6172_91::ArrayRef' has no member named 'write'.

My JSON is a configuration file for wifi[4] and serial ports[2].
{	"wifi": [{
		"DSid": "Wifi_cfg",
		"cmd": "ACK",
		"WIND": "2",
		"SSID": "INFINITUM8D73",
		"WPWS": "1894256060",
		"PHYM": "802.11N",
		"CHAN": "6",
		"TPWR": "19.5",
		"PROT": "WPA2",
		"KEY": "3",
		"HID": "0",
		"DHCP": "0",
		"NWIP": "192.168.1.80",
		"GWIP": "192.168.1.254",
		"MASK": "255.255.255.0",
		"DNS1": "8.26.56.26",
		"DNS2": "8.20.247.20"
	},{
		"DSid": "Wifi_cfg",
		"cmd": "ACK",
		"WIND": "2",
		"SSID": "INFINITUM8D73",
		"WPWS": "1894256060",
		"PHYM": "802.11N",
		"CHAN": "6",
		"TPWR": "19.5",
		"PROT": "WPA2",
		"KEY": "3",
		"HID": "0",
		"DHCP": "0",
		"NWIP": "192.168.1.80",
		"GWIP": "192.168.1.254",
		"MASK": "255.255.255.0",
		"DNS1": "8.26.56.26",
		"DNS2": "8.20.247.20"
	},{
		"DSid": "Wifi_cfg",
		"cmd": "ACK",
		"WIND": "2",
		"SSID": "INFINITUM8D73",
		"WPWS": "1894256060",
		"PHYM": "802.11N",
		"CHAN": "6",
		"TPWR": "19.5",
		"PROT": "WPA2",
		"KEY": "3",
		"HID": "0",
		"DHCP": "0",
		"NWIP": "192.168.1.80",
		"GWIP": "192.168.1.254",
		"MASK": "255.255.255.0",
		"DNS1": "8.26.56.26",
		"DNS2": "8.20.247.20"
	},{
		"DSid": "Wifi_cfg",
		"cmd": "ACK",
		"WIND": "2",
		"SSID": "INFINITUM8D73",
		"WPWS": "1894256060",
		"PHYM": "802.11N",
		"CHAN": "6",
		"TPWR": "19.5",
		"PROT": "WPA2",
		"KEY": "3",
		"HID": "0",
		"DHCP": "0",
		"NWIP": "192.168.1.80",
		"GWIP": "192.168.1.254",
		"MASK": "255.255.255.0",
		"DNS1": "8.26.56.26",
		"DNS2": "8.20.247.20"
	}],
	"Serial": [{
		"BD": "115200",
		"FRM": "8n1",
		"PRT": "1"
	},{
		"BD": "115200",
		"FRM": "8n1",
		"PRT": "1"
	}]
}

wogoos, 2021-01-12:
Further investigation I  found a part of the problem. 
The compiler seems to crash on the bottom line serializeJson(doc, output); which is in my esp8266 case serializeJson(doc, Serial);
Without that line it compiles normal.

bblanchon, 2021-01-12:
Hi Oscar,

Please share the relevant parts of your program and the complete compiler output.

Best regards,
Benoit

closed 2021-01-28 by bblanchon