bblanchon/ArduinoJson #1472: Problem Using Arduino Uno and ESP8266 HUZZAH with openweathermap API data

eclefley, 2021-01-14:
I am working on using an Arduino Uno and ESP8266 HUZZAH to get data from the open weather map api.
My end goal is to display the weather data on an Adafruit TFT Touch Screen Breakout Board. So far, I am successfully connecting to the wifi and the api server using the ESP8266 HUZZAH, and I have achieved serial communication between the Arduino Uno and the ESP8266 HUZZAH using the RX and TX pins.

My process is as follows: I open the ESP8266 HUZZAH Arduino code through the terminal and upload that code to the HUZZAH via usb serial, and I open the Uno firmware code through the Arduino 4 application and upload that code to the Uno via usb modem. Once the code is uploaded to both boards, I open the serial monitor for both sketches, and then connect the RX and TX pins for both boards accordingly (Uno RX -> HUZZAH TX, and Uno TX -> HUZZAH RX). At this point, the serial monitor for the HUZZAH states that the wifi and the server are connected.

When I open my IP address in the web browser URL, the page states that there was a problem loading the page, and in the serial monitors, I get the following errors:

The error I get in the ESP8266 HUZZAH serial monitor is: {"type":"request"}deserializeJson() failed: InvalidInput

The errors I get in the Arduino Uno serial monitor are: deserializeJson() failed: NoMemory, and deserializeJson() failed: InvalidInput

When I open my IP address in the web browser URL, I should see the data displayed for the "type" of data I requested (in the code below, I am requesting the data types "temp" and "description").

In the Arduino Uno serial monitor, I should see: {"type":"response","Temperature":57.9,"Weather:":overcast clouds}

In the ESP8266 HUZZAH serial monitor, I should see: {"type":"request"}

Once that data is properly displayed in the web browser, I can move onto displaying the data on the Adafruit TFT Touch Screen Breakout Board, but for now I am blocked by these problems.

I am using ArduinoJson 6. I have done extensive research on how to solve these problem/have tried several recommendations to fixing these problems, and still cannot seem to figure it out.

Any help would be greatly appreciated! 

Here is the code I upload to the ESP8266 HUZZAH:

`#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer server;

WiFiClient client;

// Replace with your SSID and password details
char ssid[] = "SSID";       
char pass[] = "PASSWORD";   

const char servername[] = "api.openweathermap.org";

String zip = "ZIPCODE";

String apiKey = "APIKEY";

int status = WL_IDLE_STATUS;

unsigned long lastConnectionTime = 10 * 60 * 1000;     
const unsigned long postInterval = 10 * 60 * 1000; 

void setup() {
  Serial.begin(9600);
   
  WiFi.begin(ssid,pass);
  Serial.println("connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi Connected");
  printWiFiStatus();

  server.on("/",handleIndex);
  server.begin();
}

void loop() {
  if (millis() - lastConnectionTime > postInterval) {
    // note the time that the connection was made:
    lastConnectionTime = millis();
    makehttpRequest();
  }
  server.handleClient();
}

// print Wifi status
void printWiFiStatus() {
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);
}

void makehttpRequest() {
  // if there's a successful connection:
  if (client.connect(servername, 80)) {
    Serial.println("connected to server");
   
    client.println("GET /data/2.5/weather?zip="+zip+"&units=metric&APPID="+apiKey);
    client.println("Host: api.openweathermap.org");
    client.println("User-Agent: ESP8266/0.1");
    client.println("Connection: close");
    client.println();
   
    while (client.available()) client.read();
   
    client.stop();
 }
}

void handleIndex()
{
  //send a JSON-formatted request with key "type" and value "request"
  //then parse the JSON-formatted response with keys 
  DynamicJsonDocument doc(27740);
  double tempNow = 0, weatherNow = 0;
  //sending the request
  doc["type"] = "request";
  serializeJson(doc,Serial);
  //reading the response
  boolean messageReady = false;
  String message = "";
  while(messageReady == false) { //blocking but that's ok
    if(Serial.available()) {
      message = Serial.readString();
      messageReady = true;
    }
  }
 
  //attempt to deserialize the JSON-formatted message
  DeserializationError error = deserializeJson(doc,message);
  if(error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }
  tempNow = doc["main"]["temp"];
  weatherNow = doc["weather"][0]["description"];
 
  String output = "Temperature: " + String(tempNow) + "\n";
  output += "Weather: " + String(weatherNow);

  server.send(200, "text/plain", output);
}`

Here is the code I upload to the Arduino Uno: 

`
#include <ArduinoJson.h>

String message = "";
bool messageReady = false;

void setup() {
  Serial.begin(9600);
}

void loop() {
  //Monitor serial communication
  while(Serial.available()) {
    message = Serial.readString();
    messageReady = true;
  }
  //only process message if there's one
  if(messageReady) {
    //the only messages we'll parse will be formatted in JSON
    DynamicJsonDocument doc(27740); //ArduinoJson version 6+
    //attemp to deserialize the message
    DeserializationError error = deserializeJson(doc,message);
    if(error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());
      messageReady = false;
      return;
    }
    if(doc["type"] == "request") {
      doc["type"] = "response";
      doc["main"]["temp"];
      doc["weather"][0]["description"];
      serializeJson(doc,Serial);
    }
    messageReady = false;
  }
}`

eclefley, 2021-01-14:
I meant to change the memory to 1024 in this example instead of 27740

bblanchon, 2021-01-14:
Hi @eclefley,

There is a lot of information in your message, and I don't have time to process everything.
What I can tell you, however, is that 27740 is way too big for the Arduino UNO, which has only 2KB of RAM.
Make sure to read [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/).

Also, I would be very grateful if you could summarize your question in just two short sentences.
Start by telling what works and what doesn't, and exclude irrelevant details (like the ESP8266, as I understand it).

Best regards,
Benoit

closed 2021-01-28 by bblanchon