bblanchon/ArduinoJson #1475: HTTPS Get request with Filtering - Question

victor7376, 2021-01-20:
I'm trying to connect to an API server which has two functions, one shows a fortnightly payout & 2nd showing when someone is broadcasting on a webcam.

I've read through tutorials and different guides, but I cant find examples of how to do certain functions.

Its not really an 'issue' but a question.

Is there any examples of how to make a HTTPS GET request (the site requires secure API call which I've been successful in making with a fingerprint and certificate in different scripts) and filter out the sections you dont require.

Assistant Filtering
The 'payout' filtering from 768 to 192.
The 'online' filtering from 6144 to 384

I've got an ESP8266 (filtering is for ESP8266) & ESP32.

I have watched youtube vids and followed guides on making HTTP Get requests and deserialize them successfully, its just the https ones I cant seem to find with filtering.

Cheers for your help and advice.

bblanchon, 2021-01-21:
Hi,

There are two ways to connect to an HTTPS server from an ESP8266/ESP32:

* `WiFiClientSecure`, a replacement for `WiFiClient`/`EthernetClient` that supports SSL
* `HttpClient`, an HTTP library built on top of `WiFiClientSecure`

You can find an example using `HTTPClient` in the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/).

You can find an example using `WiFiClientSecure` in the Reddit case-study of [Mastering ArduinoJson](https://arduinojson.org/book/).

In both cases, the certificate's validation can be a source of trouble, see #1465.

Best regards,
Benoit

victor7376, 2021-01-21:
Bought the book and looked at the code thats included, its helped a LOT.

Finally getting something working how I wanted.

Thank you

bblanchon, 2021-01-22:
You're welcome.
Thank you very much for purchasing the book, @victor7376!

closed 2021-01-22 by bblanchon