bblanchon/ArduinoJson #1477: Help with parsing large integer arrays in JSON

aceisace, 2021-01-22:
Apologies for opening yet another issue. As a fellow dev, I understand this well so I'll try to take as less time as possible from you.  
I also bought your great e-book a few hours ago (pretty awesome and well structured), and your ArduinoJson website is awesome too! But C++ is still new to me, so I'm having a bit of a hard time.

**Description of the problem:**
I try to parse a very large integer-only array from a JSON response, but I cannot get it to work 1) fast 2) efficiently 3) without using all the memory.

**Background info:**
I'm using an ESP32-WROVER-E (4MB SRAM) with a display. The task of the ESP is to parse the image buffer (very large int-array) from a given response (e.g. [TEST-LINK](https://ace44.eu.pythonanywhere.com/testbuffer/)). In the demo code of the display, this buffer was saved as a byte array (requiring way less memory), but JSON doesn't support bytes, so I'm using an integer-array instead. 

ArduinoJson Assistant says that parsing this responce would require 4.2 MB of RAM, but the ESP32 I'm using has 4MB only. 
I tried to comprehend the examples with chunking data and de-serializing it, but I cannot get it to work as expected. Below is my current code.

**Target Platform**: ESP32-WROVER-E with 4MB external RAM.
**compiler model and version**: Arduino IDE v1.8.13
**MVCE**: See attached code and [TEST-LINK](https://ace44.eu.pythonanywhere.com/testbuffer/) 
**compiler output**: *Out of Memory*

**Code**:
```cpp
#include <WiFi.h>
#include <HTTPClient.h>
#define ARDUINOJSON_DECODE_UNICODE 1
#include <ArduinoJson.h>
#include <Arduino.h>

// Display parameters
const int EPD_WIDTH = 960;
const int EPD_HEIGHT = 540;

//WiFi
const char *ssid = "***";
const char *password = "***";

// URL for data which needs to be parsed
const char *url = "https://ace44.eu.pythonanywhere.com/testbuffer/";

// external RAM framebuffer
uint8_t *framebuffer;

// Formalities for using SPIRAM
struct SpiRamAllocator {
    void* allocate(size_t size) { return ps_malloc(size); }
    void deallocate(void* pointer) { free(pointer); }
};

using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;

void setup(){

    delay(500); // give the ESP some time

    psramInit(); // init external RAM

    Serial.begin(115200); //init Serial

    Serial.println((String)"Memory available in PSRAM : " + ESP.getFreePsram()); // 4 MB SPIRAM

    // Allocate size for buffer which is sent to the display
    framebuffer = (uint8_t *)heap_caps_malloc(EPD_WIDTH * EPD_HEIGHT / 2, MALLOC_CAP_SPIRAM);
    memset(framebuffer, 0xFF, EPD_WIDTH * EPD_HEIGHT / 2);

    // Connect to WiFi
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED){
            delay(100);
            Serial.print(".");
    }
    Serial.print("\nWiFi connected with IP : ");
    Serial.println(WiFi.localIP());
    // WiFi connected -> ready for parsing

    Serial.println("Downloading JSON");

    HTTPClient http; //init HTTP Client
    http.useHTTP10(true); 
    http.begin(url); //Use URL from provided Link to get a response
    http.GET();

    // Tried to make it all fit in a external SPIRAM document -> didn't work -> ran out of memory
    SpiRamJsonDocument doc(3145728); //3MB for documents

    // Usual handling of errors + deserialize JSON from URL
    DeserializationError error = deserializeJson(doc, http.getStream() );
    if(error){
            Serial.print("deserializeJson() failed: ");
            Serial.println(error.c_str());
    }

    // The text file being served on the website is 1,1MB, but this part is problematic, it takes more space than it should.
    Serial.println((String)"JsonDocument Usage Memory: " + doc.memoryUsage());

    // Convert the buffer to an integer array. Load the values of the response to the integer array named framebuffer.
    // Can't get this to work as expected.
    JsonArray array = doc["data"].as<JsonArray>();
    int counter = 0;
    for(JsonVariant v : array) {
        framebuffer[counter] = (v.as<int>());
        counter++;
    }

    Serial.println('Done');
    http.end();
}

void loop(){
}
```
Note about Code:
The frambuffer-array must be sent all at once to the display or it won't function as expexted. 1MB is required for the framebuffer array, leaving a bit more than 3MB for parsing the JSON array.

However, I'm having difficulties with the parsing, I checked the FAQ page and the page about deserializing chunks or using a filter, but I couldn't get it to work..

Could you please help with parsing the very large integer array? Thanks in advance

bblanchon, 2021-01-23:
Hi @aceisace,

Thank you very much for purchasing my book, your support means a lot to me 👍 

When I saw [the sample](https://ace44.eu.pythonanywhere.com/testbuffer/), I immediately thought "this is not a use case for JSON!".
Indeed, JSON is very good at representing nested objects and arrays, but it's very bad for a flat buffer like an image.

Instead, I recommend that you use a binary format.
You can create your own format or use a simple existing one (like [bmp](https://en.wikipedia.org/wiki/BMP_file_format)).
You might be interested in [this video](https://youtu.be/hAB4TdX8dwM) where I explain how to do serialization from scratch.

By the way, there is a lot of repetition in the data, you should implement a simple compression algorithm, like [RLE](https://en.wikipedia.org/wiki/Run-length_encoding).

Best regards,
Benoit

aceisace, 2021-01-23:
Thanks for getting back so fast @bblanchon ! Awesome support : )
You're welcome! Good work deserves at least this much :D.

aceisace, 2021-01-23:
Thanks for getting back so fast @bblanchon ! Awesome support : )
About the book, you're welcome! Good work deserves at least this much :D.

I was hoping to make use of JSON because the width and height of the image need to be known too. The best way to transfer grouped data was JSON and I was familiar with it already, but I guess there are better ways then : )

Thanks for pointing out the recources about BMP, and the video! I'll see if I can make some sense out of it with my little C++ expierence. But at least I now have a direction to look for.

Once again, thanks for your awesome ArduinoJSON library, for your awesome website and Assitant and for your fast support!

Best Regards,
Ace

aceisace, 2021-01-23:
As this specific question has been resolved, I'll close this issue now : )

closed 2021-01-23 by aceisace

bblanchon, 2021-01-24:
> the width and height of the image need to be known too

There should both be present in the headers of the binary format.
For example, in a custom format, you could decide that the first word contains the width and the second contains the height.

Alternatively, if you need more information than just width and height, you could use a different URL for the JSON document and the image. For example, the JSON enpoint could return:

```json
{
  "width": 960,
  "height": 540,
  "url": "/path/to/image.bmp",
  "title": "Something else"
}
```

This is the way [RESTful APIs](https://en.wikipedia.org/wiki/Representational_state_transfer) work and should be very easy to implement with your Python backend.

(As a side note, I forgot to mention that you could use MessagePack to solve this issue, but I think the above solution is better)

aceisace, 2021-01-24:
Thanks for the suggestion @bblanchon !
Indeed I am using a RESTful API for the backend (python3 flask) so restructuring the image data is a piece of cake. 
The problem is the parsing of that data on the esp, as I'm not an expert, I look for some examples, try to understand them and implement them to fit my use-case,

My ESP does have 8MB of SPIRAM, which should be more than enough for a single image even as an integer array, but the last 4MB are treated differently than the first 4MB (hard to access), hence this isn't an option.

I can send the image as a byte array, which could save a lot of space, but the problem is how to download and parse it as a int/byte array the ESP32 can understand.
[image]

Sending a bitmap version of the image is certainly possible, but having the ESP32 compute the pixel data one by one will make it much slower. And since there is already a powerful backend program converting image to int/byte arrays, having the ESP do the extra work probably isn't the best option.

One case that seemed interesting is to chunk data, e.g. instead of a single 'data' array, 4 could be used. The ESP32 reads from the first one, writes it to the array, frees the memory, parses the next one...until all are completed. But this doesn't look like an elegant solution, there probably are better ways to do this.

I wonder if it's possible to 'remember' the position of the array form the web response. That way, the ESP32 could download e.g. 2MB of data, write it to the buffer, download the next 2 MB of data, parse it again and done. But I haven't found a way to get this working yet.

I'll take a look at MessagePack and will tinker with it to see if I can get it working : )

bblanchon, 2021-01-25:
First, you need to send a binary response, not a string representing the array.
I'm not familiar with Flask, but I think it should be something like:

```python
@app.route('/path/to/image')
def get_image():
    return make_response(bytes(image_pixels))
```

...where `image_pixels` is your existing `list` of pixels.
The python code could be even simpler if you had a file with the right format, as a call to `send_file()` would to the job.

On the ESP, you simply have to call [`Stream::readBytes()`](https://www.arduino.cc/reference/en/language/functions/communication/stream/streamreadbytes/), and if the data is already in the expected format, you don't have to do any conversion.

aceisace, 2021-01-30:
Thanks for your help so far @bblanchon !
I dug a bit deeper in C++ and the microcontroller. Finally got an image on the display.

I took your advice to make a binary response from the server, like this:
```
222222222222222222223333333333333333333343443333333333333333333333333333334444445555554444544544444444444444444444444444444444556566666666666655555544444444443333333333333333333333333333333333333333333333333333443333334444443433333344445445444444443333333333444444444444444444444444444444444444444444444454555555555555555545444444444444444444444444444444444444445455555555555555555555555
```
Every two letters e.g. 22 (0x22) represent a hex value from 0-255. I used `Stream::readBytes()` to store incoming bytes in a buffer. Then used a function to calculate the decimal value of two bytes -> 22 (= 0x22) -> 34. This value is then saved in the image buffer at the correct index.

It's probably not the ideal solution, but at least it works. Once again, thanks for the useful tips!