bblanchon/ArduinoJson #1479: Can't initialize StaticJsonDocument outside setup()

tuloski, 2021-01-26:
I'm trying try to initialize a staticJsonDocument with the following code:

const int capacity = JSON_OBJECT_SIZE(3);
StaticJsonDocument<capacity> doc;
doc["a"] = "1";
doc["b"] = "2";
doc["c"] = "3";

If the code is outside the setup(), I get the error "doc does not name a type" when I try to fill the data.
If the code is inside the setup() everything is ok.
What is going on? Is this working as supposed?

bblanchon, 2021-01-27:
Hi @tuloski,

This is not a limitation of the library but a limitation of the language.
Indeed, C++ doesn't allow statements in the global scope.
Instead, statements must be in a function, typically `main()` in a regular program, or `setup()`/`loop()` in an Arduino sketch.

Best regards,
Benoit

tuloski, 2021-01-27:
Ouch, me noob :).
Thanks for the answer.

closed 2021-01-27 by tuloski

bblanchon, 2021-01-27:
You're welcome.
Thank you for using ArduinoJson.