bblanchon/ArduinoJson #1480: JsonDocument dynamic allocation problem

Leo-C, 2021-01-26:
Hi,
the title not refers to `DynamicJsonDocument`, but it refers precisely to dinamic allocation of a `JsonDocument` with `new` !

My environment is:
**Platform**: ESP32 TTGO
**IDE**: Arduino IDE 1.8.13

The simple code below compile well, but crash at `delete`:

```
#include <Arduino.h>
#include <ArduinoJson.h>

void setup()
{
	Serial.begin(115200);
	while(!Serial) {} // Portability for Leonardo/Micro
}

void loop()
{
	Serial.println("Mem before: " + String(ESP.getFreeHeap()));
	
	JsonDocument* doc = new DynamicJsonDocument(1024);
	delete doc;
	
	Serial.println("Mem after: " + String(ESP.getFreeHeap()));
}
```
(this is a minimal example for a scenario where I allocate dinamically a `JsonDocument` into a function, return a pointer, and only after use, I call destructor in a different scope with `delete` in a different scope)

I appreciate any help to understand my error

Many thanks in advance

Leonardo

Leo-C, 2021-01-26:
Sorry,
I've found that following code works:

```
#include <Arduino.h>
#include <ArduinoJson.h>

void setup()
{
	Serial.begin(115200);
}

void loop()
{
	Serial.println("Mem before: " + String(ESP.getFreeHeap()));
	
	DynamicJsonDocument* doc = new DynamicJsonDocument(1024);
	delete doc;
	
	Serial.println("Mem after: " + String(ESP.getFreeHeap()));
}
```
(only changing `doc` from type `JsonDocument` to `DynamicJsonDocument`)

Then, `JsonDocument ` is not a superclass of `DynamicJsonDocument`?

Many thanks

Leonardo

bblanchon, 2021-01-27:
Hi Leonardo,

Thank you very much for opening this issue.

`DynamicJsonDocument` is not intended to be deleted from a pointer to base.
In fact, it has no virtual function at all, and I don't want to pay the overhead just for this edge case.
As you'll see, I made the destructor of `JsonDocument` not `virtual` but `protected`, so the compiler will issue an error next time you try to `delete` a `JsonDocument`.

Now, I don't understand why anyone would want to do that since `DynamicJsonDocument` is already in the heap.
It seems that you want to return a `JsonDocument` from a function, right?
Why don't you write something like this:

```c++
DynamicJsonDocument makeJsonDocument() {
  DynamicJsonDocument doc(1024);
  // ...
  return doc;
}
```

`DynamicJsonDocument` is moveable, so no copy of the buffer is done, even if [RVO](https://stackoverflow.com/q/12953127/1164966) doesn't kick in.

Best regards,
Benoit

Leo-C, 2021-01-27:
Many Thanks!

I admit that I did not know Copy-Elision, then your suggestion is good and efficient.

I also have attempted to write a function like:
```
JsonDocument makeJsonDocument() {
  DynamicJsonDocument doc(1024);
  // ...
  return doc;
}
```
but was not accepted by compiler ...

Leonardo

bblanchon, 2021-01-28:
Yes, that's expected.
Allowing this would result in a [slicing](https://stackoverflow.com/q/274626/1164966) of `DynamicJsonDocument` and therefore to a memory leak.

closed 2021-01-28 by bblanchon

smuellener, 2021-01-30:
One question, does that mean that DynamicJsonDocument has built-in smart shared pointer functionality? In other words:

std::shared_ptr<DynamicJsonDocument> create_a_doc_for_usage()
{
    auto doc = std::make_shared<DynamicJsonDocument>(JSON_OBJECT_SIZE(1));
    (*doc)["Content"] = "Test";
    return doc;
}

would not bring any benefit over 

DynamicJsonDocument create_a_doc_for_usage()
{
    auto doc DynamicJsonDocument(JSON_OBJECT_SIZE(1));
    (*doc)["Content"] = "Test";
    return doc;
}

Thank you for the clarification!

bblanchon, 2021-01-30:
@smuellener, in a sense, it's more like `unique_ptr` because it doesn't implement reference counting.

bblanchon, 2021-02-15:
[ArduinoJson 6.17.3](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.3) includes the protected destructor.

commit c70a491f (2021-01-28): Made JsonDocument's destructor protected (fixes #1480)