bblanchon/ArduinoJson #1481: Multiple nested arrays

alexisicte, 2021-01-26:
Hello, 

I am using esp32 and i have a goal to create and send a json file over time. The json format should be the following:

`{"protected":{"ver":"v1","alg":"HS256","iat":1594133532},"signature":"e444c199b435f754e9e69c449dfca1c6c64709bc54138ebdce9e82bdb73f5998","payload":{"device_name":"3.933848","device_type":"DATA_FORWARDER","interval_ms":100,"sensors":[{"name":"red","units":"N/A"},{"name":"ir","units":"N/A"}],"values":[[-102.5,-71.75],[-103.375,-68.25],[-98.25,-61.75],[-78.25,-92.75],[-75.375,-27]]}}`

Therefore i used the ArduinoJson Assistant in order to get the code for the above format. The Assistant recommend the following:
`JsonObject _protected = doc.createNestedObject("protected");
  _protected["ver"] = "v1";
  _protected["alg"] = "HS256";
  _protected["iat"] = 1594133532;
  doc["signature"] = "e444c199b435f754e9e69c449dfca1c6c64709bc54138ebdce9e82bdb73f5998";

  snprintf(bufferBatToSend, sizeof bufferBatToSend, "%f", getVoltage());
  JsonObject payload = doc.createNestedObject("payload");
  payload["device_name"] = bufferBatToSend;
  payload["device_type"] = "DATA_FORWARDER";
  payload["interval_ms"] = 100;

  JsonArray payload_sensors = payload.createNestedArray("sensors");

  JsonObject payload_sensors_0 = payload_sensors.createNestedObject();
  payload_sensors_0["name"] = "red";
  payload_sensors_0["units"] = "N/A";

  JsonObject payload_sensors_1 = payload_sensors.createNestedObject();
  payload_sensors_1["name"] = "ir";
  payload_sensors_1["units"] = "N/A";

  JsonArray payload_values = payload.createNestedArray("values");

  JsonArray payload_values_0 = payload_values.createNestedArray();
  payload_values_0.add(red[0]);
  payload_values_0.add(ir[0]);

  JsonArray payload_values_1 = payload_values.createNestedArray();
  payload_values_1.add(red[1]);
  payload_values_1.add(ir[1]);

  JsonArray payload_values_2 = payload_values.createNestedArray();
  payload_values_2.add(red[2]);
  payload_values_2.add(ir[2]);

  JsonArray payload_values_3 = payload_values.createNestedArray();
  payload_values_3.add(red[3]);
  payload_values_3.add(ir[3]);

  JsonArray payload_values_4 = payload_values.createNestedArray();
  payload_values_4.add(red[4]);
  payload_values_4.add(ir[4]);`

It is obvious that for 5 values of red and ir arrays it is an easy task to type the above. But what if i am not aware of arrays' size or if the size is 300 values? It is not possible to type it manually.

So, does the library supports an automated way to do the above?

bblanchon, 2021-01-27:
Hi @alexisicte,

You can write a loop:

```c++
JsonArray payload_values = payload.createNestedArray("values");
for (int i=0; i<300; i++) {
  JsonArray array = payload_values.createNestedArray();
  array.add(red[i]);
  array.add(ir[i]);
}
```

Best regards,
Benoit

alexisicte, 2021-01-27:
Hi @bblanchon,

Many thanks for the prompt reply and your work regarding ArduinoJson.

From the above code a nested array named **payload_values** will be created under **array** array and the values will be added to **array** array.  Am i wrong on that?
I am seeking to create nested arrays under **payload_values** nested array for each pair (red & ir) of values.

bblanchon, 2021-01-28:
I think the code above does exactly what you asked.
Why don't you try?

alexisicte, 2021-01-28:
It is working. I am not understanding why but it does. 
I should give a better look to the documentation.
Apologies for wasting your time.

Thanks!

bblanchon, 2021-01-29:
You're welcome; thank you for using ArduinoJson, @alexisicte.
Good luck with your project!

closed 2021-01-29 by bblanchon