bblanchon/ArduinoJson #1482: Merging deserialized json document into a JsonVariant not complete

timkoers, 2021-01-26:
Hi!

Sorry to bother you again..

I'm trying to 'merge' a deserialized json document into the AsyncJsonResponses getRoot() JsonVariant.

```
AsyncJsonResponse *response = new AsyncJsonResponse();

bool ok = getStoredProfiles(response->getRoot());
```

```
bool getStoredProfiles(JsonVariant &root)
{
    bool ok = false;
    JsonArray profiles = root.createNestedArray("Profiles");

    Dir dir = LittleFS.openDir(STORED_PROFILES_PATH);
    while (dir.next())
    {
        ESP_LOGD("", "getStoredProfiles(): next file");
        if (dir.fileSize())
        {
            JsonObject profile = profiles.createNestedObject();
            File file = dir.openFile("r");
            ok = getFileInfo(file, profile);

            if (ok)
            {
                ESP_LOGD("", "Loading of data for file completed");
            }
            file.close();

            ESP_LOGD("", "getStoredProfiles(): moving onto next file");
        }
    }

    serializeJson(root, Serial);

    return ok;
}

```

The serialization call on **root** in this function prints out incomplete arrays. 
The serialization of **profile** and **profiles** is also incomplete.

getFileInfo looks like this

```
bool getFileInfo(File file, JsonObject &output)
{
    bool ok = false;
    const size_t size = file.size();

    if (size == 0)
        return false;

    ESP_LOGD("", "Filename: %s (size %d)", file.fullName(), size);

    DynamicJsonDocument root(MAX_JSON_SIZE); // This is a little better since the root will only be the filesize now
    DeserializationError errorCode;
    ReadBufferingStream bufferedFile(file, size);

    // The (const char*) cast is needed to force ArduinoJson to make a copy of the input, the buffer will get destroyed when this function returns
    //if (ok = ((errorCode = deserializeJson(root, (const char *)buf.get()), errorCode) == DeserializationError::Code::Ok), !ok)
    if (ok = ((errorCode = deserializeJson(root, bufferedFile), errorCode) == DeserializationError::Code::Ok), !ok)
    {
        ESP_LOGE("", "Error parsing JSON %s", errorCode.c_str());
        root[MemberNames::kError] = errorCode.c_str();
    }

    file.close();

    // For now deserializeJson(JsonVariant..) is not supported. Will be in V7 (https://github.com/bblanchon/ArduinoJson/issues/1226#issuecomment-695786965)
    //output.set(root.as<JsonVariantConst>());
    merge(output, root);

    serializeJson(output, Serial);

    return ok;
}
```

The serialization call on **output** is also incomplete, but when I call serialize on **root**, the output is as expected.
The merge function is the one from your documentation. (https://arduinojson.org/v6/how-to/merge-json-objects/)

What am I doing wrong?

bblanchon, 2021-01-27:
Hi @timkoers,

I'm sorry, I read your question several times, and I don't understand it.
All I can tell you is that the document is incomplete probably because it's full.
You can check [JsonDocument::overflowed()](https://arduinojson.org/v6/api/jsondocument/overflowed/) if you want to be sure.

If you need a more helpful answer, please write an MCVE, preferably on [wandbox](https://wandbox.org/permlink/RlZSKy17DjJ6HcdN).

Best regards,
Benoit

timkoers, 2021-01-27:
Once again, you are right, I didn't have access to the DynamicJson document so I couldn't verify the overflowed() call. I didn't notice that when creating an AsyncJsonResponse you could enter a manual document size. Thanks again!

closed 2021-01-27 by timkoers