bblanchon/ArduinoJson #1483: Issue with StaticJsonDocument

edro1961, 2021-01-30:
I can't seem to get the json doc to post to my Web API service on my server. I can port with Postman and it works fine. So after hunting and pecking for hours I gave in to come ask. After checking my code I see the serial output look like it's correct with the exception of the Content - Length.. Postman has it at 239 and the jsondocument has it at 220...

Please help....

 //Create JSON doc and write a "name" attribute
  const size_t capacity = JSON_OBJECT_SIZE(12);
  StaticJsonDocument<239> doc;
  doc["DateNTime"] = "2021-01-28T08:23:44.000";
  doc["Temperature"] = 73.5;
  doc["Humidity"] = 26;
  doc["LuxValue"] = 60;
  doc["M1"] = 123;
  doc["M2"] = 128;
  doc["M3"] = 122;
  doc["M4"] = 135;
  doc["M5"] = 119;
  doc["M6"] = 101;
  doc["M7"] = 156;
  doc["M8"] = 104;

  //POST request
  Serial.println("Begin POST Request");

  client.println("POST /api/garden HTTP/1.1");
  Serial.println("POST /api/garden HTTP/1.1");

  client.println("Host: www.edronet.net");
  Serial.println("Host: www.edronet.net");
 /* client.println("User-Agent: Arduino/1.0");
  Serial.println("User-Agent: Arduino/1.0");*/

  client.println("Content-Type: application/json; charset=utf-8");
  Serial.println("Content-Type: application/json; charset=utf-8");

 /* client.println("Connection: keep-alive");
  Serial.println("Connection: keep-alive");*/

  client.print("Content-Length: ");
  Serial.print("Content-Length: ");

  client.println(measureJsonPretty(doc));
  Serial.println(measureJsonPretty(doc));

  client.println();
  Serial.print(F("Sending: "));
  Serial.println();
  client.println();
  serializeJsonPretty(doc, Serial);
  Serial.println();
  client.println();

  //This works like client.println, but prints doc to client
  serializeJsonPretty(doc, client);

  
  //To let me know that request has been completed
  Serial.println("Sent POST Request");
 
[image]

edro1961, 2021-01-30:
Here is the postman code for the post
[image]

bblanchon, 2021-01-30:
Hi @edro1961,

Your program seems to send multiple empty lines between the headers and the body, but only one empty line is allowed by HTTP. You can find a complete example of an Arduino sketch sending a POST request with ArduinoJson at the end of [the deserialization tutorial](https://arduinojson.org/v6/doc/serialization/).

BTW, instead of duplicating each line to write to both `client` and `Serial`, I recommend that you use [`WriteLoggingStream`](https://github.com/bblanchon/ArduinoStreamUtils#logging-write-operations) as shown here: 

```c++
WriteLoggingStream client2(client, Serial);

client2.println("POST /api/garden HTTP/1.1");
client2.println("Host: www.edronet.net");
client2.println("User-Agent: Arduino/1.0");
client2.println("Content-Type: application/json; charset=utf-8");
client2.print("Content-Length: ");
client2.println(measureJsonPretty(doc));
client2.println();
serializeJsonPretty(doc, client2);
```

Finally, the difference in `Content-Length` seems to be due to the indentation (ArduinoJson uses 2 spaces) and the integer values variation.

Best regards,
Benoit

edro1961, 2021-01-31:
I found the issue... I was missing the HttpClient... Ugh.. Also I cleanned up the empty line.. Now I just need to figure out how to use my variable data in the DateNTime doc object..

Thanks

closed 2021-01-31 by bblanchon