bblanchon/ArduinoJson #1485: Not closing sockets in examples

TechnicalLee, 2021-02-01:
Hardware: Arduino Uno WiFi Rev 2
Software: Arduino IDE 1.8.13 with WiFiNINA library 1.8.2

I was using your OpenWeatherMap example from your book, but I ran into a **No socket available** error after repeated http errors. This is the part of the code where you check if the header is okay:

``` c++
// Check status code
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  // should be "HTTP/1.0 200 OK"
  if (memcmp(status + 9, "200 OK", 6) != 0) {
    Serial.print(F("Unexpected HTTP status"));
    Serial.println(status);
    return;
  }
```

About 10% of the time when fetching weather I get `Unexpected HTTP status` from the above. After that happened 10 times, I would get the `No socket available` error and no further communication is possible. I also noticed the amount of available memory decreased by 66 bytes after every error. After looking through the WiFiNINA library, I found there is a 10 socket limit which matches the number of failed attempts before the no sockets error. I conclude that sockets are being left open when you are returning from errors, so that's why they eventually run out. The fix is simple, you just need to add a `client.stop()` to close the socket before each return from any error:

``` c++
// Check status code
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  // should be "HTTP/1.0 200 OK"
  if (memcmp(status + 9, "200 OK", 6) != 0) {
    Serial.print(F("Unexpected HTTP status"));
    Serial.println(status);
    client.stop(); //<---Add here
    return;
  }
```

``` c++
// Deserialize the response
  DynamicJsonDocument doc(8192);
  DeserializationError err =
      deserializeJson(doc, client, DeserializationOption::Filter(filter));
  if (err) {
    Serial.print(F("deserializeJson() returned "));
    Serial.println(err.c_str());
    client.stop(); //<---Add here
    return;
  }
```

Those are the two instances in the OpenWeatherMap example, it looks like this would also affect the Adafruitio example in your book, and the JsonServer and JsonHttpClient examples in this library.

This is probably moving into another issue, but when I get those http errors it's always because the header is blank, so it seems like it's having trouble keeping up with the stream or interruptions sometimes. Every so often I also get a `deserializeJson() returned IncompleteInput` even though the response header was okay. You mentioned this could be some kind of stream race problem back in #969, but I was curious if you had any fix or update with your StreamUtils to add to the examples? Right now I'm just living with the 10% error rate.

bblanchon, 2021-02-01:
Hi @TechnicalLee,

Thank you very much for purchasing my book and reporting this issue.

As I see it, it's more a bug in the WiFiNINA library than in the samples.
Indeed, a correctly designed C++ class should not let a socket leak like that; instead, [it should reclaim all acquired resources in its destructor](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r1-manage-resources-automatically-using-resource-handles-and-raii-resource-acquisition-is-initialization).

Nevertheless, I added the calls to `client.stop()` to the Adafruit, OpenWeatherMap, and Reddit examples.
These examples were already calling `client.stop()` from the main path, so why argue :-)

I don't know what causes the 10% error rate, but I'm pretty confident it's not related to ArduinoJson.
I recommend that you add a `ReadBufferingStream` (as shown [here](https://arduinojson.org/v6/how-to/improve-speed/)) because it will speed up the deserialization and hopefully reduce the number of errors (unless WiFiNINA already has a buffering strategy in place).

Best regards,
Benoit

closed 2021-02-01 by bblanchon

TechnicalLee, 2021-02-02:
Thanks for fixing that so fast @bblanchon!

I will try the ReadBufferingStream sometime and post back if things improve significantly.

Thanks again,
Lee

commit c72eccdd (2021-02-01): Added missing calls to client.stop() in JsonHttpClient.ino (fixes #1485)