bblanchon/ArduinoJson #1487: Problem compiling ArduinoJson >= 5.8.4 on ameba D (asdk toolchain 1.0.1) and arduino-1.8.13

edward17829991, 2021-02-02:
* short description of the problem:
ArduinoJson compiles until 5.8.4, error messages differ a bit for the different versions after 5.8.4.

* target platform:
Ameba RTL8722DM (AMB 21) version 3.0.7
https://www.amebaiot.com/en/ameba-arduino-summary/

* compiler model and version
(not sure if this info help?)
`ameba_d_asdk_toolchain/1.0.1/bin/arm-none-eabi-g++ -ffunction-sections -march=armv8-m.main+dsp -mthumb -mcmse -mfloat-abi=hard -mfpu=fpv5-sp-d16 -g -gdwarf-3 -nostartfiles -nodefaultlibs -nostdlib -O2 -D__FPU_PRESENT -gdwarf-3 -fstack-usage -fdata-sections -nostartfiles -nostdlib -Wall -Wpointer-arith -Wundef -Wno-write-strings -Wno-maybe-uninitialized -c -Wextra -w -x c++ -E -CC -mcpu=cortex-m33 -DF_CPU=200000000L -DARDUINO=10813 -DARDUINO_AMEBA -DARDUINO_ARCH_AMEBAD -mthumb -DBOARD_RTL8721D -DUSB_VID= -DUSB_PID= -DUSBCON "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"RTL8722DM/RTL8722CSM\"" -Wl,--start-group ...(lots of header files go on)`

* MVCE (https://stackoverflow.com/help/mcve)
compiling the code "StringExample" provided by ArduinoJson 6.17.2 in Arduino IDE 1.8.13.

* compiler output

```
/home/cychen1782/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/cychen1782/arduino-1.8.13/hardware -hardware /home/cychen1782/.arduino15/packages -tools /home/cychen1782/arduino-1.8.13/tools-builder -tools /home/cychen1782/arduino-1.8.13/hardware/tools/avr -tools /home/cychen1782/.arduino15/packages -built-in-libraries /home/cychen1782/arduino-1.8.13/libraries -libraries /home/cychen1782/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8721d -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_569756 -warnings=none -build-cache /tmp/arduino_cache_933258 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_tools.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -prefs=runtime.tools.ameba_d_tools-1.0.4.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -verbose /home/cychen1782/Arduino/libraries/ArduinoJson/examples/StringExample/StringExample.ino
/home/cychen1782/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/cychen1782/arduino-1.8.13/hardware -hardware /home/cychen1782/.arduino15/packages -tools /home/cychen1782/arduino-1.8.13/tools-builder -tools /home/cychen1782/arduino-1.8.13/hardware/tools/avr -tools /home/cychen1782/.arduino15/packages -built-in-libraries /home/cychen1782/arduino-1.8.13/libraries -libraries /home/cychen1782/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8721d -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_569756 -warnings=none -build-cache /tmp/arduino_cache_933258 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_tools.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -prefs=runtime.tools.ameba_d_tools-1.0.4.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -verbose /home/cychen1782/Arduino/libraries/ArduinoJson/examples/StringExample/StringExample.ino
Using board 'ameba_rtl8721d' from platform in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7
Using core 'arduino' from platform in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7
Detecting libraries used...
/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/bin/arm-none-eabi-g++ -ffunction-sections -march=armv8-m.main+dsp -mthumb -mcmse -mfloat-abi=hard -mfpu=fpv5-sp-d16 -g -gdwarf-3 -nostartfiles -nodefaultlibs -nostdlib -O2 -D__FPU_PRESENT -gdwarf-3 -fstack-usage -fdata-sections -nostartfiles -nostdlib -Wall -Wpointer-arith -Wundef -Wno-write-strings -Wno-maybe-uninitialized -c -Wextra -w -x c++ -E -CC -mcpu=cortex-m33 -DF_CPU=200000000L -DARDUINO=10813 -DARDUINO_AMEBA -DARDUINO_ARCH_AMEBAD -mthumb -DBOARD_RTL8721D -DUSB_VID= -DUSB_PID= -DUSBCON "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"RTL8722DM/RTL8722CSM\"" -Wl,--start-group -I. -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/project/realtek_amebaD_va0_example/inc/inc_hp -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos/freertos_v10.2.0/Source/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos/freertos_v10.2.0/Source/portable/GCC/ARM_CM33/non_secure -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/os_dep/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/misc -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/network/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/at_cmd -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/platform -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi/rtw_wpa_supplicant/src -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi/rtw_wpa_supplicant/src/crypto -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/application -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/media/framework -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/example -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/example/wlan_fast_connect -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/mbed/api -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/mbed/hal -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/mbed/hal_ext -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/mbed/targets/hal/rtl8721d -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/lwip/lwip_v2.0.2/port/realtek/freertos -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/lwip/lwip_v2.0.2/src/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/lwip/lwip_v2.0.2/src/include/lwip -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/lwip/lwip_v2.0.2/port/realtek -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/test -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/cmsis -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/fwlib -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/misc -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/drivers/wlan/realtek/include -I/home/cychen1782/.arduino15/pa [... cut ...]
                 from /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/cores/arduino/Arduino.h:122,
                 from /tmp/arduino_build_569756/sketch/StringExample.ino.cpp:1:
/home/cychen1782/Arduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/ctype.hpp:11:13: error: expected ')' before 'char'
 inline bool isdigit(char c) {
             ^
Using library ArduinoJson at version 6.17.2 in folder: /home/cychen1782/Arduino/libraries/ArduinoJson 
exit status 1
Error compiling for board RTL8722DM/RTL8722CSM.
```

bblanchon, 2021-02-02:
Hi @edward17829991,

Thank you very much for reporting this error.

To me, this is a bug in the Ameba core because [they define `isdigit()` as a macro](https://github.com/ambiot/ambd_arduino/blob/47cd35c7935c32e57287757e30781943d57c33b3/Arduino_package/hardware/system/libameba/sdk/component/soc/realtek/amebad/swlib/string/strproc.h#L21) instead of a function.
Anyway, I added a workaround in ArduinoJson; it should be soon merged in the master branch.

Best regards,
Benoit

edward17829991, 2021-02-02:
Nice, thank you so much for the active maintenance.
I really appreciate your work.

👍 

Best,
Edward Chen

edward17829991, 2021-02-02:
(Sorry, for the rapid reopening.)

Hi Benoit,

I append the diff you committed to the source, then a new error appeared. 
Maybe they are related (or not)? So I continued here.

* ArduinoJson version is 6.17.2 and Arduino IDE 1.8.13.
* MVCE:
Not reproducible with example code. It's only failed on my application code.
So I'll provide a minimal code to reproduce this issue if needed.

* compiler output
Arduino: 1.8.13 (Linux), Board: "RTL8722DM/RTL8722CSM"

```
/home/cychen1782/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/cychen1782/arduino-1.8.13/hardware -hardware /home/cychen1782/.arduino15/packages -tools /home/cychen1782/arduino-1.8.13/tools-builder -tools /home/cychen1782/arduino-1.8.13/hardware/tools/avr -tools /home/cychen1782/.arduino15/packages -built-in-libraries /home/cychen1782/arduino-1.8.13/libraries -libraries /home/cychen1782/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8721d -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_921440 -warnings=none -build-cache /tmp/arduino_cache_309379 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_tools.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -prefs=runtime.tools.ameba_d_tools-1.0.4.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -verbose /home/cychen1782/Arduino/RTL8720DN_ScanNetworks/RTL8720DN_ScanNetworks.ino
/home/cychen1782/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/cychen1782/arduino-1.8.13/hardware -hardware /home/cychen1782/.arduino15/packages -tools /home/cychen1782/arduino-1.8.13/tools-builder -tools /home/cychen1782/arduino-1.8.13/hardware/tools/avr -tools /home/cychen1782/.arduino15/packages -built-in-libraries /home/cychen1782/arduino-1.8.13/libraries -libraries /home/cychen1782/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8721d -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_921440 -warnings=none -build-cache /tmp/arduino_cache_309379 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_tools.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -prefs=runtime.tools.ameba_d_tools-1.0.4.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -verbose /home/cychen1782/Arduino/RTL8720DN_ScanNetworks/RTL8720DN_ScanNetworks.ino
Using board 'ameba_rtl8721d' from platform in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7
Using core 'arduino' from platform in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7
Detecting libraries used...
/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/bin/arm-none-eabi-g++ -ffunction-sections -march=armv8-m.main+dsp -mthumb -mcmse -mfloat-abi=hard -mfpu=fpv5-sp-d16 -g -gdwarf-3 -nostartfiles -nodefaultlibs -nostdlib -O2 -D__FPU_PRESENT -gdwarf-3 -fstack-usage -fdata-sections -nostartfiles -nostdlib -Wall -Wpointer-arith -Wundef -Wno-write-strings -Wno-maybe-uninitialized -c -Wextra -w -x c++ -E -CC -mcpu=cortex-m33 -DF_CPU=200000000L -DARDUINO=10813 -DARDUINO_AMEBA -DARDUINO_ARCH_AMEBAD -mthumb -DBOARD_RTL8721D -DUSB_VID= -DUSB_PID= -DUSBCON "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"RTL8722DM/RTL8722CSM\"" -Wl,--start-group -I. -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/project/realtek_amebaD_va0_example/inc/inc_hp -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos/freertos_v10.2.0/Source/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos/freertos_v10.2.0/Source/portable/GCC/ARM_CM33/non_secure -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/os_dep/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/misc -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/network/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/at_cmd -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/platform -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi/rtw_wpa_supplicant/src -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi/rtw_wpa_supplicant/src/crypto -I/ [... cut ...]
Using library ArduinoJson at version 6.17.2 in folder: /home/cychen1782/Arduino/libraries/ArduinoJson 
Using library WiFi at version 1.0.0 in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/libraries/WiFi 
Using library MQTTClient at version 1.0 in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/libraries/MQTTClient 
Using library Watchdog at version 1.0.0 in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/libraries/Watchdog 
exit status 1
Error compiling for board RTL8722DM/RTL8722CSM.

```

bblanchon, 2021-02-02:
@edward17829991, this looks like a new issue.
I cannot reproduce, and the cause isn't obvious.
Can you open a new issue and provide an MCVE?
Thanks!

closed 2021-02-02 by bblanchon

bblanchon, 2021-02-15:
The fix was published in [ArduinoJson 6.17.3](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.3)

commit d2174d15 (2021-02-02): Fixed compilation error when isdigit() is a macro (fixes #1487)