bblanchon/ArduinoJson #1489: USE_LONG_LONG flag cause error on ameba D (asdk toolchain 1.0.1) and arduino-1.8.13

edward17829991, 2021-02-02:
Hi Benoit,

* ArduinoJson version is 6.17.2 and Arduino IDE 1.8.13.
* Target platform:
Ameba RTL8722DM (AMB 21) version 3.0.7
https://www.amebaiot.com/en/ameba-arduino-summary/

* MVCE:

```
#define ARDUINOJSON_USE_LONG_LONG   true

#include <ArduinoJson.h>

StaticJsonDocument<200> list;

void setup() {
  Serial.begin(115200);

  list.to<JsonArray>();

  for (int i = 0; i < 16; ++i) {
    list.add(i);
  }

  serializeJson(list, Serial);
}

void loop() {
  // not used in this example
}
```

After some fiddling, the critical part is the first line "ARDUINOJSON_USE_LONG_LONG".
The data append part doesn't affect much.

* compiler output
Arduino: 1.8.13 (Linux), Board: "RTL8722DM/RTL8722CSM"

```
/home/cychen1782/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/cychen1782/arduino-1.8.13/hardware -hardware /home/cychen1782/.arduino15/packages -tools /home/cychen1782/arduino-1.8.13/tools-builder -tools /home/cychen1782/arduino-1.8.13/hardware/tools/avr -tools /home/cychen1782/.arduino15/packages -built-in-libraries /home/cychen1782/arduino-1.8.13/libraries -libraries /home/cychen1782/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8721d -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_176337 -warnings=none -build-cache /tmp/arduino_cache_309379 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_tools.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -prefs=runtime.tools.ameba_d_tools-1.0.4.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -verbose /tmp/arduino_modified_sketch_152609/StringExample.ino
/home/cychen1782/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/cychen1782/arduino-1.8.13/hardware -hardware /home/cychen1782/.arduino15/packages -tools /home/cychen1782/arduino-1.8.13/tools-builder -tools /home/cychen1782/arduino-1.8.13/hardware/tools/avr -tools /home/cychen1782/.arduino15/packages -built-in-libraries /home/cychen1782/arduino-1.8.13/libraries -libraries /home/cychen1782/Arduino/libraries -fqbn=realtek:AmebaD:ameba_rtl8721d -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_176337 -warnings=none -build-cache /tmp/arduino_cache_309379 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.ameba_d_asdk_toolchain.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_asdk_toolchain-1.0.1.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1 -prefs=runtime.tools.ameba_d_tools.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -prefs=runtime.tools.ameba_d_tools-1.0.4.path=/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_tools/1.0.4 -verbose /tmp/arduino_modified_sketch_152609/StringExample.ino
Using board 'ameba_rtl8721d' from platform in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7
Using core 'arduino' from platform in folder: /home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7
Detecting libraries used...
/home/cychen1782/.arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/bin/arm-none-eabi-g++ -ffunction-sections -march=armv8-m.main+dsp -mthumb -mcmse -mfloat-abi=hard -mfpu=fpv5-sp-d16 -g -gdwarf-3 -nostartfiles -nodefaultlibs -nostdlib -O2 -D__FPU_PRESENT -gdwarf-3 -fstack-usage -fdata-sections -nostartfiles -nostdlib -Wall -Wpointer-arith -Wundef -Wno-write-strings -Wno-maybe-uninitialized -c -Wextra -w -x c++ -E -CC -mcpu=cortex-m33 -DF_CPU=200000000L -DARDUINO=10813 -DARDUINO_AMEBA -DARDUINO_ARCH_AMEBAD -mthumb -DBOARD_RTL8721D -DUSB_VID= -DUSB_PID= -DUSBCON "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"RTL8722DM/RTL8722CSM\"" -Wl,--start-group -I. -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/project/realtek_amebaD_va0_example/inc/inc_hp -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos/freertos_v10.2.0/Source/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/freertos/freertos_v10.2.0/Source/portable/GCC/ARM_CM33/non_secure -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/os/os_dep/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/misc -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/network/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/at_cmd -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/platform -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi/rtw_wpa_supplicant/src -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/api/wifi/rtw_wpa_supplicant/src/crypto -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/application -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/media/framework -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/example -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/example/wlan_fast_connect -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/mbed/api -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/mbed/hal -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/mbed/hal_ext -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/mbed/targets/hal/rtl8721d -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/lwip/lwip_v2.0.2/port/realtek/freertos -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/lwip/lwip_v2.0.2/src/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/lwip/lwip_v2.0.2/src/include/lwip -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/lwip/lwip_v2.0.2/port/realtek -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/test -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/cmsis -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/fwlib -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/soc/realtek/amebad/misc -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/drivers/wlan/realtek/include -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/drivers/wlan/realtek/src/osdep -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/drivers/wlan/realtek/src/hci -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/network/ssl/ssl_ram_map/rom -I/home/cychen1782/.arduino15/packages/realtek/hardware/AmebaD/3.0.7/system/libameba/sdk/component/common/utilitie [... cut ...]
/home/cychen1782/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:70:7: error: pasting ")" and "1" does not give a valid preprocessing token
 using ARDUINOJSON_NAMESPACE::Filter;
       ^
/home/cychen1782/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:71:7: error: pasting ")" and "1" does not give a valid preprocessing token
 using ARDUINOJSON_NAMESPACE::NestingLimit;
       ^
Using library ArduinoJson at version 6.17.2 in folder: /home/cychen1782/Arduino/libraries/ArduinoJson 
exit status 1
Error compiling for board RTL8722DM/RTL8722CSM.

```

Thanks in advance.

Best,
Edward

bblanchon, 2021-02-03:
Hi @edward17829991,

For [ARDUINOJSON_USE_LONG_LONG](https://arduinojson.org/v6/api/config/use_long_long/), only 0 and 1 are supported.
Any other value, like `false` or `true` will cause a compilation error.
I added this information to the documentation.

Best regards,
Benoit

edward17829991, 2021-02-03:
Oh, really.

I used to misuse it with true and being okay.
Now I learned it.

I changed true to 1 and everything compile smoothly, thanks a lot.

Best regards,
Edward

closed 2021-02-03 by bblanchon