bblanchon/ArduinoJson #1491: serializeJson error request for member 'write' in '*....

trawinlegacy, 2021-02-03:
I get the following compiler error while trying to use the serializeJson function with a Wio LTE Cat 1 dev board (Wio LTE Tracker). I am trying to send a JSON message with mqtt, using the PubSubClient library.

serializeJson(doc, payload);
- doc is a StaticJsonDocument<256>
- payload is a 256 item char array

**Compiler Error Log:**
In file included from C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/serialize.hpp:7:0,
                 from C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Json/JsonSerializer.hpp:10,
                 from C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson.hpp:35,
                 from C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson.h:9,
                 from C:\Users\liamt\Documents\Arduino\libraries\EoTSystemJSON\EoTSystemJSON.cpp:31:
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/Writer.hpp: In instantiation of 'size_t ArduinoJson6172_11::Writer<TDestination, Enable>::write(const uint8_t*, size_t) [with TDestination = char*; Enable = void; size_t = unsigned int; uint8_t = unsigned char]':
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/CountingDecorator.hpp:21:12:   required from 'void ArduinoJson6172_11::CountingDecorator<TWriter>::write(const uint8_t*, size_t) [with TWriter = ArduinoJson6172_11::Writer<char*, void>; uint8_t = unsigned char; size_t = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Json/TextFormatter.hpp:132:5:   required from 'void ArduinoJson6172_11::TextFormatter<TWriter>::writeRaw(const char*) [with TWriter = ArduinoJson6172_11::Writer<char*, void>]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Json/JsonSerializer.hpp:90:5:   required from 'size_t ArduinoJson6172_11::JsonSerializer<TWriter>::visitNull() [with TWriter = ArduinoJson6172_11::Writer<char*, void>; size_t = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/VariantFunctions.hpp:18:30:   required from 'typename TVisitor::result_type ArduinoJson6172_11::variantAccept(const ArduinoJson6172_11::VariantData*, TVisitor&) [with TVisitor = ArduinoJson6172_11::JsonSerializer<ArduinoJson6172_11::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/VariantRef.hpp:350:40:   required from 'typename TVisitor::result_type ArduinoJson6172_11::VariantConstRef::accept(TVisitor&) const [with TVisitor = ArduinoJson6172_11::JsonSerializer<ArduinoJson6172_11::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Document/JsonDocument.hpp:20:39:   required from 'typename TVisitor::result_type ArduinoJson6172_11::JsonDocument::accept(TVisitor&) const [with TVisitor = ArduinoJson6172_11::JsonSerializer<ArduinoJson6172_11::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/serialize.hpp:15:34:   required from 'size_t ArduinoJson6172_11::doSerialize(const TSource&, TWriter) [with TSerializer = ArduinoJson6172_11::JsonSerializer; TSource = ArduinoJson6172_11::StaticJsonDocument<256u>; TWriter = ArduinoJson6172_11::Writer<char*, void>; size_t = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/serialize.hpp:22:49:   required from 'size_t ArduinoJson6172_11::serialize(const TSource&, TDestination&) [with TSerializer = ArduinoJson6172_11::JsonSerializer; TSource = ArduinoJson6172_11::StaticJsonDocument<256u>; TDestination = char*; size_t = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Json/JsonSerializer.hpp:113:55:   required from 'size_t ArduinoJson6172_11::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson6172_11::StaticJsonDocument<256u>; TDestination = char*; size_t = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\EoTSystemJSON\EoTSystemJSON.cpp:209:29:   required from here
**C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/Writer.hpp:22:29: error: request for member 'write' in '*((ArduinoJson6172_11::Writer<char*, void>*)this)->ArduinoJson6172_11::Writer<char*, void>::_dest', which is of non-class type 'char*'
     return _dest->write(s, n);**
                             ^
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/Writer.hpp: In instantiation of 'size_t ArduinoJson6172_11::Writer<TDestination, Enable>::write(uint8_t) [with TDestination = char*; Enable = void; size_t = unsigned int; uint8_t = unsigned char]':
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/CountingDecorator.hpp:17:12:   required from 'void ArduinoJson6172_11::CountingDecorator<TWriter>::write(uint8_t) [with TWriter = ArduinoJson6172_11::Writer<char*, void>; uint8_t = unsigned char]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Json/TextFormatter.hpp:149:5:   required from 'void ArduinoJson6172_11::TextFormatter<TWriter>::writeRaw(char) [with TWriter = ArduinoJson6172_11::Writer<char*, void>]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Json/TextFormatter.hpp:71:19:   required from 'void ArduinoJson6172_11::TextFormatter<TWriter>::writeFloat(T) [with T = float; TWriter = ArduinoJson6172_11::Writer<char*, void>]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Json/JsonSerializer.hpp:60:5:   required from 'size_t ArduinoJson6172_11::JsonSerializer<TWriter>::visitFloat(ArduinoJson6172_11::Float) [with TWriter = ArduinoJson6172_11::Writer<char*, void>; size_t = unsigned int; ArduinoJson6172_11::Float = float]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/VariantData.hpp:43:51:   [ skipping 2 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/VariantRef.hpp:350:40:   required from 'typename TVisitor::result_type ArduinoJson6172_11::VariantConstRef::accept(TVisitor&) const [with TVisitor = ArduinoJson6172_11::JsonSerializer<ArduinoJson6172_11::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Document/JsonDocument.hpp:20:39:   required from 'typename TVisitor::result_type ArduinoJson6172_11::JsonDocument::accept(TVisitor&) const [with TVisitor = ArduinoJson6172_11::JsonSerializer<ArduinoJson6172_11::Writer<char*, void> >; typename TVisitor::result_type = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/serialize.hpp:15:34:   required from 'size_t ArduinoJson6172_11::doSerialize(const TSource&, TWriter) [with TSerializer = ArduinoJson6172_11::JsonSerializer; TSource = ArduinoJson6172_11::StaticJsonDocument<256u>; TWriter = ArduinoJson6172_11::Writer<char*, void>; size_t = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/serialize.hpp:22:49:   required from 'size_t ArduinoJson6172_11::serialize(const TSource&, TDestination&) [with TSerializer = ArduinoJson6172_11::JsonSerializer; TSource = ArduinoJson6172_11::StaticJsonDocument<256u>; TDestination = char*; size_t = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Json/JsonSerializer.hpp:113:55:   required from 'size_t ArduinoJson6172_11::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson6172_11::StaticJsonDocument<256u>; TDestination = char*; size_t = unsigned int]'
C:\Users\liamt\Documents\Arduino\libraries\EoTSystemJSON\EoTSystemJSON.cpp:209:29:   required from here
**C:\Users\liamt\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Serialization/Writer.hpp:18:26: error: request for member 'write' in '*((ArduinoJson6172_11::Writer<char*, void>*)this)->ArduinoJson6172_11::Writer<char*, void>::_dest', which is of non-class type 'char*'
     return _dest->write(c);**
                          ^
exit status 1
Error compiling for board Wio Tracker LTE.

trawinlegacy, 2021-02-03:
SOLVED: 
Needed to use the following:
serializeJson(doc, payload, 256);

@bblanchon 
Can you please add the comment you made in the following closed issue:
#1449 
to the MQTT how to pages ie: 
https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/

closed 2021-02-03 by trawinlegacy

bblanchon, 2021-02-03:
Link added in #1449.