bblanchon/ArduinoJson #1492: Measuring json length with measureJson with a pointer of a Document as parameter (Question)

judalvarezca, 2021-02-03:
I'm trying to build a query in my sketch, and my program follows the next structure:

```
void gatherData(...) {
  DynamicJsonDocument doc(200);
   .
   .
   .

  doc["temp_celsius"] = temp;
  doc["Humidity"] = hum;
  doc["mq7"] = mq7Value;
  doc["mq135"] = mq135Value;

  postData(&doc);
}
```
```
void postData(DynamicJsonDocument *doc) {

  ClientHTTP http(client);
   .
   .
   .

  size_t payloadLength = measureJson(doc);
  if(!http.POST("/post", payloadLength)) {
    Serial.println("Failed to POST");
    return;
  }
  Serial.println("test 2");
#if ARDUINOJSON_VERSION_MAJOR == 6
  serializeJson(doc, http);
#endif
}
```

However when I build the program, I get the following error:

```
.pio/libdeps/heltec_wifi_kit_32/ArduinoJson/src/ArduinoJson/Serialization/measure.hpp:15:34: error: request for member 'accept' in 'source', which is of pointer type 'ArduinoJson6172_91::BasicJsonDocument<ArduinoJson6172_91::DefaultAllocator>* const' (maybe you meant to use '->' ?)
```

I'm a bit lost on how to pass the parameter to the measureJson method, so it will be a lot of help if you could guide me here.

Thank you very much for the hard work, the library is simply Amazing with the capital A.

bblanchon, 2021-02-05:
Hi @judalvarezca,

You simply have to dereference the pointer, like so:

```c++
measureJson(*doc);
```

However, if you had read "The Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/), you would know that this is not the way C++ programmers pass an object to a function without copying it. Instead of pointers, we prefer C++ references:

```c++
void gatherData(...) {
  DynamicJsonDocument doc(200);
  .
  .
  postData(doc);
}

void postData(const DynamicJsonDocument &doc) {
  .
  .
  size_t payloadLength = measureJson(doc);
  .
  .
  serializeJson(doc, http);
}
```

Best regards,
Benoit

judalvarezca, 2021-02-10:
Thank you very much!

closed 2021-02-10 by judalvarezca